nRF24/RF24 #160: setCRCLength(RF24_CRC_DISABLED) doesn't appear to work

RetiredPro, 2015-11-08:
I tried  setCRCLength(RF24_CRC_DISABLED) and disableCRC(). In both cases, a subsequent
getCRCLength() gives the enumerator for the 8 bit CRC.

TMRh20, 2015-11-08:
Per the data sheet v2.0 section 7.7 crc is always enabled when ESB
(autoack) is enabled.

RetiredPro, 2015-11-08:
Thanks for the prompt response.  I am really impressed with your knowledge of the nRF24L01+ and like devices. I have decided to use your library because you are actively maintaining it and are always quick to respond to questions and suggestions. I offer the following description of my experience on this topic so that others may benefit.

I looked up section 7.7 and found it was unrelated. I noticed that my pdf file  was for version 1 as indicated on the cover page. I went to the Nordic website and found that their data sheet for the nRF24l01+ was listed as version 1. I did a web search specifically looking for "nordic nrf24l01+ datasheet v2.0".  There is a v2.0 data sheet on the nordicsemi.com site (http://www.nordicsemi.com/eng/content/download/2730/34105/file/nRF24L01_Product_Specification_v2_0.pdf). In the v2.0 data sheet I found the reference you made above. Interestingly, the v2.0 data sheet does not have a version number on the title page and the date listed (July 2007) is earlier than the version 1.0 data sheet date(September 2008).

Now a tiny suggestion. Because setCRCLength(RF24_CRC_DISABLED) and disableCRC() are sometimes unable to honor the the request (autoack enabled) perhaps they should both be bool functions instead of void so they can return false if they couldn't make the change. That would be similar to how setDataRate reports success or failure.

RetiredPro, 2015-11-08:
Ooops I hit the close button by mistake

TMRh20, 2015-11-09:
Well, I guess there are two options, it could return either a boolean value indicating success or failure, or it could just combine the two functions, ```setCRCLength()``` and ```getCRCLength()``` so that setCRCLength returns the CRC length.

Boolean:
```
if(setCRCLength(RF24_CRC_DISABLED){
  Serial.println("Set CRC Fail");
}
```

rf24_crclength_e:
```
rf24_crclength_e length = setCRCLength(RF24_CRC_DISABLED);
Serial.print("CRC Length set to: ");
Serial.println(length*8);
```

I'm not sure myself which is the better option, so please let me know what you think.

RetiredPro, 2015-11-10:
I think the choice is a matter of style.  My personal choice is Boolean. With rf24_crclength_e, a test against the value I requested would determine if corrective action is necessary. That approach is a little more verbose.

How would you handle radio.disableCRC()?

TMRh20, 2015-11-10:
> How would you handle radio.disableCRC()?

The same way? Return the current value.

The external logic probably should use a boolean, as you suggest. The only reason I brought it up is because it would have to do the same logic internally to return a boolean value.

RetiredPro, 2015-11-15:
After thinking about this issue, I would like to retract the request for an enhancement. There is no need to change the existing library and a slight detriment to doing so. Returning a bool or CRC length enumerator can easily  be done with a tiny  wrapper that suits the programmer.

TMRh20, 2015-11-21:
Well, thanks for the input on this anyhow. I think it is a good idea to at least update the documentation with this info, so please leave this issue open as a reminder to do so.

closed 2015-11-25 by TMRh20