nRF24/RF24 #161: RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.

marc-gist, 2015-11-09:
using the python wrapper, after some time i get constant "RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc." errors.

How do I catch this, and restart my program or better yet, the wrapper?

Thanks,

TMRh20, 2015-11-09:
As you are probably aware, this does typically indicate a hardware problem, so it is usually best to attempt resolution of the issue as opposed to managing it. 

If you are using a breadboard, its usually best practice to limit the length of connecting wires and use a small capacitor on the 3.3V & GND pins. Using a small RF24 PCB, I rarely have seen this error.

To manage the issue, you can edit RF24_config.h and uncomment ``` #define FAILURE_HANDLING ```
Then, continue to check the radio.failureDetected boolean value in order to identify errors. Typically, calling radio.begin() and re-configuring your settings will get things up and running, however, in my limited testing, the error detection may not be 100% depending on exactly how it fails.

marc-gist, 2015-11-09:
Thank you for the fast response!
I do have have capacitor, however, it is on a breadboard for now as i'm still testing it with the RPi.
I would like the option to try and "self recover" without human interaction (or at least send an alert email/etc should something fail like this).

TMRh20, 2015-11-09:
Np. 

> I assume i can access the flag in python

Its there. You will have to recompile RF24 after changing the define though. I always forget to mention that...

closed 2015-11-25 by TMRh20

eflukx, 2016-02-05:
Having a similar problem. I'm using the cpp lib directly, and an adapted version of the `gettingstarted` example. The NRF24 hardware module is adequately decoupled (with 2x100nF and 2x10uF mlcc). I'm using a RasPi 2 with Rasbian . Could it be that the SPI/GPIO is not configured properly in the kernel and some other process/module is interfering? I've tried multiple nRF24L01+ modules, so I don't expect a one-off hw fail.

Important to know is that the module _is_ actually working, but I do get the error messages. Furthermore, CPU use seems quite high (intermittently during the RX loop).

Any ideas?

Oitzu, 2016-02-05:
@eflukx you should probably also have a look at issue #205