nRF24/RF24 #162: python wrapper doesn't work until after I've run the C code first

marc-gist, 2015-11-09:
So I've noticed some strange behaviour.

If I startup my raspberryPi, run the rxRF24Network/hellowworld_rx.py
Nothing is ever received (displayed on screen).

If I run examples_RPi/hellowworld_rx
(the C++ program), it receives the data fine.

If I then run the python code, it will now work fine.

any ideas?

all commands are run "sudo"

mz-fuzzy, 2015-11-09:
Python wrapper just propagate calls from python to C++ lib, so there is just a small chance that problem with such symptoms is in the wrapper itself.
What versions (commit ids) of rf24/rf24network are you using? In past, there was a initialization issue fixed with random impacts similar to what you are reporting: https://github.com/TMRh20/RF24/issues/101

marc-gist, 2015-11-09:
Not sure where to check the commit id... i just downloaded the zip files last week.  I will try a git pull and recompile and see what happens.

marc-gist, 2015-11-10:
pulled and compiled and installed latest master branch... still happens... after reboot, I have to run the C++ program first, then I can run the python program and receive data.

TMRh20, 2015-11-12:
Not sure if python or C++ related, but I can't seem to replicate this issue either way. I'm testing on Raspbian Jessie - 4.1.7+ on Model B Revision 2.0 (see [here](http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/)) on this RPi

That being the case I don't have many suggestions other than updating (apt-get update & rpi-update ) and/or running raspi-config if using raspbian and enable spi at boot, and make sure it isn't in the blacklist at /etc/modprobe.d/raspi-blacklist.conf (although it makes little difference on my device)

Please post the results of running the following if no luck:
``` cat /proc/cpuinfo ```
``` uname -rv ```
``` lsmod | grep spi ```

@mz-fuzzy With the RF24Network helloworld_rx example, it seems I have to change the line ``` ms, number = unpack('<qi', payload) ``` to ``` ms, number = unpack('<ii', payload) ``` or I get an error: ``` struct.error: unpack requires a string argument of length 12 ```  I could have sworn I tested this before...

marc-gist, 2015-11-12:
I am currently downloading the latest Jessie build and will try that.
I have a Model B+ Revision 1.2 (I believe) that I'm currently trying it on. I will also try on my Raspberry Pi 2 Model B v1.1 when I get the new Jessie build downloaded.

Thank you for trying to reproduce! greatly appreciate the work here!
here are results of those commands:

> processor       : 0
> model name      : ARMv6-compatible processor rev 7 (v6l)
> BogoMIPS        : 2.00
> Features        : half thumb fastmult vfp edsp java tls 
> CPU implementer : 0x41
> CPU architecture: 7
> CPU variant     : 0x0
> CPU part        : 0xb76
> CPU revision    : 7
> 
> Hardware        : BCM2708
> Revision        : 0010
> Serial          : 00000000fd00deb5

> 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015

> lsmod | grep spi 
appears to return nothing... so I'll look into why that is via the config and ensure its enabled during boot and see if that is the issue.

again, can't thank you enough for the great work and fast responses here!

marc-gist, 2015-11-12:
spi now loading at boot... didn't fix the issue. I still have to run the C++ program first to get the python to work.
again: i'll test further on the latest release of rasbian jessie when i can.

marc-gist, 2015-11-12:
Found the Issue!!!
So in the code you have this line:

> # Setup for GPIO 22 CE and CE0 CSN for RPi B+ with SPI Speed @ 8Mhz
> radio = RF24(RPI_BPLUS_GPIO_J8_22, RPI_BPLUS_GPIO_J8_24, BCM2835_SPI_SPEED_8MHZ)

So I put the select (CE) pin to GPIO22 (pin 15 on the rPi).
Doesn't work

in the CPP program you have:

> // Setup for GPIO 22 CE and CE1 CSN with SPI Speed @ 8Mhz
> RF24 radio(RPI_V2_GPIO_P1_15, BCM2835_SPI_CS0, BCM2835_SPI_SPEED_8MHZ);

After finally reading better, I see why I missed it.  In python one it appears pin 22 is selected (but really its not GPIO22, its PIN22, which is GPIO25!) doh! :)

Why when I ran the C++ code first, then the python it worked... the C++ code would select/enable the radio (which I guess stays enabled after I "CTRL-C" to stop the program) allowing the python code to work.

please fix code/notes to make it more clear that its not GPIO 22, but PIN 22.

marc-gist, 2015-11-12:
Sorry, I guess this issue/error should technically be in the RF24Network code; yet the issue is still in the code for the RF24 examples where the notes say "GPIO 22", when it should read "GPIO 25" or "PIN 22"

mz-fuzzy, 2015-11-13:
> With the RF24Network helloworld_rx example, it seems I have to change the line ms, number = unpack('<qi', payload) to ms, number = unpack('<ii', payload) or I get an error: struct.error: unpack requires a string argument of length 12 I could have sworn I tested this before...
*edit to add: Must have tested with two RPi I'm thinking, and using different lengths for millis()...

correct, c-rpi and arduino examples use 2 longs to send, so in order to make it compatible, python example should be changed to pack/unpack "<ii" in both helloworld_tx and helloworld_rx, and reading should expect 8 bytes there.

TMRh20, 2015-11-13:
@marc-gist Thanks for the input. It seems this is one of those things I don't really look at much anymore, although its right in front of me, but there definitely is some inconsistency there that should be fixed asap. See https://github.com/TMRh20/RF24/issues/158 also.

@mz-fuzzy Thanks!

marc-gist, 2015-11-13:
@mz-fuzzy Thanks! Great work... I'm glad it was just me not really reading closely and hopeful this post helps some other newbies like me until you get around to making those minor "fixes"!

closed 2015-11-25 by TMRh20