nRF24/RF24 #165: error timeot in write function

versoul, 2015-11-13:
in write function is delay of 85 milliseconds that well works at a speed 1mb but if speed 250kb the answer doesn't manage to come, I consider that on a place of rigidly written down figures 85 there has to be a variable txRxDelay which is calculated depending on the speed of the processor and the transmitter and there figure 85 too appears but only at a speed 1mb

TMRh20, 2015-11-13:
In the stopListening() function there is a delay, which in Arduino AVR devices, works out to 85ms at 1MBPS, but it is adjusted to 155 at 250KBPS as per [here](https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L1432)

Without some code demonstrating an issue, and some more info, I can't do much, because it works fine for me. [This](https://en.wikipedia.org/wiki/Wikipedia:Reference_desk/How_to_ask_a_software_question) is usually helpful.

versoul, 2015-11-14:
ok look my code

receive
```
/**********************************************************/
#include <SPI.h>
#include "RF24.h"
byte addresses[][6] = {"1Node","2Node"};
RF24 radio(9,10);

struct dataStruct{
  unsigned long _micros;
  byte cmd;
}myData;

/********************commands******************************/

/**********************************************************/

void setup() {
  Serial.begin(115200);
  Serial.println("*** Start");
  
  radio.begin();

  radio.setDataRate(RF24_250KBPS);
  //radio.setChannel(118);
  radio.setRetries(0,15);
  //radio.setAutoAck(1);
  radio.setPALevel(RF24_PA_MAX);
  radio.openWritingPipe(addresses[0]);
  radio.openReadingPipe(1,addresses[1]);

  radio.startListening();
}
void loop() { 
  if( radio.available()){
    while (radio.available()) {
      radio.read( &myData, sizeof(myData) );
    }
    radio.stopListening();
    radio.write( &myData, sizeof(myData) );     
    radio.startListening();
    Serial.print(F("Sent response "));
    Serial.print(myData._micros);  
    Serial.print(F(" : "));
    Serial.println(myData.cmd);
 }
}
```

tranmit
```
/**********************************************************/
#include <SPI.h>
#include "RF24.h"
byte addresses[][6] = {"1Node","2Node"};
RF24 radio(9,10);
struct dataStruct{
  unsigned long _micros;
  byte cmd;
}myData;

/********************commands******************************/
#define cmdLine 1;
/**********************************************************/

void setup() {
  Serial.begin(115200);
  Serial.println("*** Start");
  
  radio.begin();

  radio.setDataRate(RF24_250KBPS);
  //radio.setChannel(118);
  radio.setRetries(0,15);
  //radio.setAutoAck(1);
  radio.setPALevel(RF24_PA_MAX);
  radio.openWritingPipe(addresses[1]);
  radio.openReadingPipe(1,addresses[0]);
  
  radio.startListening();
}
void loop() { 
  radio.stopListening();
  myData._micros = micros();
  myData.cmd = cmdLine;
  Serial.print("Send, size = ");
  Serial.println(sizeof(myData));
   if (!radio.write( &myData, sizeof(myData) )){
     Serial.println(F("failed"));
   }
  radio.startListening();
  unsigned long started_waiting_at = micros();
  boolean timeout = false;
  while ( ! radio.available() ){
    if (micros() - started_waiting_at > 200000 ){
        timeout = true;
        break;
    }      
  }  
  if ( timeout ){
      Serial.println(F("Failed, response timed out."));
  }else{
      radio.read( &myData, sizeof(myData) );
      unsigned long time = micros();
      
      // Spew it
      Serial.print(F("Sent "));
      Serial.print(time);
      Serial.print(F(", Got response "));
      Serial.print(myData._micros);
      Serial.print(F(", Round-trip delay "));
      Serial.println(time-myData._micros);
  }
  delay(2000);
}
```

it writes on my pc 
```
Sent 49195388, Got response 49165604, Round-trip delay 29784 >>> Sent response 51197256 : 1
Send, size = 5
failed
Sent 51227044, Got response 51197256, Round-trip delay 29788 >>> Sent response 53228908 : 1
Send, size = 5
failed
Sent 53258692, Got response 53228908, Round-trip delay 29784 >>> Sent response 55260552 : 1
Send, size = 5
failed
Sent 55290340, Got response 55260552, Round-trip delay 29788 >>> Sent response 57292204 : 1
Send, size = 5
failed
Sent 57321992, Got response 57292204, Round-trip delay 29788 >>> Sent response 61355520 : 1
Send, size = 5
failed

```

Data successfully reach there and back, but function of write returns false always as it seems to me because isn't in time

versoul, 2015-11-14:
>>> its second serial window

TMRh20, 2015-11-14:
In 250KBPS mode, the datasheet v1.0 [here]( https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf ) says ARD (retry delay) must be 500us or more ( ```radio.setRetries(1,X);``` )

Please leave this issue open as a reminder to update the RF24 class docs with this information.

versoul, 2015-11-14:
I tried to change on 1 or 2 began to work even worse how I can include a test conclusion of messages from library in the series port for testing? after all I think matter in number 85 and I want to experiment

TMRh20, 2015-11-14:
I'm guessing english is not your first language, because I'm not sure exactly what you are asking. If you can rephrase the question it might help me to understand.

If the radios are close together maybe try ``` radio.setPALevel(RF24_PA_MIN); ``` also.

closed 2015-11-25 by TMRh20