nRF24/RF24 #167: Chip stop sending (or receiving)

madshobye, 2015-11-26:
Dear RF24

I am testing a setup with arduino to arduino communication. The setup works just fine but after some hours the receiver stops receiving messages. I might reappear but it somehow becomes unstable. My best guess is that the sender becomes unreliable somehow. A reset of the sender stations solves the problem. This is however undesirable. 

I do not think this is an issue with your library since we have seen it with a library version we made ourselves (before we found yours). This seems to be some issue with the chip or the setup we have. I am writing here hoping that someone may have knowledge of this problem and knows what to do with it.

The data send from one station to the other is stored in the cloud every 30 seconds (the local update frequency between the two rf24 stations is every second). 

[image]

[image]

As can be seen in the two images the accumulated energy (lowest graph) stops updating (becomes constant for a while). This means that the wifi module is uploading the same number over and over again and does not receive any new numbers. After a random period of time it looks like it get one more update, but then reverts back to being stalled. That is the jump in the graph on the second image. Further there seems to be faulty data once in a while (the downwards spike on the second graph).

If anybody has any pointers of how to proceed from here I would be grateful.

If have tried to ad a reset code that runs if 30 packets has been lost on the sender side:

  if (!RFradio.write((unsigned char*)datagram, 7))
      {
        failCounter++;
        if (failCounter > 30)
        {
          begin(keyidLocal);
          failCounter = 0;
        }
      }

It does not seems to do the job, but it might be too soon to tell.

TMRh20, 2015-11-26:
> If anybody has any pointers of how to proceed from here I would be grateful.

I would suggest that the problem is likely to be related to your hardware, code or environment.

**Hardware:**
*Method A:*
By default, the Arduino usually hangs if the radio stops responding. Blink an LED or something to indicate the node is active.
*Method B:*
1. Edit RF24_config.h and uncomment ``` #define FAILURE_HANDLING ```
2. See [here](http://tmrh20.github.io/RF24/classRF24.html#a2e40fe66d1231a333aa2534e8491f828) for info on detecting hardware failures

**Software:**
*Method A:*
Share your code.
*Method B:* Try using RF24Network, which reduces/removes a lot of problems otherwise encountered in custom designs with the RF24 driver.

**Environment:**
Ghosts??? lol

madshobye, 2015-11-26:
Thank you for your response. I have been testing and looking for ghosts :)
I think the problem may be the WIFI router it is close by and relatively
powerful. I will approach some of the suggestion you give.

 I have now changed the wifi channel to 13 and have the nrf on 0. This
should be the opposite ends of the spectrum.

I have not been able to find the range of the channels in the library? E.g.
what is the maximum number I an set the channel to.

Also this is a ebay copy of the Chip (chip and might be a knockoff). I hear
this is not the most reliable version - if anybody are able to confirm or
deny the different types I would be grateful.

stefa168, 2015-11-26:
Do you have any photos of your actual NRF board? It would be very helpful because from what I know there are three main versions of it:
 * one that is with real NRF24L01+ ICs
 * one that is with "SI24R1" written on it (note that this chip **doesn't** work like the real IC
 * one that [appears as the real IC, but is the knockoff with another name on it](http://hackaday.com/2015/02/23/nordic-nrf24l01-real-vs-fake/)

Since you actually can transmit, I suppose you have the real one (or at least one that works like the real one), so the best solution would be to do as @TMRh20 said: use the failure handling functions

madshobye, 2015-11-27:
Here you have an image of the one we have:

[image: Inline image 1]

[image: Inline image 2]

According to the link from hackaday it looks genuine?

Avamander, 2015-12-14:
@madshobye Your pictures are not visible.

closed 2016-02-06 by Avamander