nRF24/RF24 #168: Errors with nodeconfig.cpp in rf24mesh, rf24network, etc

pedroiriso, 2015-11-26:
Hello, i'm have a problem when i want to compile the example, an error occurred in nodeconfig.h, these are the errors.

In function "const eeprom_info_t& nodeconfig_read()"
error: "struct eeprom_info_t has no member named "temp_calibration"

In function "void nodeconfig_listen()"
error: "set_temp_calibration" was not declared in this scope.

In function "void set_temp_calibration(int16_t)"
error: "struct eeprom_info_t" has no member named "temp_calibration"

Please help me! 
BYE!

TMRh20, 2015-11-27:
The nodeconfig sketch is outdated and left included only for reference.

closed 2015-11-27 by TMRh20

pedroiriso, 2015-12-02:
TMRh20, do you have a new version of nodeconfig? or help me to compile a little example