nRF24/RF24 #170: Question: Maximum transfer-rate between RPi and Arduino?

Oitzu, 2015-12-04:
Hi,
i'm currently trying to transfer image data (about 100kb per picture) from an arduino to the RPi over the nrf24l01+.
All works fine an well but i wonder whats the max. transfer rate possible?
With the transfer examples i get about 5-10KB/s from arduino directly to RPi.
With the transfer example directly between 2 arduinos i get 40KB/s.
Outputing the results (the received data) on the arduino<->arduino setup with the serial connection between the arduino and the RPi lowers the rate to 15KB/s. (Baudrate: 1000000)

Any idea for faster setups?

Oitzu, 2015-12-05:
Okay.. i have done a failure on the arduino<->arduino setup.
I shouldn't write the received data with Serial.print but with Serial.write, Serial.print slows the arduino down due to conversion. This recovers the rate between arduino and arduino. Beginner fail.

But one question remains: The low rate of the nrf24l01+ directly connected to the RPi.
Maybe a problem with the bad regulated voltage output of the RPi?
Or some timing issues with the RPi?

I will try to smooth the voltage, but maybe someone else has an idea or experiences about the RPi?

TMRh20, 2015-12-05:
You can see my initial testing showing max transfer rates [here](https://www.youtube.com/watch?v=mODflsK1Ab0)  ~48KB/s @1MBPS w ESB, 100KB/S @1MBPS w/o ESB

Using the BCM driver for RPi, transfer rates are approximately the same between RPi and Arduino. Compiling with SPIDEV support reduces response times.

This works about the same for me using two RPi, one on a breadboard and one using a small PCB with no cap on 3.3v rail.

TMRh20, 2015-12-05:
Just a thought, if you are using the LNA+PA versions of the module with RPi, you may need a more robust power supply, or you can try calling ``` radio.setPALevel(RF24_PA_MIN); ```

Oitzu, 2015-12-05:
Okay, i need to look again on the RPi.
I'm using the LNA+PA version on the sending side (with a additional robust power supply) and the "normal" version on the receiving side. (Arduino/RPi)

Oitzu, 2015-12-05:
If i compile with SPIDEV it seems i don't get a functioning lib:
```
transfer.cpp:39:12: error: ‘RPI_V2_GPIO_P1_15’ was not declared in this scope
 RF24 radio(RPI_V2_GPIO_P1_15, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);
```

Or do i need to change this constants in der transfer example?

Oitzu, 2015-12-05:
Okay i smoothed the supply voltage to the nrf24l01+ on the RPi site and now get 
```
Transfer complete at 40.22 KB/s
0 of 10000 Packets Failed to Send
```
with the standard configuration. This seems reasonable to me and the supply voltage of the RPi seems to be the limiting factor.

Using ESB is probably not a good idea while transfering pictures?

TMRh20, 2015-12-05:
@Oitzu Yup, use GPIO numbers directly with SPIDEV. There should be a mention in the examples.

> Using ESB is probably not a good idea while transfering pictures?

Just a matter of preference and a different process IMHO.

Oitzu, 2015-12-06:
Okay. I think we can close this issue.
Tanks and keep up your great work! :)
If you would like i can report back some results later.

closed 2015-12-06 by Oitzu