nRF24/RF24 #171: RPi B+ DataRate 250kbs enable to send

tetnobic, 2015-12-04:
Hi, 
When I modify the sample file examples_RPi/gettingstarted.cpp : 
I just add this

`radio.setDataRate(RF24_250KBPS)`

after line 97

and start sample in ping_out mode
I have an Hardware Fail...
Can you help me ? 

For information : 
- In other data rate it works fine
- I use a nRF24L01+ (see log after) (I see in doc that setting RF24_250KBPS will fail for non-plus units)

copy/paste of STDOUT : 
*******
RF24/examples/GettingStarted/
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO22
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5     = 0xff 0xc4 0xc5 0xc6
TX_ADDR          = 0x65646f4e32
RX_PW_P0-6       = 0x20 0x20 0x20 0x00 0x00 0x00
EN_AA            = 0x3b
EN_RXADDR        = 0x06
RF_CH            = 0x4c
RF_SETUP         = 0x27
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 250KBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX

 ************ Role Setup ***********
Choose a role: Enter 0 for pong_back, 1 for ping_out (CTRL+C to exit)
1
Role: Ping Out, starting transmission

Now sending...
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
********

TMRh20, 2015-12-05:
If it works fine at other data rates, my only assumption would be that your module does not support 250kbps mode.

There are lots of clones that appear to be valid nrf24l01 chips.

http://sigrok.org/wiki/Protocol_decoder:Nrf24l01
http://hackaday.com/2015/02/23/nordic-nrf24l01-real-vs-fake/

tetnobic, 2015-12-14:
Hi, thanks for you response...
I try with other new module and same behaviour....

So I do some other test : 

With the same module : 
sender is an Arduino (more precisely only an ATmega328p)
receiver is an RPi
at Data Rate = 250KBPS on each side 
Everything is OK !

Have you an idea ??

Avamander, 2015-12-14:
@tetnobic Do you have capacitors attached to the modules? (A 100uF electrolytic to the module's power pins)

tetnobic, 2015-12-15:
Hi, thanks for response avamander
No I haven't capacitors attached to the modules (not on RPi Side and not on Arduino Side)
Should I use one ? on both side ?

Avamander, 2015-12-15:
You should on both sides. In many cases it fixes many transmission related
problems and reduces packet loss.

tetnobic, 2015-12-15:
Ok,so I need to soldered 100uF capacitors across the VCC/GND pins on my nRF24l01 modules in parallel : 

Connect VCC of the module to 3.3V of the arduino
Connect GND of the module to GND of the arduino
Connect the +ve side of the capacitor to the 3.3V/VCC connection
Connect the -ve side of trhe capacitor to the GND connection

I wiil try this and come back.
Thanks a lot

closed 2016-03-23 by TMRh20