nRF24/RF24 #172: Using interrupts on RPi

Oitzu, 2015-12-08:
Any idea how this could easily be done?
Using an interrupt would have the advantage to react faster on incoming messages without polling all the time.
I've seen the wiringPi library supports interrupts on the raspberry but hughly collides with the bcm... library.

Any idea or should i try to port it to the wiringPi library, if possible?

TMRh20, 2015-12-09:
Hate to say it but no idea really. Was something I wanted to look more into but haven't so wiringPi might be a good option.

Oitzu, 2015-12-09:
Okay. I will see how far i will get. The wiringPi lib seems to offer a few less SPI configuration options, but as far as i see the standard configuration provided by wiringPi should work with the nrf24l01+.

Oitzu, 2015-12-10:
Just to keep updated:
I try to solve it with pure wiringPi but didn't get the SPI part working.
The interrupt function in the wiringPi library is just spawning a thread and watching the exit status of the wiringPi gpio util. I will try to rebuild this function in the rf24 lib.

Oitzu, 2015-12-10:
In a rough hacked together interrupt based transfer version i got
```
Transfer complete at 31.25 KB/s
0 of 10000 Packets Failed to Send
``` 
The cpu usage dropped from 99% to 10%.

TMRh20, 2015-12-10:
Looking good! This could benefit RPi/Linux usage quite a bit.

Oitzu, 2015-12-11:
How to best implement this? I was thinking about an include like the bcm lib. But the interrupt handling would require installed wiringPi.
Make the interrupt functions optional through build-flags?
Its theoreticly not a new arch but an extensions to the RPi arch.

TMRh20, 2015-12-11:
Heh, I don't know, there seems to be a few options.

The bcm lib supports some sort of [event detection for GPIO](http://www.airspayce.com/mikem/bcm2835/event_8c-example.html) functionality too, but I haven't been able to look at it closely.

If  going with wiringPi, I would suggest going all the way, and just making a pure wiringPi port with a compile-time option. I can probably contribute the SPI and general porting code if you want to work on interrupts.  

I would also assume there should be a standard kind of c++ way of handling or managing interrupts in linux directly, without any additional libraries. I'll see about finding some time to look at this all a bit closer, and wait to hear what you think or have found.

Oitzu, 2015-12-11:
About interrupt bcm lib vs. wiringPi
---------------------------------------------
I've seen that event detection for GPIO while readme states:
```
Pin event detection is supported by polling (interrupts are not supported).
```
And looking at the code, it's a endless loop checking a detection bit.

The wiringPi library states:
```
With a newer kernel patched with the GPIO interrupt handling code, you can now wait for an interrupt in your program. This frees up the processor to do other tasks while you’re waiting for that interrupt.
```
http://wiringpi.com/reference/priority-interrupts-and-threads/

The library spawns a thread that do a blocking poll what seems to be a kernel provided interface. (/sys/class/gpio/gpio%d/value)

About going all the way with wiringPi:
---------------------------------------------------
This was my first approach, with a seperate RF24_arch_config, but i couldn't get the SPI working. I could clean it up and push it to branch, if you wish.
The wiringPi SPI functions are probably not as fast as the bcm provided.

About interrupts in Linux:
------------------------------------
Interrupts are provided in the kernel by device drivers in linux.
Thats why the wiringPi library states "newer kernel patched with the GPIO interrupt handling code".

Combining best of both worlds:
-------------------------------------------
The way that worked for me, was copying the interrupt relevant parts from the wiringPi library.
Maybe you should have a look at my latest commit in https://github.com/Oitzu/RF24.
An transfer example using the interrupt is included.
I reached 
```
Transfer complete at 44.09 KB/s
0 of 10000 Packets Failed to Send
```
with 7% CPU usage. The CPU usage drops to the desired 0% if no transmission is active.

But it's probably not the best way to go, my experience in this field aren't good enough to judge.

Oitzu, 2015-12-12:
Note i tried also, to include this into the RF24Network library:
```
void intHandler()
{
        network.update();
        while ( network.available() ) {     // Is there anything ready for us?
                RF24NetworkHeader header;        // If so, grab it and print it out
                payload_t payload;
                network.read(header,&payload,sizeof(payload));
        }
//        printf("Received payload # %lu at %lu \n",payload.counter,payload.ms);

        counter++;
}
```
but only reaching:
```
Transfer complete at 12.62 KB/s
0 of 10000 Packets Failed to Send
```
is this due to 'Network' overhead?

TMRh20, 2015-12-12:
Ok, I like the 'combining best of both worlds' as it seems the way you've set it up is probably the most seamless way of bringing interrupts to RPi, and I also had a bit of a chance to play with this.

I made some small changes to the [transfer_interrupt](https://gist.github.com/TMRh20/fa015027c6e39fd984f3) example which seems to run the radio at max data rate:

**1MBPS w/AA**
Rate: 46.78 KB/s
Payload Count: 1462
Rate: 46.94 KB/s
Payload Count: 1467
Rate: 47.17 KB/s
Payload Count: 1474

**1MBPS wo/AA**
Rate: 37.47 KB/s
Payload Count: 1171
Rate: 102.43 KB/s
Payload Count: 3201
Rate: 101.50 KB/s
Payload Count: 3172
Rate: 77.44 KB/s
Payload Count: 2420

**2MBPS wo/AA**
Rate: 62.50 KB/s
Payload Count: 1953
Rate: 223.17 KB/s
Payload Count: 6974
Rate: 48.64 KB/s
Payload Count: 1520
Rate: 0.00 KB/s

CPU usage in all cases is much much lower than while using the standard SPI polling method. In the first test ((RPi B v2) it was about 25% max, and at 2 MBPS it did spike quite a bit, but of course only as required. 

This is working excellent so far, but one thing to note, I changed CRC validation to 16-bit since I was noticing problems at 2MBPS with AA disabled. (extra payloads rcvd @ 8-bit)

The slow-down with RF24Network may be related to long delays, as with the transfer example, but there will be overhead since 8-bytes of every 32-byte payload is used for the network header, making the payload size only 24-bytes. 

With RF24Gateway & two RPi, I can get 20KB/s @ 1MBPS w/AA , and that includes huge TCP/IP headers, so RF24Network should be able to go much faster.

Oitzu, 2015-12-12:
I used an arduino (8Mhz pro mini) on transmitting side and an RPi on receiving side. (Which is my ultimative goal)
https://github.com/Oitzu/RF24Network/blob/master/examples/transfer_tx/transfer_tx.ino
https://github.com/Oitzu/RF24Network/blob/master/examples_RPi/transfer_rx.cpp

Do you may know if there is any delay on the arduino side due to the RF24Network lib?

TMRh20, 2015-12-13:
> Do you may know if there is any delay on the arduino side due to the RF24Network lib?

Not really, there are some minor delays that affect both, but shouldn't make a major difference.

I did some more tweaking, but with your network examples: [Arduino/TX](https://gist.github.com/TMRh20/a74e948ca4d117120b98)  & [RPi/RX](https://gist.github.com/TMRh20/75a0cc9ed27441a40125)

**Sending fragmented payloads of 144 bytes:**

Network Rate: 31.20 KB/s  Byte Count: 31200
Data Rate: 24.77 KB/s  Byte Count: 24768
Network Rate: 30.96 KB/s  Byte Count: 30960
Data Rate: 24.77 KB/s  Byte Count: 24768
Network Rate: 30.96 KB/s  Byte Count: 30960
Data Rate: 24.77 KB/s  Byte Count: 24768
Network Rate: 30.96 KB/s  Byte Count: 30960
Data Rate: 24.96 KB/s  Byte Count: 24960
Network Rate: 31.20 KB/s  Byte Count: 31200

**Sending non-fragmented payloads of 24 bytes:**

Data Rate: 17.22 KB/s  Byte Count: 17216
Network Rate: 21.52 KB/s  Byte Count: 21520
Data Rate: 17.22 KB/s  Byte Count: 17216
Network Rate: 21.52 KB/s  Byte Count: 21520
Data Rate: 17.18 KB/s  Byte Count: 17184
Network Rate: 21.48 KB/s  Byte Count: 21480

**Explanation:**
When sending normal payloads, the radio always enters RX mode after transmitting, slowing down transfers. 
When sending fragmented payloads, the radio only enters RX mode after transmitting all of the payloads, speeding up transfers.

I also added in some extra code to count every byte and calculate the overhead involved in fragmented payloads and display the user-data-rate vs the network-data-rate.

These results are a bit more on par with what I was expecting of the network lib.

Oitzu, 2015-12-13:
Ahhh okay! I was trying to avoid fragmentation, because i thought it would add even more overhead.
But the best for my project is possibly to use the plain rf24 library, because the topology consist of just 4 nodes connected in a star and needs highest rate possible. (Sending pictures from arduino to RPi)

But i'm going off-topic, i remember this topic was about interrupts. ;)
I will add a few more comments in the code an open an pull req.

Oitzu, 2015-12-13:
In the risk of going offtopic again:
Or do you have a idea to speed it up? I first tried the mysensors library, but it is to slow and bloated for this purpose and now i don't know in which direction i should invest my time.

henrikekblad, 2015-12-13:
I'm following you know ;)

Oitzu, 2015-12-13:
@henrikekblad hi there! :) it wasn't meant to discredit mysensors ^^
sorry if my words sounded like that. 
I love the mysensors library for my home-sensors etc. but maybe the comfortableness has its price i invested days to try to get it working, but don't see a way to get it running reasonable.

Since you are already reading... do you maybe have an idea for that?
I'm facing 2 problems with mysensors: Outdated Serial Gateway. (This issue was initially to get interrupts in the serial gateway) and huge processing on the sending side before sending.. This results in a much slower datarate.

henrikekblad, 2015-12-13:
Just kidding with you. MySensors isn't for every project. Especially bad for transferring images (even if people have done it). 

The RPI gateway code is _really_ outdated. Doesn't support signed messages nor saving of routing table  between restarts.
 
I hoped on some help incorporating RPI s into the "Arduino" project (which probably will be renamed/restructured) so we'd get one instance of MySensors core code to maintain.

TMRh20, 2015-12-13:
I've been looking a bit closer at RF24Network, but am still currently unsure of why exactly such a dramatic slowdown, and to see if I can speed it up, but I'm having doubts about how much performance can be improved.  
  
It seems that routing adds a large amount of overhead, since the header requires a from and to address, each 16-bits, where, in a single-level network, the sender is identified by the pipe-number and recipient by who gets the data.  Its always a balance of reliability vs speed it seems.
  
I'm going to do a bit more tweaking and testing, but I'm not expecting any dramatic improvements to come easily.  With a small non-routing network, its just that much easier to achieve high-performance transfers. 

I'm not the expert here, but transferring really large volumes of data like images is kind of a specialty thing, a bit out of scope of both mysensors and rf24network I would say, however if reliability and routing was more important that speed, I would actually go higher on the stack and use RF24Ethernet.

Avamander, 2015-12-13:
@henrikekblad Not related to this issue, but I am currently inspecting ways of implementing signing messages to RF24(Mesh).

henrikekblad, 2015-12-13:
@Avamander 
Cool. You might find bits and pieces to re-use from the MySensors library. We have both a software emulation and hardware implementation of the ATSHA chip for signing. There's also soft-encryption even if it is pretty meaningless IMHO.

Avamander, 2015-12-13:
@henrikekblad The MySensors library implementation is a PITA to read. (The documentation lacks in many places and the code is so fragmented, blergh)

Oitzu, 2015-12-13:
@henrikekblad you can't expect me to get that. The potato battery don't source enough power for higher text processing. :P
I looked again at the MySensors library do you think it might be worth the try to implement:
3 functions: startStream, sendStream, endStream
startStream: Opens sending pipe stop listining.
sendStream: sends an prepared "packet" directly through the pipe
endStream: close & start listning.

The only point i don't know is how to prepare the packet. Maybe first manipulate an MyMessage Object to have the right header and after that only manipulating the data, leaving the header untouched?

@TMRh20 well i don't know if it is really the added header. Wouldn't it that if it is only the header that the network rate should be higher? I've seen that also a lot of preprocessing on the "packets/frames" is done, could it be that the arduino just needs a little time to do that?

TMRh20, 2015-12-13:
@Oitzu Yup, its not just the header, but routing in general. Managing data in a half-duplex system is the overall challenge.

Oitzu, 2015-12-13:
Well yeah, the project is really out of scope of both. But i decided on the arduino+nrf24 combo based on low power consumption and great reliability on "long" distances with the pa/lna modules. This is going to be solar powered outdoor nodes. ;)

TMRh20, 2015-12-13:
Not that its gotten out of hand, but lets try to keep this conversation as constructive as possible. The things that make RF24 and MySensors great IMHO are the users and contributors, with some very active development, and ongoing improvements. Both libs contribute to each other, and if we all had the resources available, the docs, code etc would be awesome. We are largely dependent on contributions (and you) to make things better.  
  
Just had to throw my $0.02 in because I've received my fair share of negative comments regarding code/documentation quality, and I think any code/documentation that works is better than none.

henrikekblad, 2015-12-13:
Yep, we've started to improve on the documentation a bit..  But heavy refactoring usually throws the project into doc-limbo-land sometimes.

@fallberg actually started documenting the signing part a while ago (and with CI running it should be much easier to keep things updated).
http://ci.mysensors.org/job/MySensorsDoc/Doxygen_HTML/group__internals.html

Still a mess, but we're trying..

Avamander, 2015-12-13:
@henrikekblad I think if you'd either remove robots.txt or fix the search button, things would be much better.

henrikekblad, 2015-12-13:
@Oitzu 
Sorry for messing up this issue.

@Avamander 
What are you talking about? We might better keep this discussion elsewhere. You can email me directly on henrik.ekblad@mysensors.org.

Oitzu, 2015-12-13:
@henrikekblad no problem. I was the first going offtopic. :+1: 
For further discussion of the interrupt function a pull request is already open.
Any opinions on the Stream part?

henrikekblad, 2015-12-13:
Sure, might be doable, but not in combination with signed messages.

At I'm having a hard time seeing how a repeater node would work in such a  setup.

Oitzu, 2015-12-13:
Well first step was to get it running on a direct connection. I will give this a try.
For a repeater use i can imagine to send an internal message through the route that will tell the gateway and repeaters that a stream will start.
The repeaters would open pipes to the corresponding nodes and repeat them on a, for this stream, fixed route.
But this will probably cutting half the data-rate.

Signed messages: Only option without going low speed would to calculate a checksum on the whole stream content and send this signed on stream end.

Oitzu, 2015-12-13:
@henrikekblad 
```
MyMessage camDataMsg(CAM_CHILD_ID, DATA_STREAM);
camDataMsg.last = (uint8_t)2;
camDataMsg.sender = (uint8_t)2;
camDataMsg.destination = (uint8_t)0;
mSetVersion(camDataMsg, PROTOCOL_VERSION);
mSetCommand(camDataMsg, C_SET);
mSetRequestAck(camDataMsg,false);
gw.sendStream((uint8_t)0, &camDataMsg.set(buf, 24), 32)
```
sendStream is a call to MyTransportNRF24::sendStream(uint8_t to, const void* data, uint8_t len)
which calls only rf24.write(data, len, to == BROADCAST_ADDRESS);

Just results in failed transmits... any idea, what i'm doing wrong?

TMRh20, 2015-12-14:
@Oitzu FYI I was able to get some decent improvements with some modifications to the RF24Network lib. It looks like there was a delay affecting fragmented multicast payloads, and I worked out a problem with the radio having to drop to standby-I mode instead of staying in standby-II mode during fragmented writes.

**Standard fragmented payload of 144 bytes:**
Data Rate: 24.77 KB/s  Byte Count: 24768
Network Rate: 33.02 KB/s  Byte Count: 33024
Data Rate: 25.49 KB/s  Byte Count: 25488
Network Rate: 33.98 KB/s  Byte Count: 33984

**Direct (multicast) fragmented payload of 144 bytes**
Data Rate: 41.47 KB/s  Byte Count: 41472
Network Rate: 55.30 KB/s  Byte Count: 55296
Data Rate: 41.18 KB/s  Byte Count: 41184
Network Rate: 54.91 KB/s  Byte Count: 54912

The [helloworld_tx.ino example](https://gist.github.com/TMRh20/a74e948ca4d117120b98) gist has been updated to use the direct write on line 65. It would be changed to ``` bool ok = network.write(header,&data,144,0); ``` for a standard write. Thanks for your input, it seems to have helped make a nice bit of difference in transfer rates.

Oitzu, 2015-12-14:
@TMRh20 wow! This is really a great improvement in speed! :)
Is this also tested with Arduino/TX & RPi/RX?
Will definitely try it out as soon as i have the time to set it up.

TMRh20, 2015-12-14:
@Oitzu Yup, mostly tested with Arduino to RPi including the above results. Note: The direct write should work with any node, regardless of address, given it is in range of the master node, but of course these messages won't be routed.

Oitzu, 2015-12-15:
Highest i got out of it was:
```
Data Rate: 20.16 KB/s  Byte Count: 20160
Network Rate: 26.88 KB/s  Byte Count: 26880
```
Could be multiple reasons. Are we both using the standard settings? (1mbps, 16bit crc)
I'm using a 8mhz pro mini, you probably other 16mhz arduino? Could processing-time be a factor?

I also need to rule out on my side that my connection is as good as yours. :) My 2 nodes are in different rooms at this moment.

TMRh20, 2015-12-15:
I'm just using the examples I referenced in my [gists](https://gist.github.com/TMRh20) with CE/CSN pins adjusted as required. There were a couple bugs with the initial RF24Network commit, so make sure to update.

An 8mhz device will perform slightly slower. I am using 16mhz pro-mini and get slightly better results using an Arduino Due, worse on ATTiny so I believe processing time & SPI speed is a factor. My devices are about 8-10 meters apart in separate rooms & tested at close range also.

*Due:* 27/37KB/s @ 144byte payload, 44/59KB/s @ 144byte w/direct write and  15/20KB/s@24byte payload
*ATtiny:* 8/11KB/s @ 24byte payload

Also, a 16mhz avr device can run SPI at 8mhz, but 8mhz device can only run 4mhz if I remember correctly. (clock/2). 

If you can get fast & consistent results with core RF24 examples, that may suggest overhead in RF24Network as the main factor. If they are noticeably lower @8mhz vs 16 then it might suggest or include SPI speeds as a factor. 

In my testing with RF24 core examples, things need to be working very well to get consistent 100KB/s or 200KB/s results, so anything less at 16mhz at close range is potentially indicative of minor issues.

Oitzu, 2015-12-15:
Seen that, i merged it in my fork and used your gists.
The distance doesn't seem to affect the rate, i moved them closer.

I will try to get some 8mhz vs. 16mhz testing results tommorow, i have also 16mhz pro minis laying arround.

What could maybe also an issue: I use on one side an pa/lna module and on the other side an "normal" module. I have multiple versions laying arround and will also test this tommorow.
You will probably never know how much the china clones are out of spec.

Oitzu, 2015-12-16:
A little bit slow on testing...
After connecting the secons PA/LNA module i no longer get messages on the rpi.
Despite the status looks okay...
```
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5     = 0x33 0xce 0x3e 0xe3
TX_ADDR          = 0xe7e7e7e7e7
RX_PW_P0-6       = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA            = 0x3e
EN_RXADDR        = 0x3f
RF_CH            = 0x5a
RF_SETUP         = 0x03
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x3f 0x04
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_LOW
```

Need first du figure out what is going wrong.

Avamander, 2015-12-16:
Attach a more powerful power supply to the nrf24l01. Connect Pi and power
supply grounds together.

Oitzu, 2015-12-16:
@Avamander Thanks for the help! But i already done that. :(
I'm currently clueless whats the problem. Even the old setup don't want to work... time for checking cables and cables and cables... ^^

Oitzu, 2015-12-16:
But i seriosly don't find anything wrong.
The status displayed above seems to display a correct communication with set registers.
But if i try to send anything:
```
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
```
Naturally i checked the wiring multiple times in the meantime.

PeterSek, 2015-12-17:
Hi Oitzu
Are you mixing nrf24 module types PA/Non PA
I had a problem using sample ping program stopped receiving when using
mixed PA/Non PA models
when both ends where PA, it worked, when both end where non pa it also
worked ok
unfortunately I did not have time to get o the bottom of it
Peter

Avamander, 2015-12-17:
Seems like the fake&non-fake issue if two different types of radio don't
work.

You could try and change PA level to the lowest.

Oitzu, 2015-12-17:
@PeterSek: my modules normally work flawless between different nrf24l01+ versions. But there are a lot of different versions on the market.

@Avamander: Already tried that, too.

Most of the PA&LNA Version problems are due to too weak power supply OR! what most people don't know: improper shiedling! Most of the PA&LNA are not shielded, this is a problem on the MAX Level, because the signal itself is so strong that it overlays any incoming signal to the nrf24 chip.
This can be solved with manual shielding provided by tinfoil.

Despite all this... my problem seems to be related to the RPi itself. I tested the modules on an arduino, and it works fine.

Oitzu, 2015-12-17:
After a long powercycle over night everything working again... don't ask me f**ing why... ^^

Oitzu, 2015-12-17:
I finally got to test the RF24Network lib with 2 PA&LNA modules. I got out:
```
Data Rate: 25.20 KB/s  Byte Count: 25200
Network Rate: 33.60 KB/s  Byte Count: 33600
```
I will test it later also with an 16mhz pro mini.

TMRh20, 2015-12-20:
Big thanks to @Oitzu for all the effort in getting interrupts going and everybody for the input!

**Current State of things:**

1. Interrupts are are working and pushed to master branch w/updated examples. I have tested quite a bit at RF24 layer and a bunch with other libs. Plz advise of any bugs etc with current code or examples.

2. Interrupts use [pthreads](https://computing.llnl.gov/tutorials/pthreads/)

3. SPI functions are protected automatically now, using a [mutex](http://www.freebsd.org/cgi/man.cgi?query=pthread_mutex_init&sektion=3&apropos=0&manpath=Red+Hat+Linux%2Fi386+9) so threads should be relatively easy to manage. 
      *A thread attempting to lock a mutex that is already locked by another thread	is suspended until the	owning thread unlocks the mutex	first.*

3. The interrupt handler is protected with a separate mutex from SPI:
 ``` rfNoInterrupts() & rfInterrupts()``` provide a simple interface to protect data from being modified at the same time by separate threads, and only pauses interrupts via mutex locking. It does not disable them. 
<sup>**note:** This is similar to using ```sei``` and ```cli``` on AVR devices to protect data from modification by interrupts per [here](http://www.nongnu.org/avr-libc/user-manual/group__avr__interrupts.html) and [here](https://www.arduino.cc/en/Reference/NoInterrupts) for those more familiar with Arduino.</sup>
Users *should* be able to use their own [thread-safe queues](https://www.justsoftwaresolutions.co.uk/threading/implementing-a-thread-safe-queue-using-condition-variables.html) 

4. @Oitzu you suggested a function to disable interrupts, which I think it is a good idea. Are you still planning to add that?

5. Documentation will be updated with some info regarding interrupts soon, so now is a great time for any suggestions, questions etc.

Oitzu, 2015-12-21:
@TMRh20 i tried with your example https://gist.github.com/TMRh20/a74e948ca4d117120b98 to transfer image data. But getting garbage data on the other end out of it.
Anything i need consider on the receiving side?
It's working fine without the ,0.

TMRh20, 2015-12-22:
The ,0 is a direct write using multicast, so packets are not retried or acknowledged at any level, radio or network. The gist example was missing quotes around the array data, but I doubt that is the issue.

Since you are sending image files, I assume they are larger than your defined MAX_PAYLOAD_SIZE, which would mean you are sending in chunks and re-assembling them at the receiver. My best guess in this case, is that you are missing a few chunks. The header.id is incremented every send, so that can be used to identify missing chunks and re-request.

Oitzu, 2015-12-22:
Yeah you are right, the image resides in an fifo buffer, read out over spi. I'm reading and sending in 144byte chunks. There are probably missing a few chunks on the receiving side. It should also possible to "rewind" in the fifo buffer to get past packages and retransmit them.
But it is probably not worth the hassle, because in the end the overhead of manual acknowledge and rerequest would slow down the process more then simply using the retry and auto-ack functions.

wiredman, 2016-01-10:
okay to get interupts running. you need to have the latest wiringpi from git, do the ./build step then, sudo ldconfig so you can use the new gpio tool. then you have to use the command "gpio edge wiringpi.pinnumber falling", then inside of the examples you use the port gpio number. the DEFAULT GPIO utility in Raspbian, will not work.

schmurfy, 2016-01-10:
thanks for your work on this, this is awesome !
I made an attempt a few months ago but could not get anything working :/

Oitzu, 2016-01-10:
@wiredman i think i do not fully understand your comment. Do you have any problems with the new interrupt functionality?
This function uses the wiringPi gpio util because its the only tool afaik that supports interrupts on the RPi.
You do not need to call the gpio util manually the rf24 library will do this internally when you attach an interrupt. The examples use the bcm gpio numbers to not introduce 2 Pin numbering concepts to the class.
Whats the default gpio utility in raspbian? The wiringpi gpio util is the only gpio util that i know that calls itself the gpio util.

wiredman, 2016-01-10:
oitzu, my comment was just more instructions on how to get it working, I had a few issues while getting it to work; and thats how I worked around it.

prophet-roshak, 2016-01-18:
Btw, MRAA has built in Interrupt capabilities on GPIO pins, with opening secondary thread and waiting for event, like wiringpi does, so I would like to go ahead and try this on my Galileo board, will report speed upon completion :)

mz-fuzzy, 2016-01-24:
to leverage all advantages of the interrupt handling in examples, add some sleep/delay into role_pong_back branch of the infinite while loop in the main function. This will prevent burning cpu at 100%.

mz-fuzzy, 2016-01-25:
After checking the implementation pf IRQ support and thinking how to expose it to python, I found out that perhaps it's not needed - it's sufficient just to use existing RPi.GPIO lib as suggested by @dmarkey in #175. Commited in 031d3888e837fa2ef23e92e3923ddc10defbcbf8.

closed 2016-11-10 by TMRh20