nRF24/RF24 #174: NRF module won't receive but will send

Sharan123, 2015-12-14:
Hi i have a ebay bought NRF module (meaning most probably not genuine)
One on the RPi one on the arduino. Now RPi one can send and receive packets - no problem, but the arduino one can just send packets and cannot receive. Can someone guide me on what the problem might be , how to detect it and solve it ? Note - original arduino UNO works no problem but the ebay UNO is having issues receiving packets.

Avamander, 2015-12-14:
Try attaching a capacitor to the power pins directly on the NRF module. (100uF should be enough)

Sharan123, 2015-12-14:
10uF attached on GND ,VCC on the NRF module both on RPi and Arduino.

Avamander, 2015-12-14:
Try larger.

Sharan123, 2015-12-14:
Sorry  to bother you but why would a capacitor that size be needed for receiving packets , since i am able to send them, especially since on the original arduino the packets gets received? I suspect that MOSI, MISO is the problem just don't know where i would pinpoint. However i will try that also as a last resort :)

Avamander, 2015-12-14:
If I understood your first description. The issue is only on clone Uno. Clones tend to have cheaper parts and the 3.3V regulator is one of the parts that they are saving on. Noisy power would maybe not influence sending but in receiving noise plays a bigger role.

itsjustvenky, 2015-12-14:
capacitors are used to filter noise. We have to eliminate noise in electronic components esp. when there is a RF.

Sharan123, 2015-12-14:
Ok, that makes sense, will try 100uF capacitor and tell you how it went ^^

Sharan123, 2015-12-14:
Ok the 100uF capacitor is doing something thats for sure. I have the 100uF electrolytic capacitor (i don't know if it should be tantalum one since usually i had a 10uF tantalum? )
Now when i send data (from arduino) in the beginning i got the ACK for every 2nd packet (but later on the more i tested the less ACK's i received - 1 in 10 got the ack)
Also arduino on the receiving side still doesn't get the packets.

Do i need yet a bigger capacitor and how much bigger ? :/ (also the type you recommend if needed tantalum or electrolytic ?)

Avamander, 2015-12-14:
What happens if you send less often?

Sharan123, 2015-12-24:
Ok, first sorry for the delay.

I have tried adding a ceramic capacitor of 10nF - which resulted in nothing. 
I have tried replacing the power source from the Fake Arduino to the 2x1.5V series connected batteries giving me the 3V power needed. 

Here is how that went:

Rpi2 Running helloworld_tx:
Sending ..
failed.
Sending ..
failed.
_________________________________
Arduino running helloworld_rx: 
Package #0 at 0
Package #0 at 0 

First i was thinking that the code is all messed up because arduino int and rpi int are not the same length after modifying the code so that the payload uses uint32_t i got
#0 at 0
#33554432 at 0 
#33554432 at 0  
... 

So yeah that didn't worked. 
I am willing to try changing the frequency (SPI_SPEED) , but how do i do that on Arduino - can i use the setClockDivider() in SPI and when do i call it ? (After or before radio.begin() , after or before network.begin ?)

I was almost sure that the battery would work since that should be a "stable power source" ?

Avamander, 2015-12-24:
Did you connect battery ground with Arduino ground?

Sharan123, 2015-12-24:
Exactly the issue i came here to address just now. I forgot hence creating a "not well connected circuit"

After the connection " Battery -"  -  " Arduino GND " AND after tightening the connections IT FINALLY WORKS the same as an original UNO.  Rpi is receiving packets and arduino is sending them and vice versa - when arduino is on the receiving end it actually receives a "non-garbage" packet and RPI2 gets the ACK. 

Case closed. Sorry for the long thread where the main answer is:
--------------------------------------------------------------------------------------------------
*Sometimes there just isn't a capacitor that can compensate for the noise coming from 3.3V fake arduino. Other 3.3V sources MUST be applied IN COMBINATION with the 100uF capacitor (maybe having in mind the ceramic one also if that fails). 3.3V Voltage regulators might also be useful here*

PeterSek, 2015-12-25:
I find using PA_Min when radios are near each other also helps?

closed 2016-02-20 by TMRh20