nRF24/RF24 #175: Python receive code spins

dmarkey, 2015-12-15:
Hi there,

The while loop in the python example spins and hogs the cpu.

Is there a better design here?

mz-fuzzy, 2015-12-15:
This is not python-specific issue, the same situation is in RPi C++ example.
See discussion https://github.com/TMRh20/RF24/issues/172.

dmarkey, 2015-12-15:
OK fine, I think I could possibly use GPIO and the extra pin interrupt?

mz-fuzzy, 2015-12-15:
Yes, read the neighboring thread I linked above.

dmarkey, 2015-12-16:
```python
import time
from RF24 import *

radio = RF24(RPI_BPLUS_GPIO_J8_22, RPI_BPLUS_GPIO_J8_24, BCM2835_SPI_SPEED_8MHZ)

def my_callback(channel):
    if radio.available():
        while radio.available():
            len = radio.getDynamicPayloadSize()
            receive_payload = radio.read(len)
            print('Got payload size={} value="{}"'.format(len, receive_payload.decode('utf-8')))

import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)

GPIO.setup(2, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.add_event_detect(2, GPIO.FALLING, callback=my_callback)

pipes = [0xF0F0F0F0E1, 0xF0F0F0F0D2]

radio.begin()
radio.enableDynamicPayloads()
radio.setRetries(5,15)
radio.printDetails()

radio.openWritingPipe(pipes[1])
radio.openReadingPipe(1,pipes[0])
radio.startListening()

while 1:
    time.sleep(1000)
```

Here is a working example that doesn't spin the CPU, just connect GPIO2 to the INT pin.

mz-fuzzy, 2015-12-17:
@dmarkey : Excellent, thanks! It would be great if you could you prepare alternative pingpair_dyn_int.py example with utilizing the interrupt pin, test it thoroughly and post a pull request.

mz-fuzzy, 2016-02-17:
Resolved by 031d3888e837fa2ef23e92e3923ddc10defbcbf8, closing.

closed 2016-02-17 by mz-fuzzy