nRF24/RF24 #176: QUESTION: Compilation memory usage and artifact size of python wrapper. (RPi2)

yozik04, 2015-12-16:
I am compiling on RPi2 with 1GB of memory.
In lib makefiles I have changed *-march=armv6zk* occurrences to *-march=armv7-a*. Otherwise it does not compile.

When I compile in RPi/pyRF24
```CC=gcc CFLAGS="-Ofast -mfpu=vfp -mfloat-abi=hard -march=armv7-a -mtune=arm1176jzf-s" ./setup.py build```
During this command gcc takes up to 83% of my RPi memory(1GB-256Mb video memory).
```gcc -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv7-a -mtune=arm1176jzf-s -fPIC -I/usr/include/python2.7 -c pyRF24.cpp -o build/temp.linux-armv7l-2.7/pyRF24.o```
*time* tells that process uses 626Mb.
I have to stop all my services to compile the wrapper. Otherwise it hangs with out of memory.

Why it needs so much memory?

Produced RF24.so artifact weights 1,56 Mb. Why it is so heavy? Can it be optimized?

pyRF24.o is 2,57Mb

Do you have same artifact sizes and same memory consumption during compilation?

Note: *time -v* is a good tool to measure memory consumption. *sudo apt-get install time*
```CC=gcc CFLAGS="-Ofast -mfpu=vfp -mfloat-abi=hard -march=armv7-a -mtune=arm1176jzf-s" time -v ./setup.py build```

TMRh20, 2015-12-17:
> In lib makefiles I have changed -march=armv6zk occurrences to -march=armv7-a. Otherwise it does not compile.

Hmm, there should be an easy way to auto-detect this, maybe based on revision number or something?

Per [this](https://github.com/TMRh20/RF24Network/issues/44#issuecomment-159992832) I also had problems compiling the python wrapper on RPi-Bv2 and had to [increase swap size](https://www.bitpi.co/2015/02/11/how-to-change-raspberry-pis-swapfile-size-on-rasbian/) to 758MB

Not sure why it takes so much, but it seems to be a general thing. @mz-fuzzy may be able to provide some info.

mz-fuzzy, 2016-01-20:
Big memory footprint is disadvantage of boost.python library which vastly utilizes templates and macros. It's a price paid for simple implementation of the interface. I guess substantial improvement would be only to use a different library/approach to interface C from python.

A bit smaller library footprint can be reached by compiling using optimization for code size by "-Os" option.

closed 2016-02-20 by TMRh20