nRF24/RF24 #177: well, couple of things going wrong since p3 update,

WERAQS, 2015-12-20:
there is a problem with one of the new cards on my network after the update.. any solutions?

1) -there is a problem with len, why?-
pingpair_dyn.cpp:201:42: warning: ‘len’ may be used uninitialized in this function [-Wmaybe-uninitialized]
       radio.write( receive_payload, len ); 

2) -compile error, nothing major I presume?-
root@RPi2_16G_01:~/RPi/pyRF24# ./setup.py build
running build
running build_ext
building 'RF24' extension
creating build
creating build/temp.linux-armv7l-2.7
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c pyRF24.cpp -o build/temp.linux-armv7l-2.7/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++

3) -it just hangs and does nothing..-
root@RPi2_16G_01:~/examples_RPi# python pingpair_dyn.py
pyRF24/examples/pingpair_dyn/

TMRh20, 2015-12-20:
1. Its a minor warning because: ``` uint8_t len; ``` should be ``` uint8_t len=0; ```

2. Python warning is normal as pre-p3 update

3. Hrrmm, not sure. Make sure no other RF24 software is running. I recently tested pingpair_dyn on two RPi and an Arduino with C++, python, and new interrupt examples. Make sure you are running the latest code also.

WERAQS, 2015-12-23:
well, a fresh raspbian install solved it all, thanks.

WERAQS, 2015-12-23:
[image]
updating to ubuntu mate was nice till this :)

TMRh20, 2015-12-24:
Hrmm... it looks like its not liking the compiler flags, but I dunno.

You could try editing or commenting out the makefile line ``` CCFLAGS=-Ofast -mfpu=vfp -mfloat-abi=hard -march=$(ARCH) -mtune=arm1176jzf-s ``` but running RF24 on Ubuntu your kind of on your own unless somebody else has some experience with it.

closed 2016-02-06 by Avamander