nRF24/RF24 #178: Compiling on Edison (mraa)

Sulejman, 2015-12-23:
Hi, when i try to compile this on Edison it requires me Arduino.h file:

:~/RF24# make install -B RF24_MRAA=1
cp utility/MRAA/includes.h utility/includes.h
g++ -Wall -fPIC  -c RF24.cpp
In file included from RF24.cpp:10:0:
RF24_config.h:58:23: fatal error: Arduino.h: No such file or directory
   #include <Arduino.h>
                       ^
compilation terminated.
Makefile:78: recipe for target 'RF24.o' failed
make: *** [RF24.o] Error 1

TMRh20, 2015-12-24:
The MRAA dev was/is difficult because I don't have an Intel board, but it looks like its not picking up the proper defines for linux or mraa at all.

**Workaround**
You could try running ```sudo make install -B RF24_MRAA=1 CCFLAGS=-DMRAA``` but you still might end up having to add a ```#define MRAA``` to your progs that use RF24.

**Fix**
Are you running a linux OS or other? In RF24_config.h there is a line ```#if ( defined (__linux) || defined (LINUX) ) && defined( __arm__ ) || defined(MRAA) ``` which could be modified to add an include for your system. If you are able to identify one that works, please advise so it can be added.

madarax64, 2016-01-11:
Hello,
I've had this exact same issue on an Intel Galileo, running the Ubilinux distro. I've documented some of it here: http://terminal28.blogspot.com.ng/2015/10/ubilinux-on-intel-galileo-tmrh20s-rf24_16.html
In summary, editing  the RF24_config.h file as follows fixed it for me:

Locate the section called "User Defines". At its end, simply add "#define MRAA" (minus the quotes). That should fix the issue. All you need to do would be to go ahead and run make -B RF24_MRAA=1

Sulejman, 2016-01-11:
Thaank you very much,

I have to try it as soon as I come to the Edison again tomorrow.

madarax64, 2016-01-11:
Okay,
Let us know how it goes. Good luck!

rimolive, 2016-04-16:
I'm trying to compile RF24 lib in Intel Edison too. The issue with Adruino.h is solved but the Makefile uses install command and I think that it is not provided in Yocto installed in the board.

If anyone knows how to workaround this I really appreciate.

mz-fuzzy, 2016-04-28:
@rimolive you can grab 'install' script for example from automake http://git.savannah.gnu.org/cgit/automake.git/tree/lib/install-sh
and put it into your system.
It's also possible to include the install script to RF24 repo, but I'm not sure whether it's worth it.

closed 2016-11-10 by TMRh20