nRF24/RF24 #180: Attiny861 + nRF24L01 work fine

nopnop2002, 2015-12-26:
I tried ATtiny861+NRF24L01.
It's work fine.
It's Japanese, but I have a report below.

http://nopnop2002.webcrow.jp/nRF24L01/nRF24L01-ATtiny-3.html

Modify to RF24_config.h

#elif defined(__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__) || defined(__AVR_ATtiny2313__) || defined(__AVR_ATtiny4313__) || defined(__AVR_ATtiny861__)

Add to RF24.cpp

#elif defined(__AVR_ATtiny861__)
// these depend on the core used (check pins_arduino.h)
// tested with google-code core
#    define DI   9   // PB0
#    define DO   8   // PB1
#    define USCK 7   // PB2
#    define SS   6   // PB3

Pin Coneection

 * for ATtiny861
 *  VCC  3.3V
 *  MISO PB0(Pin#1)
 *  MOSI PB1(Pin#2)
 *  SCK  PB2(Pin#3)
 *  CE   PA6(Pin#12)
 *  CSN  PA7(Pin#11)

Avamander, 2015-12-26:
Nice to know, if I have time I'll try and update documentation.

TMRh20, 2016-03-23:
Added

closed 2016-03-23 by nopnop2002