nRF24/RF24 #182: Some module works when SPI_CLOCK_DIV4, not SPI_CLOCK_DIV2 (Arduino Uno 16MHz)

NicksonYap, 2015-12-26:
I am using Arduino Uno 16MHz, and sometimes I test the module/library using Mega 
3.3V supply decoupling capacitors are added close to IC for all NRF24 modules.

I have tested various NRF24 modules.
[image]
I'll name the module from the left M1 to the last one, M4. All are labeled as NRF24L01+ (with the plus) and RF24 library detects to be so.
M1: most common version of the NRF24L01 module, could be using clone chips. med price
M2: very cheap version of the module, quite certain it's a clone. cheapest
M3: same as M1 but with power amplifier (PA). higher price
M4: 100mW shielded module with PA, highest price.

For each type of module, I have a few quantity. And I test using all of them to make sure they are consistent, not a faulty module.

Using GettingStarted example with radio.printDetails(); and printf included:
All M1, M2, M3 modules outputs status register 0x0e can TX and RX but all M4 outputs 0x06 and does not work.

I then tested M4 using Maniacbug's RF24 library and RadioHead's. They work well will M4. 
M2 is unreliable using RadioHead's library. Maniacbug's library works with all M1 to M4 modules.

I then roughly compare Maniacbug's library with TMRh20 and changed TMRh20's RF24.cpp line 44

		_SPI.setClockDivider(SPI_CLOCK_DIV2);

Into

		_SPI.setClockDivider(SPI_CLOCK_DIV4);

And then tested the GettingStarted example and it works.

With ~200Ohm resistors added between Arduino output and NRF24 module input, all M1 and M4 does not work with TMRh20 RF24 library, and also outputs 0x06 status reg.
I think it is due to RC constant on the SPI pins.
Currently I do not have access to an oscilloscope to probe the SPI signals nor view the schematic of the M4 module. Might desolder the shielding later.

Avamander, 2015-12-26:
M2 is definetly fake, throw that away before you lose any more brain cells, clones tend to not work with originals and most likely lack features, are slower, use more power, transmit weaker or are more unreliable in terms of packet loss. M1, M3 (questionable as it seems to be a modified uncommon module) and M4 seems to be original, but do measure the thickness and compare it to the datasheet. 

Attach a fairly good power supply to the M4 and M3, they need quite a lot of juice. (Main reason M4 probably doesn't work)

Add a 100uF (or even larger) decoupling capacitor to all modules (exept M2, that one you really shouldn't use as it will cause more issues than it's worth), lower values might not work (you already have something soldered to them, but make sure it is 100uF or larger). This fixes quite a lot of problems for many.

When you've done this, take the M1 modules and try the transfer speeds again but make sure it's a continuous transmission of packets without auto ACK.

NicksonYap, 2015-12-26:
Haha, you're right about M2, they caused quite some problems.
I think some M1 are original (maybe only 2 out of 10) but hasn't caused any issues (yet).
I can't measure M4's thickness coz I soldered the shielding back. Even with the shielding removed, seems difficult to measure, haha. With the price I paid for, I'm pretty sure M4's original.

On the TX, clone Arduino Uno, using AMS1117 regulator (could be a clone regulator, again)
I check the regulators once in a while,  at TX it outputs 3.29-3.30V even when transferring at max speed: (after disabling autoACK)
DIV4, 176.89KB/s
DIV2  204.47 KB/s 
For both M1 and M4 
still only using 4.7uF SMD ceramic caps near the NRF24 module. 
SPI clock speed does affect transfer speed after all.  (I was really surprised Arduino could push it over over 200KB/s) 
In my previous attemps ( like, half a year ago I thought ~50KB/s is the max Arduino can push)
This library is efficient indeed!

I was once desperate trying to get NRF24 module to work and ended up using normal performance 100uF electrolytic cap.
For both TX side M1 and M4, large cap values doesn't seem to make a difference, maybe low ESR caps does the trick? The M4 worked well even without the 4.7uF caps. I just added it to avoid problems later on.

So conclusion is to make sure power supply maintains 3.3V, 4.7uF low esr caps are good enough?
And... DIV2 SPI clock speed does improve performance by around 15% (at transfer speed approx > 70KB/s)

So the problem is due to the design of the module?
Bad move to add 1kOhm resistors on the SPI pins, then. 

Since I've removed the 1kOhm resistors, I'll set RF24.cpp back to DIV2, 8MHz. SPI clock speed.
The best way to help others in this problem is to mention that the status register returns 0x06 and let others find this article.
Because I couldn't find an article about 0x06.
Turns out resistors in series are the problem.

Thanks, Avamander!

closed 2015-12-26 by NicksonYap

Oitzu, 2015-12-26:
@NicksonYap do you have any more perfomance reports from M4? I'm currently using the "long" PA+LNA Version without shielding and the missing shielding is really a problem. Do this works better with the M4 Version?

NicksonYap, 2015-12-27:
No,  I have just got M4 to work.  I will do some testing later on :)

NicksonYap, 2015-12-27:
Apparently, the TX did sent 204KB/s but the RX could not receive as fast.
Only received 47716/50000 (around 5% lost)
TX side marks as no failed packets (50000 sent) but RX side counts only sums up to around 47716.

So I suspect the FIFO was full and data discarded.
At RX side where counts of payload is printed every second, radio.rxFifoFull() returns 1 when data is being received and 0 when the TX stops for a moment.
```C++
  while (radio.available()) {
    radio.read(&data, 32);
    counter++;
  }
  if (millis() - rxTimer > 1000) {
    rxTimer = millis();
    unsigned long numBytes = counter * 32;
    totalCounter += counter;
    Serial.print(F("Rate: "));
    //Prevent dividing into 0, which will cause issues over a period of time
    Serial.println(numBytes > 0 ? numBytes / 1000.0 : 0);
    Serial.print(F("Payload Count: "));
    Serial.println(counter);
    Serial.print(F("Total Count: "));
    Serial.println(totalCounter);
    Serial.print(F("FIFO full? "));
    Serial.println(radio.rxFifoFull());
    counter = 0;
  }
```

Trying to make the RX faster, but the original one seem fast enough already...

NicksonYap, 2015-12-27:
adding 
```C++
    delayMicroseconds(500);
```
after

```C++
radio.writeFast(&data, 32, 1));
  ```

47KB/s , RX 49997/50000 

And rxFifoFull() returns TRUE once in a while.

That's quite terrible, half a millisec delay and still unable to keep up with the RX FIFO...

Is this due to Arduino speed limitation?

TMRh20, 2015-12-27:
@NicksonYap Placing the rxFifoFull() check before the calls to ```serial.print``` should prevent it from returning true. 50KB/s == about 1500 payloads/second, so any delays can prevent the receiver from keeping up. Calls to serial.print seem to take some time. Also note that using the multicast option as ``` radio.writeFast(&data, 32, 1)); ``` requires ```enableDynamicAck() ``` to be called, and doesn't seem to work properly on some clones.

NicksonYap, 2015-12-29:
I was able to push Arduino Uno and NRF24 up to 215KB/s, both TX and RX side. So far the fastest that can be done by NRF24L01+. Channel 125, CRC 8 bit, address width 3, 2MBPS, SPI DIV 2 but DIV 4 did not make a difference coz it was waiting for TX FIFO to make space. 

[image]
'P' is the number of packets TXed or RXed.
't' is time taken to send the 10000 packets.
'S' is rate of transfer
'T' is sum of 'P'
'fF' increments if RX_FULL (checked the same time as radio.available()); (more details later)
'F' counts the number of times "P" is not equal to 10000.

My test yesterday is that I had address width 5 so the speed was 204KB/s. I let it on for a few hours at night and covered with a box to reduce changed of cockroach or mice touching it and interesting with my test.  And NO packets lost, AT ALL.

[image]
Left is TX, right is RX.
1565ms taken to send 10000 packets in each attempt at 204KB/s.
"Total count " sums "Payload count"
fifoFullCounter increments if RX_FULL (checked the same time as radio.available()); (more details later)
"Fails" counts the number of times "Payload count" is not equal to 10000.

To achieve such speed, I removed delayMicroseconds(5); and used direct port manipulation of CSN and CE pins. digitalWrite itself had about 7us delay. Datasheet did not mention a delay needed for CSN and CE pins and the module worked well so far.

In my code, DIV4 or DIV2 did not  make a difference in transfer speed.

Will clean up my code and upload soon, I'll try to use more RF24 library functions since to optimize speed I had to copy and edit parts of RF24 lib.

NicksonYap, 2015-12-29:
TX side:
```C++

#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"

#define CEpin 7
#define CSNpin 8
/*************  USER Configuration *****************************/
// Hardware configuration
RF24 radio(CEpin, CSNpin);            // Set up nRF24L01 radio on SPI bus plus pins 7 & 8

/***************************************************************/

const uint64_t pipes[2] = { 0xABCDABCD71LL, 0x544d52687CLL };   // Radio pipe addresses for the 2 nodes to communicate.

byte data[32];                           //Data buffer for testing data transfer speeds

unsigned long maxRetriesCounter, rxTimer;          //Counter and timer for keeping track transfer info
long startMicros, stopMicros;

void setup(void) {

  Serial.begin(115200);
  printf_begin();

  radio.begin();                           // Setup and configure rf radio
  radio.setChannel(125);
  radio.setPALevel(RF24_PA_MAX);
  radio.setDataRate(RF24_2MBPS);
  radio.setAutoAck(0);                     // disable for speed
  radio.setRetries(2, 15);                 // Optionally, increase the delay between retries & # of retries

  radio.setAddressWidth(3);
  radio.setCRCLength(RF24_CRC_8);          // Use 8-bit CRC for performance

  radio.openWritingPipe(pipes[1]);
  radio.openReadingPipe(1, pipes[0]);
  radio.stopListening();

  //radio.startListening();                 // Start listening
  radio.printDetails();                   // Dump the configuration of the rf unit for debugging

  Serial.println(F("\n\rRF24/examples/Transfer/"));

  randomSeed(analogRead(0));              //Seed for random number generation
  for (int i = 1; i < 32; i++) { //leave first byte for payload for identification, (set when being transmitted)
    data[i] = random(255);               //Load the buffer with random data
  }
  radio.powerUp();                        //Power up the radio
  delay(100);
}

unsigned long num_packets = 10000; //Change this to a higher or lower number.
unsigned long total_packets = 0;

void loop(void) {

  delay(100);

  startMicros = micros();
  long pauseMicros = startMicros;

  for (unsigned long i = 0; i < num_packets; i++) {  //Loop through a number of packets
    if (i == 0) {
      data[0] = 'S';
    } else if (i < num_packets - 1) {
      data[0] = 'T';                      //Change the first byte to inditate being sent
[... 253 lines cut ...]
    fifoFullCounter = 0;
    num_packets = 0;
    // Serial.println();
  }

  if (fifoStatus & _BV(RX_FULL)) {
    fifoFullCounter++;
  }

}

void beginTransaction() {
  PORTB &= ~(1 << PB0);
 // PORTH &= ~(1 << PH5); //mega
  //  delayMicroseconds(4);
}

void endTransaction() {
  PORTB |= (1 << PB0);
//  PORTH |= (1 << PH5); //mega
  //  delayMicroseconds(4);
}
```

[TX RX up to 215KB per sec.zip](https://github.com/TMRh20/RF24/files/73927/TX.RX.up.to.215KB.per.sec.zip)

TMRh20, 2015-12-29:
Nice analysis and testing!

A couple points that may be of benefit:
- The delayMicroseconds(5) in CSN toggle was added to cope with breadboards & long wires. Testing on a range of devices seems to show the most benefit when hardware/wiring etc. is not optimal. (Category: general hackery )
- Commenting [this line](https://github.com/TMRh20/RF24/blob/Interrupts/RF24.cpp#L1081) tends to make a fair bit of difference in read speeds. The, idea was to clear interrupts after every read, so users don't necessarily need to do it manually via ```whatHappened()```   
After taking a look at your results, and mine with the single modification, I find clearing the interrupt flags may not be worth the tradeoff for simplicity??

Very interesting stuff, will have to keep an eye on this thread.

penguin007-1, 2017-06-12:
Thanks a lot! @NicksonYap, you did a wonderful job! I've got the same problem: AS01-ML01DP5. It didn't work with RF24 library and I didn't know the reason. I tried some other nRF24 libraries, it worked, but the library looks not so nice as TMRh20 RF24 library.