nRF24/RF24 #183: Only up to 126 channels, not 128.

NicksonYap, 2015-12-27:
In the NRF24L01+ datasheet, it's only up to 126 channels.

At 1MHZ channel bandwidth, (max freq - min freq) 2525 - 2400 MHz = 125 channels + 1 to include the the first channel, channel 0. Total 126 channels.
Using the Transfer example, the module can only communicate at radio.setChannel(125);

The "scanner" example scans up to channel 127, which is invalid. 
When I run the "scanner" example, I often see the last two channels (CH 126 and 127) jumping up and down while 125 down to around 100 are all zeros.

Perhaps the documentation should state that it only works from CH 0 to 125?
And change the scanner example from 128 to 126 channels.

TMRh20, 2015-12-28:
Fixed in examples and docs. Thanks for recognizing this.

NicksonYap, 2015-12-29:
https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L447

```C++
void RF24::setChannel(uint8_t channel)
{
  const uint8_t max_channel = 127;
  write_register(RF_CH,rf24_min(channel,max_channel));
}
```

I think max_channel = 125?

TMRh20, 2015-12-29:
I think so too :+1:

closed 2016-02-20 by TMRh20