nRF24/RF24 #184: Transfer example, false packet failure count.

NicksonYap, 2015-12-28:
In "Transfer" example.

RF24::write(...) and RF24::writeFast(...) returns 0 if max retries has reached and the TX FIFO is full and then clears the MAX_RT flag. Should the TX FIFO be flushed if max retried? Why is reUseTX(); commented?  Should the user should manually retry TX?

https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L896
```C++
bool RF24::writeFast( const void* buf, uint8_t len, const bool multicast )
{
...
while( ( get_status()  & ( _BV(TX_FULL) ))) {	
if( get_status() & _BV(MAX_RT)){
	//reUseTX();	//Set re-transmit
	write_register(NRF_STATUS,_BV(MAX_RT) );	 //Clear max retry flag
	return 0;	  
//Return 0. The previous payload has been retransmitted
  //From the user perspective, if you get a 0, just keep trying to send the same payload
}

...

```
I assume when TX_FULL and MAX_RT, the number of failed packets is 3? Or one? Or more than one?

Similarly for RF24::txStandby(), it returns 0 if the TX is NOT empty and max retries reached.
https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L970
```C++
RF24::txStandBy(){
...
while( ! (read_register(FIFO_STATUS) & _BV(TX_EMPTY)) ){
		if( get_status() & _BV(MAX_RT)){
			write_register(NRF_STATUS,_BV(MAX_RT) );
			ce(LOW);
			flush_tx();    //Non blocking, flush the data
			return 0;       
		}
...
```
I think that the when NOT TX_EMPTY and MAX_RT, one or more packets has failed.

I believe both txStandBy and the 'write' functions can only count the number of MAX_RT occurrences, not the number of failed packets since we cannot count how many packets left in the TX FIFO

If this is true, then then line 94 and 80 in Transfer.ino should have the same increments and only count number of MAX_RT occurrences.

Therefore the serial prints in Line 101  is misleading.

Line 94:
https://github.com/TMRh20/RF24/blob/master/examples/Transfer/Transfer.ino#L94
```C++
  if(!radio.txStandBy()){ counter+=3; } 
```

Line 80:
https://github.com/TMRh20/RF24/blob/master/examples/Transfer/Transfer.ino#L80
```C++
 if(!radio.writeFast(&data,32)){   //Write to the FIFO buffers        
        counter++;                      //Keep count of failed payloads
      }
```

Line 101:
https://github.com/TMRh20/RF24/blob/master/examples/Transfer/Transfer.ino#L101
```C+++
 Serial.print(counter); Serial.print(" of "); Serial.print(cycles); Serial.println(" Packets Failed to Send");
```

TMRh20, 2015-12-28:
The behavior is demonstrated fairly well in the transfer example, and you can easily test any modifications regarding flushing, retrys etc. Its probably easier than a long-winded explanation, but I'll give it a shot.

I guess the best way to look at it is to go through the process of sending a stream of data vs individual packets.

a: Initiate 3 calls to: ``` radio.writeFast() ``` on startup
b: If running SPI at 8mhz, the 3 FIFO buffers will be filled before the first is sent
c: Initiate a 4th call to ```radio.writeFast()```
- **Possibility 1:** Success:  What we know: If called immediately after the 3rd write, one payload has succeeded for sure. *Action* Write the next payload
- **Possibility 0:** Fail: What we know: The first payload has failed (none sent) and the current payload has failed *Action* Keep trying the current payload as required or quit. 

d: **Possibility 1:** All is well. Call ``` radio.txStandBy() ``` to finish. If it fails, n payloads have failed, where n==1,2 or 3.
    **Possibility 0:** All is not well. The stream of 4 payloads has failed. Call ```radio.txStandBy()``` If it fails, and the preceding ``` writeFast()``` call failed, at least 2 payloads failed, likely all 4. If it succeeds, only the last payload failed.

[nrf24l01 datasheet](https://www.nordicsemi.com/chi/content/download/2730/34105/file/nRF24L01_Product_Specification_v2_0.pdf)
> If the auto retransmit counter (ARC_CNT) exceeds the programmed maximum limit (ARC), the MAX_RT IRQ
is asserted. In Figure 21. the packet transmission ends with a MAX_RT IRQ. The payload in TX FIFO is
NOT removed and the MCU decides the next step in the protocol. A toggle of the CE starts a new
sequence of transmitting the same packet. The payload can be removed from the TX FIFO using the
FLUSH_TX command.

**Conclusion:** 
*Accuracy:* Only when calling radio.txStandBy() , the specific payload or number of payloads that fail or succeed cannot be identified. I used 3 because it is a simple indicator of the maximum error rate. Feel free to submit a pull request if you would like to adjust this behavior.

*Method* The current method appears to offer the most flexibility to the user in managing a reliable data stream, since all users need to do is re-attempt the current payload until success, or give up by calling txStandBy();.

[sub] Note: reUseTx() command is not needed since CE is left HIGH[/sub]

NicksonYap, 2015-12-31:
I try to keep in mind that MAX_RT only works if AutoAck is enabled. True?

I read the excerpt of the data sheet you included and this line
>  The payload in TX FIFO is NOT removed and the MCU decides the next step in the protocol.
Lets say while performing writeFast() and the TX FIFO is indeed full and found out that MAX_RT IRQ is asserted, it only masks the IRQ and then what happens to the payload that failed to send? Since data sheet mentions that the payload is not removed, It just stays there and stops transmitting the next one? 
The Transfer example does not take action if failed, only counts.
Even if writeFast() is called again, it will not resend the failed payload nor remove it, seems like the space will be occupied.
At least txStandBy() will flush the TX FIFO after clearing the MAX_RT flag.

We are assuming the FIFO is pushed (inserted) faster than it is pop'ed (removed) so it is very likely writeFast() waits for TX_FULL to clear only then it writes the TX FIFO, so the MAX_RT flag is polled very often, more often that the FIFO being written.
Then what if this is not the case where MAX_RT is not polled often enough that more than one payloads failed and MAX_RT can only represent 1 or more payloads has failed.

So we continue assuming that MAX_RT is polled very often.

In step 'c' if writeFast() returns 0 (Possiblity '0') because MAX_RT is asserted so one of the possibility is that no payload was sent as the first one maxed out the number of retries, so only 1 payload failed to send. 
If the first one did succeed, and it is the 2nd one that failed and MAX_RT asserted, again, only 1 payload failed to send.

Same question: What happens to the payload that failed to send?
I cannot continue analyzing until I know what happens to that payload that failed.

TMRh20, 2015-12-31:
> Same question: What happens to the payload that failed to send?

Nothing. FIFO stays full. User payload that fails is not written.

If you want a full understanding with this level of detail, you need to write more code and test as I do. This kind of comment seems to get negative reactions sometimes, but honestly, this is the only way I could provide you with proper info, since I can't remember everything at this level. 

Try this:

```
      if(!radio.writeFast(&data,32)){   //Write to the FIFO buffers        
        counter++;                      //Keep count of failed payloads
        i--;
      }
```
Then use ```Serial.print``` to view the payload identifier (first byte received) on the receiver, as you intermittently call ```stopListening()``` & ```startListening()```

You will notice data is never lost. Further testing can be introduced by using interrupts to count/view the flags as they happen.

NicksonYap, 2015-12-31:
I'm sorry... I'll do some more tests.
In your code snippet, what is 'i--;' for?
Other than that, it's the same as the transfer example....

I roughly understand what you want me to do, as in run a program to check for what happens to the packets in the FIFO.

By "payload identifier", do you mean PID? I couldn't find info/register on the datasheet that allows me to read the PID, other than the possibility of using manually a byte in the payload as an ID to keep track of lost packets.

Why would I intermittently call `stopListening()` & `startListening()`?
I thought it's only needed once in `void setup(){ ... }` and AutoAck handles the retransmission?
Or is it because I have to call `stopListening()` to read the flags?

Thanks~

TMRh20, 2015-12-31:
*Modification to transfer.ino example:*
```
    for(int i=0; i<cycles; i++){        //Loop through a number of cycles
      data[0] = i;                      //Change the first byte of the payload for identification
      if(!radio.writeFast(&data,32)){   //Write to the FIFO buffers        
        counter++;                      //Keep count of failed payloads
        i--; // Retry all payloads that fail
      }
```

*Explanation:*
The ``` i-- ``` decrements the i in ``` for(int i=0;i<cycles;i++)``` causing the radio to retry a new payload with the same identifier when a failure occurs.

The first byte in the payload is set to i, in the line ``` data[0]=i; ``` so it can be used to identify the payload. 

``` Why would I intermittently call stopListening() & startListening()? ``` 

To demonstrate what happens to data in the FIFO, and prevent the RX radio from receiving data. You can also grab one of the devices and run far away, then back, or use a jammer, but I find this easier. :p

closed 2016-02-20 by TMRh20