nRF24/RF24 #185: Transmission fails with RPi

aloz77, 2015-12-31:
I'm trying to make nRF24L01+ (bought at eBay/Aliexpress) working with RPi B+ V1.2. I double checked the connections and I tried different caps values (1uF-1000uF) but it always fails sending. :-( I have 4 modules from 2 different sellers and still no success. Is there a way to check why the transmission fails?

```
pi@raspberrypi:~ $ sudo RF24/examples_RPi/gettingstarted
RF24/examples/GettingStarted/
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO22
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0x5a5a5a5a5a 0x4242414141
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0x5a5a5a5a5a
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x02
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX

 ************ Role Setup ***********
Choose a role: Enter 0 for pong_back, 1 for ping_out (CTRL+C to exit)
>1
Role: Ping Out, starting transmission

Now sending...
failed.
Failed, response timed out.
Now sending...
failed.
Failed, response timed out.
```

TMRh20, 2015-12-31:
The addresses RX_ADDR_P0-1     = 0x5a5a5a5a5a 0x4242414141 are wrong for the getting started sketch, so that would infer a general problem with SPI not working.

See https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md for info on SPI & troubleshooting. If pins are connected accordingly and SPI working, with compatible modules, it should work.

aloz77, 2015-12-31:
The SPI communication seems to work all right as I changed the pipe names in gettingstarted.cpp in order to test it. So it matches the RX_ADDR_P0-1.

```
const uint8_t pipes[][6] = {"ZZZZZ","AAABB"};
```

Can the TX issue be somehow related to the 4.x kernel which is new for Raspi or is it a hardware issue?

```
pi@raspberrypi:~/RF24/examples_RPi $ uname -a
Linux raspberrypi 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015 armv6l GNU/Linux
```

I also tried changing the channel and the PAlevel with no success.

```
  radio.setChannel(0x5C);
  radio.setPALevel(RF24_PA_MIN);
```

TMRh20, 2015-12-31:
4.x kernel is ok for me: ``` Linux Pie-is-Round 4.1.15+ #830 Tue Dec 15 16:58:28 GMT 2015 armv6l GNU/Linux ```

If it can receive but not send, usually is the CE and/or CS pin confusion. RPI pins labeled CE0 and CE1 are used for radio CSN, not radio CE. With RPi, if the pins are correct, it should pretty much just work.

aloz77, 2016-01-01:
I did the wiring exactly like described at http://tmrh20.github.io/RF24/RPi.html

[image]
[image]
[image]

Is there another way to check the hardware integrity of the RF24 modules?

Happy new year and keep up your great work!

TMRh20, 2016-01-01:
The easiest way to verify is if you have an Arduino, or even another RPi, just test on multiple devices.  Running a simple sketch and calling setChannel then getChannel would also tell you if there is communication, but won't necessarily identify if hardware, or not.

There are some notes on the BCM driver page regarding RPi2:
http://www.airspayce.com/mikem/bcm2835/

```
Raspberry Pi 2 (RPI2)
For this library to work correctly on RPI2, you MUST have the device tree support enabled in the kernel. You should also ensure you are using the latest version of Linux. The library has been tested on RPI2 with 2015-02-16-raspbian-wheezy and ArchLinuxARM-rpi-2 as of 2015-03-29.

When device tree suport is enabled, the file /proc/device-tree/soc/ranges will appear in the file system, and the bcm2835 module relies on its presence to correctly run on RPI2 (it is optional for RPI1). Without device tree support enabled and the presence of this file, it will not work on RPI2.

To enable device tree support:

sudo raspi-config
 under Advanced Options - enable Device Tree
 Reboot.
```

If the device tree thing is the problem, it should probably get added to the RF24 documentation.

You can also test with SPIDEV, see http://tmrh20.github.io/RF24/RPi.html for how to compile with SPIDEV.

aloz77, 2016-01-01:
Hmm, tried the RF24 modules with Arduino Uno. The same TX failure. Max retries seem to get exceeded here in RF24:: write().

```
  //Max retries exceeded
  if( status & _BV(MAX_RT)){
  	flush_tx(); //Only going to be 1 packet int the FIFO at a time using this method, so just flush
  	return 0;
  }
```

However the setChannel() test works as expected:

```
#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"

RF24 radio(7,8);

int channel=1;

void setup() {
  Serial.begin(115200);
  radio.begin();
}

void loop() {
  radio.setChannel(channel);
  delay(100);
  Serial.print("Channel set ");
  Serial.print(channel);
  Serial.print(", get ");
  Serial.println(radio.getChannel());
  channel++;
  delay(1000);
}
```

Does it look like a hardware issue?

BTW: I don't have an RPi2, mine is just B+.

TMRh20, 2016-01-01:
Something is up, thats for sure. See https://github.com/TMRh20/RF24/issues/182#issuecomment-167337905 for some good info on different modules.

*"BTW: I don't have an RPi2"*
Oops, my bad, its probably good to read things.

danielamkaer, 2016-01-01:
I'm also experiencing some weirdness. I've setup an RPi b+ and an Arduino Pro mini (3 V, 8 MHz).
Using the GettingStarted examples, the RPi receives payloads, but the Arduino never receives a response. I checked the registers, and they seem to be correctly filled.

I hope to get your RF24Network/Gateway code up and running.

aloz77, 2016-01-02:
I give up. My RF24 modules don't run even with 10uF X7R ceramic caps both with Arduino Uno R3 and with Raspi B+.

Avamander, 2016-01-02:
Try 100+uF electrolytic just as a test.

aloz77, 2016-01-02:
I already tried 1uF, 4.7 uF, 10 uF, 100 uF and 1000 uF. As well as separate 3.3 power supply and other Arduino Uno. As well as all other comment hints like powerUp and "antenna capacitor" available at http://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-HowTo. Nothing helps on my 4 modules (all looking like M1 according to #182 )

Avamander, 2016-01-02:
Then there are three things.

a) You code is wrong (either installation issue or you have defined the
wrong pins)

b) Your modules are broken

c) You have wired them incorrectly

The second case is the least likely.

aloz77, 2016-01-02:
Hmm... Wrong code and/or wiring both on Arduino and Raspi? The module communication works as you can see in my first post. I rather think now that the modules are broken. Or it's a kind of new series which need some new hardware/software workaround.

Avamander, 2016-01-02:
How are you connecting the modules to Arduinos?

aloz77, 2016-01-03:
[image]

Avamander, 2016-01-07:
Okay, but with what? Ribbon cable?

aloz77, 2016-01-07:
Hmm... Today I've got a new mini RF24 module like this.

[image]

I soldered 10 cm dupont wires, added 0.1uF and 100uF caps and connected it to a fresh and almoust unused Arduino Uno. 

[image]

And: Exactly the same thing happens as with all other (usual style) RF24 modules I own. Module status is displayed ok, but radio.write() fails 100% with MAX_RT flag set. :-(

Avamander, 2016-01-07:
a) The cap should be as near as possible to the NRF module. Solder it
straight to the power pins.

b) WIpe your Arduino IDE and RasPi install and download everything again.

aloz77, 2016-01-09:
I gave it a last try. I soldered 100 uF directly to the RF24 mini board and setup a new Arduino IDE 1.6.7 on another PC. Installed the last RF24 lib through the Library Manager of the IDE. GettingStarted.ino fails the same way on radio.write() while printDetails() gets good values. Something is going wrong. I give up.

[image]

Oitzu, 2016-01-09:
The problem are most probably due to wrong connection of the modules.. maybe you should divide a drawing or a fritzing sketch how you connected it. It isn't clearly to see on the pictures.

jpdw, 2016-01-21:
Hi.  

I hope you've not given up completely yet....

Over the past 2-3 weeks I was having similar issues getting an "arduino-like" board and a RPi talking well with these modules.  Mine are now working.  Some suggestions in case they are of any use in your efforts:

1/
I switched "down" to Arduino <-> Arduino to simplify things - allows to validate the nRF24L01s are ok, eliminates any possible Linux OS or BCM driver problem and means you are running the same code on both Tx & Rx boards.  (What you've described implies you have more than 1 arduino).

2/
Check & re-check cabling - in found I had a slightly loose cable pushed onto either the CE or CSN pin - that alone cost me 3-4 hours of frustrated debugging.

3/
Use the simplest of the examples to start with -- eliminate any other code you eventually want to have around the radios

4/
If you've got 3 radios... then once you've got Arduino-Arduino working, keep these two and put the 3rd radio on the RPi.  Try to run the same example on Arduino-RPi - but you'll still have the ability to drop back to an Arduino-Arduino setup (without rewiring anything) to validate things if the RPi isn't working.

I basically abandoned the RPi till I'd got 1 Arduino talking to 2 others on different "pipes", then switched the RPi back in as the "master"/PRX knowing I the PTX ends were tested fine (and I didn't change the code on them again till after the RPi was stable as the PRX.

5/
Depending on your knowledge / inclination - consider using the Python library on the RPi.  Whilst this *adds* a layer of complexity, I found it quicker/easier as I'm more familiar with the toolchain for python (just run it !) than building C/CPP on a pc-like device.  It also meant I didn't keep tripping myself up with string pointer issues.... (which says something about my rusty C skills...)

closed 2016-03-23 by TMRh20

randym52, 2016-05-02:
I realize this is closed, but, I just had a similar problem.  I am using a pi to replace an existing Arduino unit in a multi point wireless network.  printDetails() displayed perfectly.  I was getting RX data. But transmit failed every time.

The cause was different setup on the nRF24's.   The existing Arduino units were using crc16.  
The Pi unit was set to crc8.   Danged cut & paste.

MAKOMO, 2017-04-16:
Just want to report a similar issue on my RPi 2. Default installation from installer script. The gettingstarted script sending from the RPi 2 to an Arduino node showed random write failures although all data arrived on the Arduino node (the other direction, Arduino sending out and RPi receiving worked flawless). By accident (after hours of messing around) I tried the exact same software (running from the same SD card) on an RPi 3 (using the exact same nRF module setup). There the gettingstarted script worked without any error. 

It turned out that I could resolve the write errors on the RPi 2 by switching from the default BCM2835 driver to the SPIDEV, recompiling the RF24 lib via

```
sudo ./configure --driver=SPIDEV
sudo make install -B
```

Nothing else changed. So it turned out that those write errors were somehow caused by an issue in the SPI communication.

HannHank, 2018-01-02:
Hey , on my github I solved the biggest Problems of the RF24