nRF24/RF24 #186: which mode nrf24l01+ is in when in pingpair_irq.ino

shirish47, 2016-01-02:
Well I wanted to check out ur library for use in Low power designs. I was testing it out with a pro mini and my setup works. I am trying to check current consumption of nrf24 module using a multimeter between its VCC and my pro mini is in receiver role. while the mini waits for IRQ I stop the sender so I can check the current consumption when nothing is happening and I see that nrf24l01+ module (No LNA PA ,simple modules) is eating up 19.5mA!! 
how to get it to standby mode or any other mode such that nrf24 can IRQ on getting data but consumes less current (uA) other time? 
I actually want to know the mode it is in this example after loop().

Avamander, 2016-01-02:
You are measuring the power consumption wrong. It should never go over 14mA.

shirish47, 2016-01-02:
on pingpair sleepy nrf24 still consumes ~17mA while when I remove arduino and just use power supply to nrf24 it still consumes ~2mA. seems like modules are not designed for low power.
14 mA is minimum nef24 takes even if sleeping or not . I m not sure if the library is making it go to sleep or not. :( I thought there would be current change. but nothing seems to happen with current consumption..

Avamander, 2016-01-02:
Your modules are either fake or you have a modified version. Nordic
Semiconductors say that a NRF24L01+ consumes 14mA peak and sub μA when
powered down and this has been confirmed multiple times.

shirish47, 2016-01-02:
possible.. I think I am going to make other circuit with arduino promini like 3.3 volts and measure overall consumption.

Avamander, 2016-01-02:
You said that your simple NRF module is drawing 195mA, that is 100% wrong
reading for any type of module you might have. Check again.

shirish47, 2016-01-02:
19.5mA I don't know first it used to, wait for some time I am making another test circuit.

shirish47, 2016-01-02:
another arduino promini 3.3 volt.
 pingpair sleepy  Receiver does not sleep go to stanby as current consumption remains same after arduino sleeps. 14.7mAs...

shirish47, 2016-01-02:
I am not sure if the module is going in standby or not.... while the arduino sleeps code does not seem to do anything related to standby mode...
here is the example code with my pin Modifications it receives and reply too so module is comunicating with arduino and other modules over air. https://gist.github.com/shirish47/e48284eb0c6ff5593935

TMRh20, 2016-01-02:
Radio Modes & Current Consumption by function, when run in this order:

| Function |  Radio Mode | Current Consumption |
|-------------|-----------------|-----------------------------|
| radio.begin();              | Standby-I | .026mA
| radio.startListening(); | Active       | 11-15mA
| delay(1000);                 | Active       | 11-15mA
| radio.stopListening(); | Standby-I | .026mA
| radio.write();               | Active      | 11-15mA
| delay(1000);                 | Standby-I | .026mA
| radio.writeFast()         | Active       | 11-15mA
| delay(1000);                 | Standby-II | 0.320mA
| radio.powerDown();    | PowerDown | .0009mA

shirish47, 2016-01-02:
ok so I can't listen when I want to maintain amperes in uA ? :(
I thought nrf24 would stay in standby mode and when the RX fifo is full it will interrupt on IRQ. 
Is it possible however to make NRF24 to listen to any incoming msgs and Interrupt controller also keeping low uA.

TMRh20, 2016-01-03:
Nope. Ack payloads are a close second.

shirish47, 2016-01-04:
thank you so much. I think what I am looking for is called Wake on Radio (WOR functionality ) which is not available in NRF24l01+.

closed 2016-01-26 by Avamander