nRF24/RF24 #187: Image transfer with 32byte packages

WERAQS, 2016-01-03:
Been experimenting with STM32 and cameras, do you think it is possible to transmit images over RF24 network?

TMRh20, 2016-01-03:
Nothing is impossible

See https://github.com/TMRh20/RF24/issues/172#issuecomment-166370029
and
https://gist.github.com/TMRh20

<sub>Note: I apologize in advance for the following :p</sub>

*Network lib network lib
Does whatever a network can
Routes all packets
any size
caches data
just like flies
LOOK OUT!
Yeah its the network lib...*

Oitzu, 2016-01-03:
@TMRh20 oh god why :cold_sweat: 

@WERAQS its indeed possible, i'm actually doing it right now. ;)
But i would recommend, if you can, bigger packages of 144bytes (or anything other multiple of 24 bytes). 
It is more perfomant.

iamthemovie, 2016-01-03:
@TMRh20 ... wow ...

WERAQS, 2016-01-04:
ah I see, its the spider pig. thanks, I'll try to transmit using 144bytes.

closed 2016-02-20 by TMRh20