nRF24/RF24 #188: ESP8266 Arduino v2.0 not working

sebastius, 2016-01-03:
I can't seem to get your code working on an ESP8266. Works fine on a regular Atmega 328 Arduino.

Compiles okay although i get errors like:

RF24-master/RF24_config.h:119:0: warning: "PSTR" redefined [enabled by default]

In file included from /Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266/Arduino.h:242:0,
                 from /Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/SPI/SPI.h:24,
                 from GettingStarted.ino:8:
/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266/pgmspace.h:18:0: note: this is the location of the previous definition
 #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))

and lots more.

The ESP boots but when i call the printDetails(); i only get this:

STATUS		 = 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	 = 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	 = 0xff 0xff 0xff 0xff
TX_ADDR		 = 0xffffffffff
RX_PW_P0-6	 = 0xff 0xff 0xff 0xff 0xff 0xff
EN_AA		 = 0xff
EN_RXADDR	 = 0xff
RF_CH		 = 0xff
RF_SETUP	 = 0xff
CONFIG		 = 0xff
DYNPD/FEATURE	 = 0xff 0xff
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = 16 bits
PA Power	 = PA_MAX

Any hints on what i should do/try?

Avamander, 2016-01-04:
Start with the hardware.

Is the wiring absolutely 1000% correct?

Are the wires longer or somehow different when the Nrf24l01 was connected
to a Nano/Uno?

Have you tried a 100+uF electrolytic capacitor on the 3.3V and GND pins on
the Nrf24l01?

sebastius, 2016-01-04:
I also get lots and lots of trouble from PrintF. It won't compile unless i remove/change all printf commands. 

libraries/RF24-master/printf.h:31:29: error: 'fdevopen' was not declared in this scope

Wires are the same, wiring is ok.

TMRh20, 2016-01-05:
The warnings are because RF24 disables PROGMEM functionality by default, since I cannot get printDetails working properly with it enabled.

ESP support is tested using the [GIT version](https://github.com/esp8266/Arduino#using-git-version-)

The printDetails() output indicates basic SPI communication is not working.

Wiring should be as per http://www.mysensors.org/build/esp8266_gateway

With pins as shown above, use RF24 constructor:
``` RF24 radio(2,15); ```

Don't ```#include <printf.h>``` and there should be no problems. ( It is enabled by default )

wmarkow, 2016-02-11:
@TMRh20, shouldn't it be:
`
RF24 radio(4,15)
`
As I see in ESP8266 Arduino core, pin D2 (chip enable) is mapped to Arduino pin number 4.

chaeplin, 2016-02-17:
@wmarkow I use RF24 radio(3, 15). CE is any DOUT.

wmarkow, 2016-02-21:
@chaeplin, I know that you can use any DOUT as a CE. My previous comment has been written in context of this http://www.mysensors.org/build/esp8266_gateway post. They are using D2 pin as a CE and D8 as CS which - as I see in the [NodeMCU ESP8266 core system](https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h) - gos down to the constructor:

```
RF24 radio(4, 15);
```

BTW, in my NodeMCU I'm using D4 as a CE and it works fine as well.

zgoda, 2016-08-29:
I'm getting ton of warnings related to pgmspace use when compiling for NodeMCUv2 board:

```
In file included from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:0:
/home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:124:0: warning: "PSTR" redefined [enabled by default]
#define PSTR(x) (x)
^
In file included from /home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/Arduino.h:240:0,
from /home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:63,
from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:
/home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/pgmspace.h:21:0: note: this is the location of the previous definition
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
^
In file included from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:0:
/home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:128:0: warning: "strlen_P" redefined [enabled by default]
#define strlen_P strlen
^
In file included from /home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/Arduino.h:240:0,
from /home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:63,
from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:
/home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/pgmspace.h:73:0: note: this is the location of the previous definition
#define strlen_P(strP)          strnlen_P((strP), SIZE_IRRELEVANT)
^
In file included from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:0:
/home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:129:0: warning: "PROGMEM" redefined [enabled by default]
#define PROGMEM
^
In file included from /home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/Arduino.h:240:0,
from /home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:63,
from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:
/home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/pgmspace.h:18:0: note: this is the location of the previous definition
#define PROGMEM     ICACHE_RODATA_ATTR
^
In file included from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:0:
/home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:130:0: warning: "pgm_read_byte" redefined [enabled by default]
#define pgm_read_byte(addr) (*(const unsigned char *)(addr))
^
In file included from /home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/Arduino.h:240:0,
from /home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:63,
from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:
/home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/pgmspace.h:90:0: note: this is the location of the previous definition
#define pgm_read_byte(addr)                                              \
^
In file included from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:0:
/home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:131:0: warning: "pgm_read_word" redefined [enabled by default]
#define pgm_read_word(p) (*(p))
^
In file included from /home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/Arduino.h:240:0,
from /home/jazg/.platformio/lib/RF24_ID433/RF24_config.h:63,
from /home/jazg/.platformio/lib/RF24_ID433/RF24.cpp:10:
/home/jazg/.platformio/packages/framework-arduinoespressif/cores/esp8266/pgmspace.h:99:0: note: this is the location of the previous definition
#define pgm_read_word(addr)                                              \
^
```

ESP8266 core for Arduino implements these already in `pgmspace.h`.

sidoh, 2017-07-24:
From what I can tell in RF24_config.h, PROGMEM is disabled for ESP8266. The macros are all essentially no-ops. I'm guessing that PROGMEM wasn't properly implemented for ESP8266 Arduino at the time this was written.

I just opened #375 to reverse this. Running my project with this change and it works like a charm. I'm enjoying my KBs of freed RAM :)

closed 2019-05-10 by TMRh20