nRF24/RF24 #189: radio.printDetails()

Avamander, 2016-01-09:
I just got a few new modules and I discovered that radio.printDetails works only with pingpair_ack example. Pingpair_dyn for example doesn't print anything.

TMRh20, 2016-01-10:
Per #130 it does need to be enabled. The examples aren't very consistent with printf/printDetails() inclusion, but I don't necessarily think it should be enabled by default in every sketch, because it uses extra memory & program space.
The printDetails() functionality has bugged me since day 1, but converting it to use Serial.print instead of printf, but it complicates portability. I suppose it could be detected, but that adds more complexity to a relatively simple thing.
 I'm up for any suggestions or changes to clean it up or explain the functionality a bit better somehow.

Avamander, 2016-01-10:
Let's create a new .print function and then include it with RF24? That way the platform could be detected and correct printing method selected without cluttering the examples itself?

TMRh20, 2016-01-10:
A couple things to consider, if we want to maintain backwards compatibility (to avoid further confusion & difficulty) we would still have to maintain/keep the old printDetails() code in some form or make the new version compatible. Device detection has been a bit difficult without actually having a wide range of hardware to test on, so that can be a bit tricky. Other than that, I can't think of too many problems with the idea other than having to code & maintain it, but if it can be done nicely, maybe its worth doing?

2bndy5, 2020-09-14:
I would be in favor of a `registerDump()` function that uses a tailored `typdef`' object to return all the available registers' contents.

closed 2020-09-19 by Avamander