nRF24/RF24 #190: project instant messenger, lot of problem

joihn, 2016-01-10:
I'm trying to create an instant messenger using 2 arduino and NRF24L01 chips.
here's the problem with my messenger : most of the time, the receiver doesn't receive the message, and sometime he receive it but everything is mixed up, like a bad encoding error...

here is the code: http://pastebin.com/nEXzCq2w
(for one of the board, you have to change the number on line 10)

ps : if I try the "GettingStarted_HandlingData", everything work fine, so the problem isn't my arduinos...

TMRh20, 2016-01-10:
> most of the time, the receiver doesn't receive the message, and sometime he receive it but everything is mixed up, like a bad encoding error...

Good call.

The first thing to maybe do is print the size of your string, but try this also:
```
Serial.print("Size : ");
Serial.println(message.length()); 
```

You will note that string objects are handled much differently than simple character arrays & such. As always, see the [documentation](https://www.arduino.cc/en/Reference/StringObject)

The next apparent problem is the lack of dynamic payloads, which would let the receiver know how many characters were sent. See the included examples & documentation.

Depending on what your requirements & goals are, the RF24Network or RF24Mesh libraries may make implementing this type of thing a bit simpler, since they include features like fragmentation & reassembly.

joihn, 2016-01-10:
okay thanks,  I have solved most of my issue by using "char array" instead of "String" :)

you can see the code here if you want http://pastebin.com/Lcsqredx

the last problem is that maxim lenght of my transmitted message is 32 caracter (32 bytes I assume).
Is 32 bytes the maximum payload ? is there a way to transmit more than 32 bytes ?

thanks for your help

closed 2016-02-20 by TMRh20