nRF24/RF24 #194: RF24 without sudo / root access?

n8henrie, 2016-01-15:
I am interested in using the RF24 library on my Raspberry Pi without running as root / sudo, especially since I have a number of other processes (webapps, home automation daemons, etc.) that I'd love to be able to trigger sending or receiving through my NRF24L01+.

Even though I hardly know any c++, I've tried editing BCM2835 to use [the new `/dev/gpiomem` interface](https://github.com/raspberrypi/linux/pull/1112/commits) instead of [`/dev/mem`](https://github.com/TMRh20/RF24/blob/master/utility/RPi/bcm2835.c#L1311), and my efforts over a couple weeks have been totally fruitless. (see also: https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/bcm2835/ThGy-PVslE8/9coKpQTlCgAJ). I'm guessing that it may be a wasted effort because RF24 requires SPI, which is not supported by `/dev/gpiomem`. Can anybody tell me if this is the case?

If so, is there another good way to accomplish this? Perhaps by somehow using the [wiringPi SPI library](https://projects.drogon.net/raspberry-pi/wiringpi/spi-library/)?

Or should I just do some setuid magic and give permissions to the daemon user in question? Any other suggestions? Thanks in advance.

pmochocki, 2016-01-16:
Is this what you are looking for: [Driving LED stripes as non-root user](http://raspberrypi.stackexchange.com/questions/26278/driving-led-stripes-as-non-root-user)?

Best Regards
Piotr

n8henrie, 2016-01-19:
Thanks for the response -- 

Thanks for the response. I had tried that months ago and didn't have any luck. I just tried again and *still* wasn't having any luck and was in the process of replying as such, and I just realized that I never logged out and back in again after adding myself to the `spi` group. Doh!

Seems to be working great now.

If any others come across this thread, my steps on a Raspberry Pi 2 were:

1. Edit `RF24/utility/BBB/spi.cpp` and install RF24 for `spidev` use as per the existing [TMRh20 Docs](http://tmrh20.github.io/RF24/RPi.html)
1. Edit `/boot/config.txt` and make sure `dtparam=spi=on` is present and not commented
1. Add your user to the `spi` group (requires logout to take effect): `sudo usermod -a -G spi $(whoami)`
1. Reboot

If having issues, you should be able to verify permissions on your spi devices and your group membership:

```
$ ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Jan 18 13:20 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Jan 18 13:20 /dev/spidev0.1

$  grep spi /etc/group
spi: # Look for your username in this line
```

closed 2016-01-19 by n8henrie

dveltool, 2016-05-16:
hello mz-fuzzy,

I followed your guidance about RF24 with non root, but still out any luck (http://quick2wire.com/non-root-access-to-spi-on-the-pi/).

When I run the program as non root account, yield "Segmentation fault".
Do you have any idea? Pls help.

/Dedy Yasriady

yasriady, 2016-05-16:
hello mz-fuzzy,

I followed your guidance about RF24 with non root, but still out any luck (http://quick2wire.com/non-root-access-to-spi-on-the-pi/).

When I run the program as non root account, yield "Segmentation fault".
Do you have any idea? Pls help.

/Dedy Yasriady

noeldum, 2017-03-31:
Also getting segfault if not using sudo. 

Thanks for the steps n8henrie. What should be edited exactly in spi.cpp?
I compiled RF24 with
```
./configure --driver=SPIDEV
sudo make install -B
```

noeldum, 2017-04-04:
Going back to this one again. I beleive I have all it needs for allowing my pi user to use the spi. Still getting segfault however.

```
pi@raspberrypi:~ $ ls -l /dev/spidev0.*
crw-rw---- 1 root spi 153, 0 Apr  4 14:17 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Apr  4 14:17 /dev/spidev0.1
pi@raspberrypi:~ $ grep spi /etc/group
spi:x:999:pi
```

I strongly suspect the issue is maybe somewhere else than just spi device permissions. I tried even making pi the owner of the /dev/spidev0.*.

Any ideas?

n8henrie, 2017-04-05:
Have you logged in and logged out since adding yourself to the `spi` group? That was my issue.

noeldum, 2017-04-06:
Thanks Henrie. Yes I did restart the rpi many times. No luck. I don't get it!

without sudo the program runs up to when it meets that line in spi.cpp

`22        setDataMode(settings.dmode);
`

Not sure where to look now. My os is Raspbian Jessie.

n8henrie, 2017-04-10:
Can you provide more context and information?

- The script you're trying to run
- The exact command you're running

svenbw, 2017-04-16:
On my platform the segmentation fault is thrown by vsprintf. Stepping into the code shows that the error occurs when writing to /sys/class/gpio if there are no user rights to that folder.
You can create a gpio group, add yourself to that group and modify the rights of that folder. That should solve it.