nRF24/RF24 #195: pingpair_sleepy.ino question

pmochocki, 2016-01-16:
Hi,
The first time using your library - first of all I wanted to thank you for all your effort.

I have a question regarding the pingpair_sleepy example.
There is a comment:
```
 TMRh20 2014 - Updates to the library allow sleeping both in TX and RX modes:
      TX Mode: The radio can be powered down (.9uA current) and the Arduino slept using the watchdog timer
      RX Mode: The radio can be left in standby mode (22uA current) and the Arduino slept using an interrupt pin
```

I went trough the code of this example and the library and TX behaviors exactly as descried. But it is unclear for me what you mean by:
> "The radio can be left in standby mode (22uA current)". 

In the pingpair_sleepy example for the Pong node we leave the radio in 
> "startListening()".

I went trough the library documentation and found:
> After calling startListening(), a basic radio will consume about 13.5mA at max PA level. During active transmission, the radio will consume about 11.5mA, but this will be reduced to 26uA (.026mA) between sending. In full powerDown mode, the radio will consume approximately 900nA (.0009mA)

But I have a doubt this is the case. In the [nRF24L01+ Product Specification](https://www.nordicsemi.com/eng/nordic/download_resource/8765/2/15376264) on page 22 on Figure 4 the diagram states that CE needs to be 0 to switch to Standby-I mode.
I think this in not the case. I looked at:
> void RF24::startListening(void)

and it holds CE high all the time. Do I assume correctly that in this example the Pong node consumes about 11.5mA all the time?

Best Regards
Piotr

TMRh20, 2016-01-20:
> I went trough the code of this example and the library and TX behaviors exactly as descried. But it is unclear for me what you mean...

Unfortunately, it is unclear to me too. Really that comment of mine makes no sense :question:  Needs to be changed.

Radio Modes & Current Consumption by function, when run in this order:

| Function |  Radio Mode | Current Consumption |
|-------------|-----------------|-----------------------------|
| radio.begin();              | Standby-I | .026mA
| radio.startListening(); | Active       | 11-15mA
| delay(1000);                 | Active       | 11-15mA
| radio.stopListening(); | Standby-I | .026mA
| radio.write();               | Active      | 11-15mA
| delay(1000);                 | Standby-I | .026mA
| radio.writeFast()         | Active       | 11-15mA
| delay(1000);                 | Standby-II | 0.320mA
| radio.powerDown();    | PowerDown | .0009mA

> Do I assume correctly that in this example the Pong node consumes about 11.5mA all the time?

Yeah, pretty much. Slight variation when sending.

pmochocki, 2016-01-20:
"A picture is worth a thousand words"
This is true for a table, too. I would like to encourage you to put this table somewhere in your documentation.
Right now I am reviewing your documentation and examples. I am looking for a high level overview of your library. Especially about auto-ack, auto-retry, dynamic payloads and it seems that the examples are the best source. 
I played with the code only one day, and maybe the usage of the lib is really easy, but a high level overview would make the ramp up easier.

Once again a big thanks for this lib - it makes the playing with RF24 modules a pleasure.

TMRh20, 2016-01-23:
Re-open pending fixes/updates

ortegafernando, 2017-06-20:
So there is no option to save power in RX mode, isn`t it ?

Regards,

wmarkow, 2017-08-16:
I took a look at the chip's manual. In the **5.1 Power consumption** section it shows that the chip in RX mode can take up to 13.5 mA. In the TX mode it consumes up to 11.3 mA. On the other hand there is a diagram available in **1.2 Block diagram** section. The transmitter uses the **PA (Power Amplifier)** which consumes the power while transmitting the data. It is possible to set its gain by writing some value to chip's specific register. The RX mode uses something called **LNA (Low Noise Amplifier)** which - I suspect - consumes some power as well. I think there is no software option to have an impact on **LNA**. In RX mode the receiver constantly demodulates the signal to search a valid incoming packet.

I would say then that it is not possible to save much power in RX mode.

closed 2019-05-10 by TMRh20