nRF24/RF24 #196: python module memory issue with RF24.read() calls ??

jpdw, 2016-01-21:
I've been using the python module on the RPi successfully for a couple of weeks now, acting as PRX to 2 remotes as PTX's.  I've had no issues with the python module at all - always my stupid code!!

Anyway, over the past few days I've started to get the code bail out with:
*** glibc detected *** python: free(): invalid next size (fast): 0x01249738 ***
Google suggests glibc has caught some sort of buffer overrun

I've tracked the trigger down in my code to a call to
 `receive_payload = radio.read(length)`

which in the python module appears to create a buffer then pass it into the C++ RF24.read() method.

As this has only started happening the past few days, I think it might have started after I'd updated the code on one of the remote (arduino-like) nodes.  But whatever the trigger, I don't yet see why it should blow up the receiver !

I just wondered if anyone else (reading this) had seen anything similar with the Python module?

jpdw, 2016-01-22:
A few more hours of hunting and I think I've found the issue.  'length' was being set to the return value of `radio.getDynamicPayloadSize()` and on a small number of occasions this was returning `0`, so the C++ wrapper was trying to allocate zero bytes.  Oddly, when I added to my code `if length == 0: length = 32` to forcably avoid the problem, the resulting payload read was exactly as would be expect (i.e. not zero bytes long at all).

For now, I've put some additional debugging around it (if getDynamicPayloadSize() returns zero, override to 32, get the data and log it).  I'll come back to it another day...

Avamander, 2016-01-22:
I think it'd be better if would be fixed in the library if possible. I'll reopen this for further investigation.

jpdw, 2016-01-22:
Yes - I thought further about it - whilst it was my 'error' to allow setting zero byte payload by not checking the value passed in, it would be useful if the library checked for this and handled it.  

A test for `maxlen` being =1 or -1 in `RF24/pyRF24.cpp/read_wrap` would have prevented the library from quitting.

I've yet to work out why I got a 0 payload size from `getDynamicPayloadSize()` though.  I've not yet noticed it again since adding the code to handle it - so could have been anything from shoddy code on the PTX to either the PTX or PRX radio having got into an unusual state.

TMRh20, 2016-01-23:
@jpdw There is a comment in one of the [datasheets for nrf24l01](http://www.nordicsemi.com/eng/content/download/2726/34069/file/nRF24L01P_Product_Specification_1_0.pdf) regarding dynamicPayloadsize:

> With the DPL feature the nRF24L01+ can decode the payload length of the received packet automatically
instead of using the RX_PW_Px registers. The MCU can read the length of the received payload by using
the R_RX_PL_WID command.
Note: Always check if the packet width reported is 32 bytes or shorter when using the
R_RX_PL_WID command. If its width is longer than 32 bytes then the packet contains errors
and must be discarded. Discard the packet by using the Flush_RX command.

When/if this happens, the driver [drops all payloads reported as >32 bytes](https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L1042) and returns 0 for length.

mz-fuzzy, 2016-01-24:
The error is not caused by wrapper code itself. It works ok for maxlen=0 (memory allocation is done for maxlen+1). The error raised in wrapper's free call is caused by improper handling during RF24::read which causes memory corruption when len=0.

In particular, as far as I was able to analyze, the problem could be in **RF24::read_payload** writing into unallocated memory:

```
while ( --data_len ) // Decrement before to skip 1st status byte
    *current++ = *prx++;
```
which overwrites 255 bytes in case that data_len=0 I guess.

In general, the python wrapper is intended only to wrap, not to include any RF24 related functionality including boundary checks. So basically it behaves the same as you would call C++ library code, where you could see the same problem.

jpdw, 2016-01-26:
> When/if this happens, the driver drops all payloads reported as >32 bytes and returns 0 for length.
I hadn't looked closely enough at the driver code to spot why it would assigned 0 bytes - I had just (for now) put in code to handle getting 0.  It doesn't happen very often, fortunately.  Suggests that it could have been caused by interference or some such as most of the time the modules seem to "just work" (a testament to great library... thank you!)

closed 2016-03-23 by TMRh20