nRF24/RF24 #198: XMega support

akatran, 2016-01-25:
I'm trying to bring the **RF24** library into the **XMEGA** family of micro-Controllers.

Currently working with an **ATXMEGA256D3**.

Because the development IDE is **Atmel Studio** (*version 7*) I don't know how to make this code a library for my solution and link to it. 
Instead I put your code inside my solution.

So far I'm able to send and receive data with success, even with the **RF24Network** library!

Is there any way to include the changes I've made to your code into the mainstream development?

I will also provide the appropriate */utility* folder.

Avamander, 2016-01-25:
You are able to create a pull request but it will only be merged if it does
not break compatibility with any other supported platform or IDE.

akatran, 2016-01-26:
I will test it on my Raspberry then.

akatran, 2016-01-26:
OK, I've tested and it is **OK** on **Raspberry PI**. 

I'll try to send you a pull request!

closed 2016-02-20 by Avamander