nRF24/RF24 #201: Why appending blank to write_payload and read_payload?

eos1d3, 2016-01-28:
If data_len is not equal to payload_size, it is the programmer's faults. The library should not sending unwanted data. So there is no need to add blank data to meet the payload_size.

And if data_len is longer than payload_size, this will create negative blank_len too.

ps: The original from maniacbug is extremely buggy. It even used nRF24LE01 power up delay 
for nRF24L01. So the write operation never succeed with just 150us delay (in nRF24LE01 datasheet) . It is 1.5ms instead.

TMRh20, 2016-02-20:
Good question. I remember messing around with this a long long time ago, but cannot remember why it was left as-is. I would say this would probably just need a bit of testing to ensure functionality and compatibility with old code.

TMRh20, 2016-11-17:
I had a chance to look at this finally and it works like this:

The radio expects ``` write/read length == payload_size ? OK : Error```

1. ```if( data_len == payload_size ){ "No unwanted data will be added"; } ```
2. ```if( data_size < payload_size ){ "You **must** manually add blank data"; } ```
3. ```if( data_len > payload_size ) { data_len = payload_size; } ```

**Summary:**
Removing added blanks adds huge complexity to otherwise simple sketches, especially if using variable length payloads, since payload_size cannot be changed on-the-fly.

 If this is a concern, ensure that ``` data_len == payload size || enableDynamicPayloads(); ```

closed 2016-11-17 by TMRh20