nRF24/RF24 #204: Devices Supported

shan-94, 2016-01-31:
HI TMRh20,
I am trying to use your RF24 library to communicate between Raspi 2 B+ and FRDM KL46z. I have got the getting started example working on the pi but haven't got the comms working yet. I am not sure whether your library will allow me to communicate with KL46z. Can you give me some hints. Thank you

Oitzu, 2016-01-31:
Do you have more infos about the FRDM KL46z?
Can you verify that i correctly communicate with the nrf24l01+ module?

shan-94, 2016-01-31:
http://www.nxp.com/products/software-and-tools/hardware-development-tools/freedom-development-boards/freedom-development-platform-for-kinetis-kl3x-and-kl4x-mcus:FRDM-KL46Z
I am currenly using mBed to program the KL46z. There is a library for the nrf24L01+ on mbed: 
https://developer.mbed.org/users/Owen/code/nRF24L01P/ 
I am able to use the nrf's on KL46z. But unable to solve issues while communicating between raspi and KL46z

Oitzu, 2016-01-31:
Do you checked if the registers are correctly set?
CRC, Datarate and frequency need to be the same on both machines, also the tx and rx addresses need to be set correctly.

When the RPi is the sending site, check if sending was successful.

shan-94, 2016-02-01:
The CRC, Datarate and Frequency are matching
I am having issues with address assignment.

Oitzu, 2016-02-01:
Any way we can help you with that?

shan-94, 2016-02-01:
I am using this code on the KL46z : https://developer.mbed.org/compiler/#nav:/nRF24L01P_TX/main.cpp;

I have tested the nrf's on Raspberry pi and they are working fine. The address assignment problem is solved.

Oitzu, 2016-02-01:
Hm.. Login or Signup needed. ;) Would be great if you could post a gist, or something like that, for other people that maybe have the same problem.

Great that you got it working. :+1:

shan-94, 2016-02-01:
I have only got the nrf24l01+ working independently on the two boards.. I haven't got the radio comms working yet.. My last reply was a bit vague ..Sorry

Oitzu, 2016-02-01:
Oh okay, maybe you can share your code of both sides so that we maybe can spot whats the problem?

shan-94, 2016-02-01:
https://developer.mbed.org/users/Owen/code/nRF24L01P_Hello_World/ - This is the code & library for KL46z

For the Pi I am using the GettingStarted example under the examples_RPi folder of this repository

Oitzu, 2016-02-01:
Can you also post your output of the test? We really have no KL46z laying arround to test that. ;)

shan-94, 2016-02-01:
[image]

shan-94, 2016-02-01:
If I enter characters through a keyboard then the KL46 automatically transmits the data to an available radio ...
Similarly when it receives data, it automatically displays it

Oitzu, 2016-02-01:
And the other side? I mean the RPi side. Does the RPi side outputs errors, when you try to transmit?

shan-94, 2016-02-01:
When I try to transmit via the Raspi, I get the "failed transmission" error probably because it does not get an acknowledgement on transmitting data 
In receive mode, there is no output as the Raspi is just waiting for a valid re

shan-94, 2016-02-01:
When I try to transmit via the Raspi, I get the "failed transmission" error probably because it does not get an acknowledgement on transmitting data 
In receive mode, there is no output as the Raspi is just waiting for a valid data pack from the transmitter

Oitzu, 2016-02-01:
Can you despite this post the status output of the RPi? Just to make sure that the registers look right on both nodes.

shan-94, 2016-02-02:
[image]

Oitzu, 2016-02-02:
You maybe want to do a 
```

shan-94, 2016-02-02:
OK .. I  have a problem.. Everytime I execute the "gettingstarted" example for the first time after powering the pi, it shows the above output.
But when I consecutively run the code the Rx and Tx addresses change automatically : 
[image]

Oitzu, 2016-02-02:
Ok... i see so they really have the correct addresses.

shan-94, 2016-02-02:
I think the problem is with the pipes.. If you look at the KL46z code, do you think the pipes for write and read are correct ?

Oitzu, 2016-02-02:
Seems legit to me. As long as the receiving module is listening on the correct address it should send back an "auto-ack".
The "failed." message means, that no auto-ack was received.
The auto-ack itself happens on the hardware, if it is correctly configured.

shan-94, 2016-02-02:
The auto ack is ocnfigured on boh devices .. I will try again in a while and let you know if I can get them to communicate

Oitzu, 2016-02-02:
Just to clear up your "gettingstarted" problem:
I looked into the code and the addresses are set after the printing of the details.
So on first run it shows the wrong (default) addresses on the second run it shows the correct setted addresses.

Oitzu, 2016-02-09:
@shan-94 any progress on this?

shan-94, 2016-02-09:
Hi Oitzu,
I have paused work on it for now .. Will resume in a while .. I will update you then

closed 2016-11-10 by TMRh20