nRF24/RF24 #207: RPi not working with gettingstarted_call_response

jetiston, 2016-02-06:
Hello. I wanted to set up an RF24 mesh so I started out with testing the examples.

I'm using a Raspberry Pi 2 and arduino uno. I was able to make the gettingstarted sketch work between them so I'm pretty sure my wirings are correct. I've read that RF24Mesh uses dynamic payload and auto-ack so I tested gettingstarted_call_response but it doesn't work.

I tried the gettingstarted_call_response between two arduinos and they work.

If anyone has experienced this or any suggestion on what I should test next would be very helpful. Thanks in advance.

TMRh20, 2016-02-20:
Don't have much for suggestions here unfortunately, beyond ensuring the radio CE & CS pins are defined the same, and you have the radioNumber configured appropriately in the example.

If the above is confirmed and it is not working, I would assume a device/module issue.

ab316, 2016-02-23:
I have this issue with Pi 2 and STM32 Nucleo L053, to which I have ported the library.
RF24Mesh works between two Nucleos.
But with PI as master and Nucleo as the only node the mesh doesn't work. The master is responding to the NETWORK_POLL but the node isn't receiving anything.

jetiston, 2016-02-24:
It was hardware after all. We soldered 100uF caps to the NRF modules and gettingstarted_call_response now works smoothly.

But I still can't make the RF24Mesh examples work.

Avamander, 2016-02-24:
It never hurts trying bigger caps.

closed 2016-03-23 by TMRh20