nRF24/RF24 #208: Problem with pipe address ending on 0x00

fjuedes, 2016-02-06:
Might be a little bit off-topic as it may be an issue with the radio chip itself, but here are the really competent people...
I wrote a little chat-program for Arduino which prints everything it receives from the radio to the serial port and sends everything it receives from the serial out to the radio, just a little practice:
```
#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"

#define PAYLOAD_SIZE 32
#define PIPE_ID 0xDeadBeef01LL

RF24 radio(9,10);
String gv_SerialInput = "";

void serialEvent(){
  gv_SerialInput = Serial.readStringUntil('\n');
} // END serialEvent

void setup(void){
  Serial.begin(115200);
  Serial.setTimeout(50); // 50ms
  printf_begin();

  // Setup and configure rf radio
  radio.begin();
  radio.setRetries(15,15);
  radio.setPayloadSize(PAYLOAD_SIZE);
  radio.setChannel(100);
  radio.setDataRate(RF24_1MBPS);
  radio.openReadingPipe(0,PIPE_ID);
  radio.openWritingPipe(PIPE_ID);
  radio.startListening();
  radio.printDetails();

  Serial.println("RF24-example, chat:\n");
  Serial.println("I will print everything i receive to Serial.");
  Serial.println("I will send everything you type to wireless.");
} // END setup 

void loop(void) {

  if (radio.available()) {
    char v_data[33];
    radio.read(&v_data,PAYLOAD_SIZE);
    Serial.print("Received: ");
    Serial.println(v_data);
  } // END if

  if (gv_SerialInput.length() > 0) {
    radio.stopListening();
    Serial.print("Sending  :"); Serial.println(gv_SerialInput);
    bool ok = radio.write(gv_SerialInput.c_str(),PAYLOAD_SIZE);
    if (ok) {
      Serial.println("ok...");
    } else {
      Serial.println("failed.");
      delay(100);
    } // END if
    radio.startListening();
    gv_SerialInput = "";
  } // END if

} // END loop
```
Whenever i use a pipe address ending on 0x00, for example:
```
#define PIPE_ID 0xDeadBeef00LL
```
The radio reports an error when sending and the receiving station does not receive anything.
Am i making a mistake, is this a problem with the library or is that another thing that this "diva-chip" doesn't like?
Thank you very much in advance for your help.

Best regards from Houston (TX),
     Frank/2

TMRh20, 2016-02-20:
After a bit of testing, I believe this is due to the library itself.

Pipe 0 is special, because it can be used to transmit, meaning it establishes bidirectional communication using a single address, lets call Addr00 and it can also listen on a separate address lets call Addr01

When calling the startListening() function, the library checks the Least Significant Byte of the pipe0 reading address, and if it is equal to 0, the pipe0 reading pipe is closed. 

1. Radio #0 sends to radio#1:  Open pipe 0 with Addr01 for reading and writing
2. Radio #0 calls stopListening: Open pipe 0 with Addr00 for reading only

What is happening here, is that when you call startListening(), the radio is detecting pipe0 as having a blank address, and closing the reading pipe. 

*Why?* Closing the pipe0 reading pipe when not in use prevents a number of issues.

*Workaround:* Use pipe1 as in the gettingstarted example, or stay away from using 0 in the LSB.

*Potential Fix:* The library could check every byte, but of course that increases code size... This probably should be mentioned in the docs at least though I think.

TMRh20, 2016-03-23:
I was thinking of doing a full fix, but realized address 00 should really be avoided anyway. See http://maniacalbits.blogspot.ca/2013/04/rf24-addressing-nrf24l01-radios-require.html

closed 2016-03-23 by TMRh20

fjuedes, 2016-03-23:
> Oracle <http://www.oracle.com>
> Frank Jüdes | Senior Principal Consultant | +1.713.885.4421
> Oracle America Inc.
> 1200 Smith Street, Suite 1500
> Houston, TX 77002
> Frank Jüdes
> 2419 Kenbridge Drive
> Houston, TX 77067-1252, USA
> Cell: +1-713-885-4421

Hi TMRh20,

Thank you very much for that link, it lead me to this article 
http://maniacalbits.blogspot.ca/2013/04/rf24-avoiding-rx-pipe-0-for-enhanced.html
which explains the problems i am encountering with the radio: I was 
heavily using pipe-0 and with timing being unpredictable under Linux, 
especially on a RasPi-B1 i encountered frequent package losses.
I will have to re-think my concept for a self configuring sensor-network...

Best regards from Houston (TX),
      Frank/2

TMRh20, 2016-03-23:
With RF24Mesh I use pipe0 just for writing and for multicast in listening
mode. Pipe5 is the dedicated parent pipe which leaves room for 4
children/node with 5 children on master node because no parent pipe.

http://maniacalbits.blogspot.ca/2013/04/rf24-avoiding-rx-pipe-0-for-enhanced.html

>
> I was thinking of doing a full fix, but realized address 00 should
> really be avoided anyway. See
http://maniacalbits.blogspot.ca/2013/04/rf24-addressing-nrf24l01-radios-require.html