nRF24/RF24 #211: NRF24 Comunication samples doesn't works

Xenon7697, 2016-02-10:
I have plugged NRF24l01 exactly as writen. Power to 3.3V on power have 100uF and 100nF capacitor. But I have unreasonable incoming digits
Sent response 4294967167
Sent response 4294967167
Sent response 4294954815
Sent response 4286578559
Sent response 1811939328
Sent response 4294471680
Sent response 4294966543
Sent response 4294967167
Sent response 2133917696
Sent response 0
Sent response 671105792
Sent response 0
Sent response 0
Sent response 4294967166
Sent response 4294967167

and can't make it work, how to solve this?

And there is Joystick example from:
http://arduino-info.wikispaces.com/Nrf24L01-2.4GHz-ExampleSketches#js1

Packet Received - Sent response 0uS X= 0 Y= 0 Switch OFF
Packet Received - Sent response 2020360192uS X= 16504 Y= 24624 Switch ON
Packet Received - Sent response 4194303871uS X= -1 Y= -1 Switch ON
Packet Received - Sent response 1749043720uS X= 16384 Y= 26688 Switch ON
Packet Received - Sent response 4294967167uS X= -1 Y= -1 Switch ON
Packet Received - Sent response 4294932753uS X= 29695 Y= -1 Switch ON
Packet Received - Sent response 0uS X= 0 Y= 0 Switch OFF
Packet Received - Sent response 2097152uS X= 0 Y= 0 Switch OFF
Packet Received - Sent response 4294967119uS X= -3 Y= -1 Switch ON
Packet Received - Sent response 4294967167uS X= -2 Y= -1 Switch ON
Packet Received - Sent response 0uS X= 24800 Y= 112 Switch ON
Packet Received - Sent response 2115584uS X= 16416 Y= 24896 Switch ON
Packet Received - Sent response 0uS X= 0 Y= 0 Switch OFF
Packet Received - Sent response 2137014336uS X= 32638 Y= 26876 Switch ON

What can be wrong?

Avamander, 2016-02-10:
Seems like bad wiring. Triple check it.

Plus the sketch you are using is extremely old. Use the ones built into the
library instead.

Try 1000uF capacitor soldered (!) directly to the VCC and GND pins of the
nRF24L01+.

Make sure the distance of the Arduinos is at least 1m but not much more.

Make sure you have correctly defined the CS and CSN pins (see PingPair
example to see how it's defined).

Make sure you have correctly changed the radio number.

Make sure your wires are not too long 15cm max, the shorter the better plus
make sure it's thick enough so no ribbon cable.

>

Xenon7697, 2016-02-25:
ok, I have checked wiring several times, 
NRF             Arduino
GND            GND
VCC            VCC3.3+
CE              Set to 9
CSN            Set to 10
SCK            13
MOSI           11
MISO           12

I use example what is coming with library. (Getting Started) example in TMRh20 library
CS an CSN pins is set as it need to be, I have changed to diferent pins too. 
I have trying several distances from 1m to 10m
Radio number on one arduino is 0 and on second arduino I have 1
wires I have 4cm short.

Stil no luck with comunication. I have changed NRF boards to with new one, no result.

Avamander, 2016-02-25:
How does your module look like?

Xenon7697, 2016-02-25:
https://www.google.lv/search?q=nrf24l01&espv=2&biw=1920&bih=912&source=lnms&tbm=isch&sa=X&ved=0ahUKEwj39bfugpPLAhWFNJoKHbQnDPcQ_AUIBigB#imgrc=28rfhNohACvfRM%3A

Avamander, 2016-02-25:
Check if your 3.3V supply is 3.3V supply. Also attach 500uF electrolytic
capacitor to your module's GND and VCC pins.

Xenon7697, 2016-02-26:
I have measure, board have exactly 3.3052V
Capacitor is added, but result is the same.
I also have nRF24L01+PA+LNA and with this board result is the same.
I have desperate, what is wrong?

Avamander, 2016-02-26:
As a one last thing you can try removing your Arduino IDE and all RF24
libraries you can find (Use the file search just in case). Then redownload
latest Arduino IDE (1.6.*) and use the Library Manager to download the
latest library.

Also post printdetails() output here (there are examples where it's done if
you do not know how).

closed 2016-03-23 by TMRh20

Allecsxx, 2017-12-08:
HI,

HAVING SAME PROBLEM... 

This is my 3rd day trying to send "hello world"  using this tutorial https://www.youtube.com/watch?v=7rcVeFFHcFM&t=353s

I've  tripled check all connections ,,, added 1000uf cap... stable 3.3 v using osciloscope ... reinstalled IDE  and RF24 lib... 

nothing ... doesn't work 

i'm just going crazy... such simple should be but ......

Avamander, 2017-12-08:
Stable 3.3V but do you actually have communication? Does printDetails() print the right details? Are you actually using the latest stable IDE and library on a clean install? There are quite a few variables here and you mess one up and you won't get it working.

Allecsxx, 2017-12-09:
DOWNLOADED 1.8.5 IDE and installed after using ctrl panel to uninstall old one, deleted all old libraries files etc
Installed RF24  tmrh via Arduino. 

arduino UNO connections : 

SCK =13
MISO =12
MOSI=11
CE = 7
CSN= 8

MEGA 2560
CE =7
CSN =8
MISO=50
MOSI=51
SCK=52
ALSO added pinmode 53 output in void setup

Im using 3.3 v source from https://www.eetimes.com/document.asp?doc_id=1298322

ADDED 1000uf cap aswell

soldered connections between nrf24 and arduinos... 

I have 4 days now trying to send hello world ... i 'm going super crazy

ortegafernando, 2017-12-09:
I dont remember exactly, but try to CHANGE radio Number at the begining of the sketch. Uno could be 0, and MEGA could be 1. Check that way

Allecsxx, 2017-12-09:
BOTH NRF24 can't be faulty because i have a RC remote controll modified by somebody to drive the DRONE via rf24 module and both of them work ...

Allecsxx, 2017-12-09:
done the thing with radio 1 and 0 ... no luck...

Allecsxx, 2017-12-09:
USING getting started..

SERIAL monitor for UNO responds to T and R key but mega doesn't .... on mega it is stuck on RF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node
if I desolder CSN or CE it responds to buttons... 
also there is no doubt that CE and CSN are not connected correctly.

Allecsxx, 2017-12-09:
i think is just a bad library ... lots of people having issues...

Avamander, 2017-12-09:
@Allecsxx If it were a bad library it wouldn't work for anyone, "on mega it is stuck on RF24/examples/GettingStarted" sounds like your connection is messed up.

Allecsxx, 2017-12-09:
are not messed up .. just 7 cables to follow... i'm not an idiot... to prove i've asked my wife to check them just in case i'm too mad to see the problem. she said is correct...

Allecsxx, 2017-12-09:
ok... how do i do that ?

Allecsxx, 2017-12-09:
i'm just too mad nothing is clear anymore i can't concentrate

Allecsxx, 2017-12-09:
include prinf 

and printdetails () on void loop?

Avamander, 2017-12-09:
Put it into setup()'s last line.

Allecsxx, 2017-12-09:
'printDetails' was not declared in this scope

Allecsxx, 2017-12-09:
#include <nRF24L01.h>
#include <printf.h>
#include <RF24.h>
#include <RF24_config.h>

Allecsxx, 2017-12-09:
ok ok radio.printdetails();

Allecsxx, 2017-12-09:
well... nothing is coming... still stuck on press T

Allecsxx, 2017-12-09:
desoldered CE CSN ... serial monitor is printing 

Sent 87974644, Got response 4294967167, Round-trip delay 10984 microseconds

Avamander, 2017-12-09:
> desoldered CE CSN
????

Allecsxx, 2017-12-09:
yes ... i have soldered wires between NRFmodule pins to arduino pins... (on the back of arduino)

there is no short ...

Allecsxx, 2017-12-09:
believe me i arrieved to a stage that i wanted to avoid any dummy problem like lose connections and so on

Allecsxx, 2017-12-09:
ok i'm getting out to buy other 2 boards

1 arduino uno
1 mega

new pin cables and so on

i will get a breadboard psu just to make everything different

i will be back in 2 hours 

will you be able to help me?

i will provide team viewer and live images showing what i'm doing

Avamander, 2017-12-09:
The modules will *never* work without CE and CSN connection. It's quite likely you've forgotten to update the pin definitions in the sketch following that YT tutorial instead of the documentation.

Allecsxx, 2017-12-09:
i have radio pins 7 and 8

i'm just using getting started example... not modified anything

Avamander, 2017-12-09:
Then connect CE to 7 and CSN to 8 accordingly.

Allecsxx, 2017-12-09:
Done this but is not printing anything. .... It only prints "serialprint(whatever)-> press T

Allecsxx, 2017-12-09:
And get stuck.... If csn ce removed..  it prints send response random numbers.. if reverse them it prints random numbers. If csn ce connected correctly it gets stuck with the last print.

Allecsxx, 2017-12-09:
If somebody wants to see this i will provide team viwer and live images with the boards connected. Believe me guys i have 4 days with this... I eliminated all possible mistake

Avamander, 2017-12-09:
> Done this but is not printing anything. .... It only prints

It **is** printing something.

> it gets stuck

This is as specific as the sentence "some car did stopped somewhere at some point". Details are extremely needed. On what line exactly does the code freeze? What board are you currently using? What OS? What kind of wires are you using (how long are they)? Is multimeter showing the connections are continuous and there are no shorts?

Allecsxx, 2017-12-09:
Boards:

1.Arduino Uno R3
2.Mega2560

3.3 V out from breadboard PSU (tested with osciloscope)i will provide pictures in next minutes

10cm longest cable male to female breadboard cables

Multimeter showing continues conections between male to female of each pin.

Windows 8

Arduino 1.8.5  freshly installed yesterday
Rf24 tmrh freshly installed yesterday after deleting all files from previous installation

Uploaded getting started example on both boards and modified radio number to 1 for MEGA board. Also added pinmode 53 as output.(uploaded without pinmode 53 output aswell and no luck)

Allecsxx, 2017-12-09:
Added caps up to 1000uf 
Radio pins  CE CSN 7 and 8 to each board
Replaced psu with a laboratory psu the one in the link on a comment above.
Replaced cables
Replaced boards (new uno r3 , new mega2560)
Used different radio pins 9 and 10
Uploaded same sketch on new boards (the one that didn't work with old boards)
Uploaded 3 different sketchs found on tutorials where people were saying it works

What else i should do to make it work?

Allecsxx, 2017-12-09:
As i said i can easily provide team viewer and live streaming of my desk

Avamander, 2017-12-09:
One board at a time. On what line exactly does the code freeze on the Uno?

Allecsxx, 2017-12-09:
https://ibb.co/nFB7Hw
https://ibb.co/eArk4b
https://ibb.co/gKA1cw
https://ibb.co/i8adPb

Uno connections

Allecsxx, 2017-12-09:
https://ibb.co/eBPGAG
https://ibb.co/d1BJPb
https://ibb.co/eUEiqG
https://ibb.co/eUEiqG

MEGA connections

Allecsxx, 2017-12-09:
print details is not printing anything on serial monitor

Allecsxx, 2017-12-09:
https://youtu.be/kFgmGcbcZrU

Allecsxx, 2017-12-09:
Ok... Now it's quite... Thanks for help. Definitely there is a issue with the library. If at least somebody was ibterested to debug i would be keeping it... But as nobody is interested i will move on

Allecsxx, 2017-12-09:
Haha... Shit library... Installed other library work perfect..

Allecsxx, 2017-12-09:
Sorry... But you deserve.. You never believed ... Everything was to prove there is something wrong on my end.... Guess what... Didn't touched the boards ... Used your library instead and is not working. How about that?

wmarkow, 2017-12-09:
It can be that @Allecsxx had the similar issue with the **GettingStarted** as @RaviOli [here in issue #217](https://github.com/nRF24/RF24/issues/217#issuecomment-186177943) and me [as well in #217](https://github.com/nRF24/RF24/issues/217#issuecomment-186283789). 

As @ortegafernando has [mentioned that before](https://github.com/nRF24/RF24/issues/211#issuecomment-350435771)
>I dont remember exactly, but try to CHANGE radio Number at the begining of the sketch. Uno could be 0, and MEGA could be 1. Check that way

there is an issue with the **GettingStarted** example; if you upload the same sketch to both Arduino's then it will not work. You need to change the **radioNumber** variable to **0** and upload the sketch to **Arduino Uno**, then you need to change the **radioNumber** to **1** and upload the sketch to **Arduino Mega**.

But anyway - before proceed with two boards communication - first I would make sure that every board can correctly communicate with their RF24 chip; so printing the **printDetails()** result to serial port would be helpful.

Allecsxx, 2017-12-09:
You have a video above comment showing uploadind print details function and not printing.

Allecsxx, 2017-12-09:
Something is going wrong with the library as no matter to which example i add printdetails it will not print any details on serial monitor. I included printf and added radio.printDetails(); last line in void setup. Not working this way.

Allecsxx, 2017-12-09:
Also i uploaded with radio 0to UNO and radio 1 to MEGA and no improvements :) its funny now as nothing can make it work. But other maniac library works perfectly.

Allecsxx, 2017-12-10:
STATUS		 = 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1

Avamander, 2017-12-10:
From that half of printdetails, I can quite safely assume one of four things listed (in the rank of probability):
1) Bad wiring, connection, module - see [the documentation](http://tmrh20.github.io/RF24/) how it should be
2) Bad pin definitions 
3) Faulty SPI library
4) Black magic

And you still haven't spent the effort to determine on what line the code freezes.

Allecsxx, 2017-12-10:
Successfully made it to work with 2 pro mini boards. ... I think there is a issue with hardware spi

wmarkow, 2017-12-11:
So it works with two pro mini boards. It doesn't work with Uno vs Mega. I'm using the library with Uno and Pro Mini and Node MCU and the communication work fine between those three boards.

As @Allecsxx comented (here)[https://github.com/nRF24/RF24/issues/211#issuecomment-350442994] (and he has also mentioned that other maniac library works) I can say that it may freeze on **radio.begin();** on the **Arduino Mega** board.

Allecsxx, 2017-12-11:
Yes.. it works for a little... But freeze after some time... Is not stable. But at least it is f'ing printing details which tmrh is not..... Whyyyyyyy?:(((

Allecsxx, 2017-12-11:
Is not printing on. Pro mini either. Just a led that blinks ... when pressed a button from transmitter All i could make to work

Allecsxx, 2017-12-11:
Can somebody give me an example of printdetails in serial monitor that proves good connections ? Also a bit elaborated?

Allecsxx, 2017-12-11:
@Avamander

Don't know exactly how to determine on what libe the code get stuck. I can provide team viewer and you may see the problem clearly. I'm not that advanced.

[33 more replies]