nRF24/RF24 #212: Problem using Cortex M0 (SAMD21)

alejoast, 2016-02-11:
Hi.
I'm trying to create a wireless sensor network using 6 nRF24L01+ with 6 Adafruit Feather M0 Basic Proto based on a SAMD21 processor.
When I compile any code (basic or complex) I receive this error:

"In file included from C:\Users\Alejo\Documents\Arduino\libraries\RF24\RF24.cpp:9:0:

C:\Users\Alejo\Documents\Arduino\libraries\RF24\nRF24L01.h:27:21: error: expected unqualified-id before numeric constant

 #define CONFIG      0x00

                     ^

C:\Users\Alejo\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel/Device/ATMEL/samd21/include/component/eic.h:705:34: note: in expansion of macro 'CONFIG'

   __IO EIC_CONFIG_Type           CONFIG[3];   /**< \brief Offset: 0x18 (R/W 32) Configuration n */

                                  ^

C:\Users\Alejo\Documents\Arduino\libraries\RF24\nRF24L01.h:27:21: error: expected unqualified-id before numeric constant

 #define CONFIG      0x00

                     ^

C:\Users\Alejo\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel/Device/ATMEL/samd21/include/component/wdt.h:290:34: note: in expansion of macro 'CONFIG'

   __IO WDT_CONFIG_Type           CONFIG;      /**< \brief Offset: 0x1 (R/W  8) Configuration */

                                  ^

exit status 1
Compilation Error"

This is not happening if I use an Arduino Mega or Nano, but only for SAMD boards.
What can I do to use this library with a SAMD board such as the Adafruit Feather M0 Basic Proto?

Thanks

Avamander, 2016-02-14:
There seems to be a conflict with the samd21 wdt.h. It should be possible to rename all instances of "CONFIG" to "CONF" in all the places it exists in RF24. Or you can edit the file wdt.h and do the same.

tapanther, 2016-02-18:
I had this same issue, and changing CONFIG to CONFIG_RF24 in all library files worked perfectly. I was able to get the wireless modules to communicate without issue. I can therefore confirm that Avamander's comment is a correct solution.

TMRh20, 2016-02-20:
Maybe this change should be incorporated into the library, to prevent any issues with other devices/libs that make use of a CONFIG variable?

Avamander, 2016-02-20:
Indeed that seems to be a good idea.

alejoast, 2016-02-27:
It will be great if TMRh20 incorporate this change into the library.

TMRh20, 2016-02-27:
It's on the list but every time we make the list smaller it just gets bigger! Soon.

TMRh20, 2016-03-23:
Added... finally

closed 2016-03-23 by TMRh20

robertcasanova, 2016-06-10:
@alejoast Hi, i also have a Feather M0. Does it still work for you? 
I try upload a simple sketch but program hangs when i call radio.write. I'm trying this code: [gist](https://gist.github.com/robertcasanova/d8d67fde85edcdd62f836b70345975bc)

Thx for help!

alejoast, 2016-06-10:
Hi @robertcasanova It works fine for me. I think it could be a problem with the message itself. Try with a simple code as this and tell me how it goes:
[featherRF24simpleTest.zip](https://github.com/TMRh20/RF24/files/308468/featherRF24simpleTest.zip)

robertcasanova, 2016-06-10:
Thx @alejoast Trying sending with Adafruit Metro and it works fine but on Feather it fails. I change this in your code to test failures:
`if(!radio.write(msg, 1)) {
      Serial.println("fail");  
    } else {
      Serial.println("Sent!");  
    }`

alex007badass, 2017-10-24:
What wiring did you use? Does it matter which pin I use as interrupt? Thx for helping a noob.