nRF24/RF24 #213: RF24::begin(void) always returns false

wmarkow, 2016-02-11:
I have noticed that the RF24::begin(void) always returns false which may confuse a developer who is using this library for the first time. I think commit with SHA 16c999e8c71134401a78d4d46435517b2271d6ac comments out a little to much in the RF24::begin(void) function.

I can provide a very simple fix for that.

I was thinking that - by the way - RF24::begin(void) could also set the **failureDetected** variable to true if needed. What do you think? I can provide a fix for that as well.

Avamander, 2016-02-15:
It should be tested before merging by people with different hardware.

TMRh20, 2016-02-20:
The mess of code in begin() is due to some suggested changes to detect if setup was successful, but it never worked quite right, so I think removing that code is fine.

wmarkow, 2016-02-21:
Thanks for merging my pull request. So for now we have the following piece of code:
```
uint8_t setup=0;
...
if( setDataRate( RF24_250KBPS ) )
{
  p_variant = true ;
}
setup = read_register(RF_SETUP);
...
// if setup is 0 or ff then there was no response from module
return ( setup != 0 && setup != 0xff );
```
My source code static analysis and datasheets for both RF24L01 and RF24L01+ shows that after hardware power up (when the power supply is attached) begin() method return will return true (for both RF24L01 and RF24L01+ chips) if modules are wired correctly and are not damaged. 
It should be ok even if someone will call begin() method multiple times (and sets by the way the power level RF_PWR to '00'). For both chips the **setup** value should be different than 0x00 or 0xff. Some issues me come up when we start touching the LNA_HCURR bit in RF24L01 (but for now we do not do that).

wmarkow, 2017-05-04:
I think I can close this issue now. Now there is a new API method available that is able to check if the nRF24 module is connected, correctly powered up in responsive.

closed 2017-05-04 by wmarkow