nRF24/RF24 #214: Does not compile on the Genuino 101 (Intel Curie based)

t3chguy, 2016-02-13:
Compiling the `pingpair_ack` example leads to:

```
In file included from /tmp/arduino_68b1fd984563d5bd42a76711d8af580d/pingpair_ack.ino:20:0:
/home/t3chguy/Arduino/libraries/RF24/printf.h: In function ‘void printf_begin()’:
/home/t3chguy/Arduino/libraries/RF24/printf.h:31:29: error: ‘fdevopen’ was not declared in this scope
   fdevopen( &serial_putc, 0 );
                             ^
exit status 1
Error compiling.
```

t3chguy, 2016-02-13:
Modifying `printf.h` to compile but be essentially useless (same as __arm__) I get:

```
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:569:17: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
   printf_P(PSTR("Data Rate\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_datarate_e_str_P[getDataRate()]));
                 ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:570:17: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
   printf_P(PSTR("Model\t\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_model_e_str_P[isPVariant()]));
                 ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:571:17: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
   printf_P(PSTR("CRC Length\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_crclength_e_str_P[getCRCLength()]));
                 ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:572:17: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
   printf_P(PSTR("PA Power\t = "PRIPSTR"\r\n"),  pgm_read_word(&rf24_pa_dbm_e_str_P[getPALevel()]));
                 ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp: In member function ‘void RF24::print_status(uint8_t)’:
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:370:11: error: ‘printf_P’ was not declared in this scope
           );
           ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp: In member function ‘void RF24::print_observe_tx(uint8_t)’:
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:381:11: error: ‘printf_P’ was not declared in this scope
           );
           ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp: In member function ‘void RF24::print_byte_register(const char*, uint8_t, uint8_t)’:
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:393:38: error: ‘printf_P’ was not declared in this scope
     printf_P(PSTR(PRIPSTR"\t ="),name);
                                      ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp: In member function ‘void RF24::print_address_register(const char*, uint8_t, uint8_t)’:
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:408:38: error: ‘printf_P’ was not declared in this scope
     printf_P(PSTR(PRIPSTR"\t ="),name);
                                      ^
In file included from /home/t3chguy/.arduino15/packages/Intel/hardware/arc32/1.0.4/cores/arduino/WString.h:34:0,
                 from /home/t3chguy/.arduino15/packages/Intel/hardware/arc32/1.0.4/cores/arduino/Arduino.h:100,
                 from /home/t3chguy/Arduino/libraries/RF24/RF24_config.h:58,
                 from /home/t3chguy/Arduino/libraries/RF24/RF24.cpp:10:
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp: In member function ‘void RF24::printDetails()’:
/home/t3chguy/.arduino15/packages/Intel/hardware/arc32/1.0.4/cores/arduino/avr/pgmspace.h:54:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 #define pgm_read_word(addr) (*(const unsigned short *)(addr))
                                                            ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:569:48: note: in expansion of macro ‘pgm_read_word’
   printf_P(PSTR("Data Rate\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_datarate_e_str_P[getDataRate()]));
                                                ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:569:100: error: ‘printf_P’ was not declared in this scope
   printf_P(PSTR("Data Rate\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_datarate_e_str_P[getDataRate()]));
                                                                                                    ^
In file included from /home/t3chguy/.arduino15/packages/Intel/hardware/arc32/1.0.4/cores/arduino/WString.h:34:0,
                 from /home/t3chguy/.arduino15/packages/Intel/hardware/arc32/1.0.4/cores/arduino/Arduino.h:100,
                 from /home/t3chguy/Arduino/libraries/RF24/RF24_config.h:58,
                 from /home/t3chguy/Arduino/libraries/RF24/RF24.cpp:10:
/home/t3chguy/.arduino15/packages/Intel/hardware/arc32/1.0.4/cores/arduino/avr/pgmspace.h:54:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 #define pgm_read_word(addr) (*(const unsigned short *)(addr))
                                                            ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:570:46: note: in expansion of macro ‘pgm_read_word’
   printf_P(PSTR("Model\t\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_model_e_str_P[isPVariant()]));
                                              ^
/home/t3chguy/.arduino15/packages/Intel/hardware/arc32/1.0.4/cores/arduino/avr/pgmspace.h:54:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 #define pgm_read_word(addr) (*(const unsigned short *)(addr))
                                                            ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:571:49: note: in expansion of macro ‘pgm_read_word’
   printf_P(PSTR("CRC Length\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_crclength_e_str_P[getCRCLength()]));
                                                 ^
/home/t3chguy/.arduino15/packages/Intel/hardware/arc32/1.0.4/cores/arduino/avr/pgmspace.h:54:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 #define pgm_read_word(addr) (*(const unsigned short *)(addr))
                                                            ^
/home/t3chguy/Arduino/libraries/RF24/RF24.cpp:572:49: note: in expansion of macro ‘pgm_read_word’
   printf_P(PSTR("PA Power\t = "PRIPSTR"\r\n"),  pgm_read_word(&rf24_pa_dbm_e_str_P[getPALevel()]));
                                                 ^
exit status 1
Error compiling.
```

Avamander, 2016-02-19:
I am not familiar with your platform. Can you please tell what library/functions you use to print data to serial port?

t3chguy, 2016-02-19:
@Avamander well the Serial library works on this platform as I'd expect it to (since it is an official Arduino product). So Serial.write, print, println etc are all functional as on all other Arduino boards.

Avamander, 2016-02-19:
@t3chguy Then I guess you could replace all calls of `printf_P(PSTR(` with `Serial.print(` . It has been discussed to fix these kind of issues by creating a nice library-like print function that's multiplatform. See https://github.com/TMRh20/RF24/issues/189.

t3chguy, 2016-02-19:
Thanks @Avamander I will probably try it simply for test purposes, and to verify that the rest of the lib works on this platform (don't see any reason it shouldn't)
the way you worded it (as well as #189) suggest that this is a temporary "bug" until the printing mechanic is standardised.

TMRh20, 2016-02-20:
Not necessarily a 'bug' just a platform not specifically supported. See http://tmrh20.github.io/RF24/Portability.html
 
Removing the inclusion of printf.h & printdetails() might get it working, but its usually a matter of doing device detection, and adjusting for the use of program-memory specific functions like printf_P in the RF24_config.h file.

t3chguy, 2016-02-20:
I used "bug" due to thats what this issue has been labelled as.

madias123, 2016-03-14:
Just as info: They library is nearly 100% compatible with stm32duino (stm32 devices with arduino IDE, formerly known as "leafabs maple"), but the printf (we don't use/have it) is really an annoying thing. Furthermore it was too much affort for me building up #ifdefs so we've done a "own" branch for it, called "RF24_STM32". Not nice and I'm unhappy with that fact, but this was the fastest solution. So maybe this will work Genuino 101 also (maybe other SPI speed devider settings needed)

dvmitry, 2016-07-28:
Is there a solution for this problem? 
With library from previous post I have error messages:
`
E:\prog\arduino\libraries\RF24_STM32\RF24_STM32.cpp: In member function 'void RF24::print_observe_tx(uint8_t)':

E:\prog\arduino\libraries\RF24_STM32\RF24_STM32.cpp:380:11: error: 'printf_P' was not declared in this scope

           );

           ^
`

Avamander, 2016-07-28:
The solution would be to declare printf_P or comment out the lines where it
is used.

3jackattack3, 2017-05-25:
Just had the same issue with printf on the arduino 101 board. Has anyone found a stable solution? I only need very basic functionality to send/receive data for my project.

srolf, 2017-06-11:
It would be really nice to see a modification so that 101 works. I would like to use it in the Adeept Smart Car Kit as a replacement for the Uno.

closed 2019-05-10 by TMRh20