nRF24/RF24 #215: pyRF24

ajitam, 2016-02-14:
Hi,

I wish to use pyRF24 for a project I'm working on but I can't manage to run simple script. My setup is a bit different than "gettingstarted", but the thing is that C++ version is working and PY is not. Here are both programs

**C++**
```
#include <cstdlib>
#include <iostream>
#include <sstream>
#include <string>
#include <unistd.h>
#include <RF24/RF24.h>

using namespace std;

#define PACKAGE_SIZE 9
#define CHANNEL 76

RF24 radio(22,0);

int main(int argc, char** argv)
{

  printf("\n> begin");
  radio.begin();  
  radio.setAutoAck(false);
  radio.setDataRate(RF24_2MBPS); 
  radio.setPALevel(RF24_PA_MAX);
  radio.setPayloadSize(PACKAGE_SIZE);
  radio.setChannel(CHANNEL);
  radio.setCRCLength(RF24_CRC_16);
  printf("...done\n");

  printf("\n");
  radio.printDetails();
  printf("\n\n");

  printf("\n> setting pipes\n");

  radio.openReadingPipe(1, 0x3030303030);

  printf("\n> listening");
  radio.startListening();
  printf("...done\n");

  while (1) {

    if (radio.available()) {
      unsigned long got_time;

      while(radio.available()){
        radio.read( &got_time, sizeof(unsigned long) );
      }

      printf("*RCV\n");
      delay(925);
    }

  }

  return 0;

  printf("\n\n");

}
```

**PY**

```
#!/usr/bin/env python

from __future__ import print_function
import time
from RF24 import *
import RPi.GPIO as GPIO

PACKAGE_SIZE =  9
CHANNEL =  76

radio = RF24(22, 0);

radio.begin()
radio.setAutoAck(False)
radio.setDataRate(RF24_2MBPS)
radio.setPALevel(RF24_PA_MAX)
radio.setPayloadSize = PACKAGE_SIZE
radio.setChannel(CHANNEL)
radio.setCRCLength(RF24_CRC_16)

radio.printDetails()

radio.openReadingPipe(1,0x3030303030)
radio.startListening()

while 1:
  if radio.available():
    while radio.available():
      print('*RCV')

```

both `.printDetails()` outputs the same values except `RX_PW_P0-6` which is:
* **C++**: `0x00 0x09 0x00 0x00 0x00 0x00`
* **PY**: `0x00 0x20 0x00 0x00 0x00 0x00`

any ideas?

mz-fuzzy, 2016-02-14:
at first glance you are missing read(...) call in your python version.

ajitam, 2016-02-15:
yes a know, but this doesn't affect the logic. Only this that read does is that if you don't read data `radio. available()` is always `true`. But id PY I never fall into `if radio.available():`

wmarkow, 2016-02-15:
I can't help with Python version but I can help with C++. Let's find out if you C++ version is really working. @ajitam, can you answer some questions first:
1. Can you post the output of **radio.printDetails()**
2. Can you post what is the result of **radio.begin()**
3. If your hardware setup is right and **radio.available()** returns true then it means that some data has been received by RF24L01 chip. Do you have other chip sending data? Is the wiring for transmitter correct?
4. Maybe you are generally facing some hardware issues? Is your wiring correct? Do you have that additional ceramic capacitor soldered on your RF24L01 between VCC and GND pins (I've read that it may help).

Avamander, 2016-02-15:
Electrolytic capacitors seem to work better usually (100-1500uF).

wmarkow, 2016-02-15:
>both .printDetails() outputs the same values except RX_PW_P0-6 which is:
>
> *C++: 0x00 0x09 0x00 0x00 0x00 0x00
> *PY: 0x00 0x20 0x00 0x00 0x00 0x00

However these above means that for C++ there are 9 bytes available in the chip's receive buffer but for PY there are 32 bytes available (which means that the receive buffer is full).

ajitam, 2016-02-15:
@wmarkow 

1. `.printDetails()` output is

```
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0x2020202020
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x0f
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 2MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
```

currently (as you can see from first post) I've set only reading pipe 1 `radio.openReadingPipe(1, 0x2020202020)`

2. can you explain how can I post `.begin()` results?

3. Yes Arduino is sending data.

4. Currently I don't have capacitor but I didn't think that can be a problem since also "gettingstarted" example with sending and receiving is working without problems in C++. (I will solder capacitors tomorrow)

Additional notes: I just install a fresh copy of RF on another Raspberry Pi (first gen) and I have the same results - C++ work, PY doesn't

wmarkow, 2016-02-15:
> can you explain how can I post .begin() results?
It simply returns a **bool** value.

ajitam, 2016-02-15:
Solved!

There was an error in line `radio.setPayloadSize = PACKAGE_SIZE`, must be  `radio.payloadSize = PACKAGE_SIZE`.

don't know why I did get an error (?)
Now `radio.available()` gets triggered - so this issue is resolved.

closed 2016-02-15 by ajitam