nRF24/RF24 #217: Hanging up at radio.write

Ravi0li, 2016-02-18:
Hallo,

I have the problem, that the GettingStarted is hanging up, when I would transmit a packed.
This is the console output from the GettingStarted porgramme, when I push T:
> RF24/examples/GettingStarted
> *** PRESS 'T' to begin transmitting to the other node
> *** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
> Now sending
Than i can't do anything and I must restart the Arduino.

Some facts to my setup:
- I use the actual version of the RF24-Library from GitHub
- I have a 220uF capacitor between 3,3V and GND
- I build it on a perfboard (wirelenght < 5cm)
- I use a Ardino Nano Clone (but i have the same problem with a original Arduino Uno)
- I have checked all connections with a mulitmeter. They are right.
- I use this board: [NRF24L01 PA LNA SMA](http://www.ebay.de/itm/201493808442)
- I have switch between three of the NRF24L01 PA LNA SMA boards. I have at all three this problem.

Have somebody I tip for me? Or what I can do now?

p.s. sorry for my bad English :smile:

Avamander, 2016-02-18:
The Nano's builtin regulator is insufficient for that module. Please buy a
separate regulator that is able to supply that much power to the module or
downgrade to non-PA & non-LNA module.

Ravi0li, 2016-02-18:
OK, now i use for a LM2596S Down-Stepper with two 220uF capacitors behinde. With the oscilloscope, I can only detect some +20mV peaks from the LM2596S. I thing, now its not a VCC problem?

But I have the same problem. Now with `#define FAILURE_HANDLING`

> RF24/examples/GettingStarted
> *** PRESS 'T' to begin transmitting to the other node
> *** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
> Now sending
> failed
> Failed, response timed out.
> Now sending
> failed
> Failed, response timed out.
> Now sending
> failed
> Failed, response timed out.
> ............

And when I let write out the status every try. I become this:
> E
> E
> F
> F
> ...

Also he write something into the TX buffer, but don't send it. And after 3 times is the TX buffer full...

Avamander, 2016-02-18:
Triple check that the wiring is absolutely correct. Also, did you solder the capacitors (I had the best results with 800uF electrolytic ones) to the module's power pins? 
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

Avamander, 2016-02-18:
Wait, what does the second Arduino print out? 
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

Ravi0li, 2016-02-19:
I check the wiring again. It is absolut correct. And I connected GND from the Arduino to the other Power supply. 
Now, I solder on both module's a 1000uF electrolytic capacitor direktly on the power pins of the module.
But this make it not better. I try both sides as Transmitter and Receiver, with the same solution:

Transmitter:
> *** PRESS 'T' to begin transmitting to the other node
> *** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
> Now sending
> failed
> Failed, response timed out.
> Now sending
> failed
> Failed, response timed out.

Receiver:
> RF24/examples/GettingStarted
> *** PRESS 'T' to begin transmitting to the other node

Avamander, 2016-02-19:
Then there are two things most likely. Your library is somehow outdated or broken or you have defined the CS and CS pins wrong. If you can please attach a picture of the connection/build and your code.

t3chguy, 2016-02-19:
Best way to test wiring is `radio.printDetails();` which (if printf is functional) will print the content of some registers, if the data that gets dumped to Serial looks like zeros or Null then something is wrong.

Ravi0li, 2016-02-19:
Here is a picture from the build: http://www.pic-upload.de/view-29785724/funkmodul.jpg.html
And I use exactly the same code like the [GettingStarted](https://github.com/TMRh20/RF24/blob/master/examples/GettingStarted/GettingStarted.ino) programm.

I modified the print code a little bit, an becomme this for `radio.printDetails();`
> STATUS = 0E
> RX_ADDR_P0-1 = 0x65646F4E31 0x65646F4E32
> RX_ADDR_P2-5 = C3 C4 C5 C6
> TX_ADDR	 = 0x65646F4E31
> RX_PW_P0-6 = 20 20 0 0 0 0
> EN_AA	 = 3F
> EN_RXADDR = 2
> RF_CH	 = 4C
> RF_SETUP = 3
> CONFIG	 = E
> DYNPD/FEATURE = 0 0

wmarkow, 2016-02-19:
I have built the test environment that contains two boards: Arduino Uno and NodeMCU (with ESP8266). I'm sure that both boards and chips RF24L01 are correctly wired. After uploading the exact same sketches to both boards and making one of them to transmitter (by sending 'T' in the terminal) I must say that It doesn't work; transmitter sends the data but it doesn't receive any. So I confirm the version of @Ravi0li.

But - after some source code investigations I've figured out that both boards flashed with the exact same code have the **radioNumber** variable set to zero/false and the write/read pipes are set only once in the **setup()** method. These pipes are not reconfigured later when you set one board to the transmition state and the packet is sent to an incorrect address.

As a fix I suggest to get rid of that **radioNumber** variable and just relay on what **role** says and reconfigure the pipes correctly according when changing from receiver to transmitter (and from transmitter to receiver). It will simplify the getting started scenario as both boards will be flashed withe same code.

I can provide a pull request.

wmarkow, 2016-02-19:
@Ravi0li, can you please retest if that fix works with your boards?

Ravi0li, 2016-02-19:
Thx wmarkow, but it is not better.
And the tests before, i have change the `radioNumber` for the two arduinos.

I see the problem at the transmission, because no time come one of this Flags in the Status Register: `TX_DS` or `MAX_RT`.
Alltimes I fall in this Line: https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L803
And I measure the CE-Level. This goes to 0V for every transmittion try.

wmarkow, 2016-02-19:
Ok @Ravioli, I thought you have put the same software on two boards without changing the **radioNumber**.
Take a look at your posted **radio.printDetails()**:
> RX_PW_P0-6 = 20 20 0 0 0 0
Correct me if I'm wrong: does it mean that there are 32 bytes of data available in RX pipes? If so then your chip has received something. It may be a clue.

Avamander, 2016-02-19:
It again seems an issue caused by clone modules. If possible try getting your hands on another set (of non PA LNA) originals.

Ravi0li, 2016-02-19:
> does it mean that there are 32 bytes of data available in RX pipes?

I read in the datasheet, that this mean: You can receive 32 bytes at this pipe.
The RX_FIFO is free, what I can see at the **STATUS** register with the Value **E**.
-> **RX_P_NO** -> Bits 3:1 -> **111** -> RX FIFO Empty

I think, that i destroy my PA & LNA Chip. The NRE24L01+ work and answer on all SPI-Calls. But he can't send or recieve anything over the amplifier. 

@wmarkow Thank you for the good explanation from **TX_DS** and **MAX_RT**!
@Avamander I order new modules. Some Orginal, and some with PA LNA

phillipgreenii, 2016-02-20:
Was the ultimate source of the problem being cloned components?  I'm seeing a similar problem with mysetup where I can get one "ping" and "pong" to work, but then the second cycle hangs on write.  This sounds similar to the problem mentioned above.  I am using a base module which includes a capacitor.

Avamander, 2016-02-20:
The thing with nRF24L01+ modules is that there are many clones that work
but not exactly as specified in the documentation. So when these kinds of
issues arise and for others the same thing has been tested to work, the
hardware side is most likely the issue and should be triple checked from
any faults.

Now I'd try if I were you I'd check if the CE and CSN pins aren't switched
(and are right) plus make sure there aren't any bad connections (bad solder
joints for example).

Ravi0li, 2016-02-21:
I have read a little more over the PA-LNA Moduls in the www and found some interesting informations. I try now some things and write than my result.

But one question I have for you: When delete the nRF24L01+ modul something from the TX-FIFO? When he send the packege, or not before he come into TX_DS/MAX_RT?

TotoGe, 2016-02-21:
Hello. The same trouble. regular NRF24l01+   connect to  mini pro, and 2 modules were talking. I connect them to their place NRF24L01+PA+LNA and the problems begin. Communication is, but it is very unstable and at a short distance. It seems to me that the PA does not work. It may be that the library does not include the PA?
   Incidentally, the current consumption of 30 mA , instead of 100 mA,

Avamander, 2016-02-21:
PA should not affect the way the library works. Your description sounds like bad connection (wiring) or power supply issues.

Ravi0li, 2016-02-21:
I have found out, that the CE-Pin was not connected with the NRF24L01 and the amplifier.
Now I have repair this PCB-Fail. I have the same fails now, but the TX-FIFO have no more a overflow.
I'm looking for further errors :smile:

TotoGe, 2016-02-21:
soldered to the inputs nrf + pa modules 2 capacitors, tantalum to 10 microfarads and a barrel of 50 microfarads. Communication has become stable at 20m 5 concrete walls :). But still it is not enough. Most of all I do not like that 30mA current, instead of 100 mA consumption nrf + pa

TotoGe, 2016-02-21:
Example "Getting Started". baud rate defaults to 1 MB? decrease - will be on longer distance?
and yet it is not clear why the consumption of 30 mA.  corrected radio.setPALevel(RF24_PA_MAX);  Or 100mA - it pulses during transmission, that is very short? Maybe that's why the meter shows just 30mA?

Ravi0li, 2016-02-21:
@TotoGe: http://www.rfmw.com/data/RFaxis_Nordic_nRF24LE1_Range_Extension_with_RFaxis%20RFX2401_App_Note.pdf

TMRh20, 2016-02-21:
@TotoGe Also using 250KBPS via ``` radio.setDataRate(RF24_250KBPS); ``` should give max range and longer transmit period, making it easier to test with a meter.

TotoGe, 2016-02-21:
TMRh20 ,you have a lot of experience, please tell me what the real distance for Chinese NRF + pa? And then maybe I want a lot of them? )
  I read your library several times, now studying examples. Thank you for your hard work

TotoGe, 2016-02-21:
Ravi0li . Yes, your link is also at the maximum consumption should be up to 140mA. But I do not understand, it is a constant current consumed nrf + pa, or only during the transfer? Can I see these 140mA multimeter or oscilloscope only?

Avamander, 2016-02-21:
Most likely, the modules are actually quite fast.

TMRh20, 2016-02-21:
The short answer is I don't really know.

The long answer is that wireless is & can be directional and largely
affected by the environment. In a perfect testing environment, or very
rural area, results should be much better than in a city apartment block so
mileage will vary. Take a look at
http://n8henrie.com/2015/03/range-testing-for-wireless-arduino-projects-rf-433-mhz-and-nrf24l01/
for a good summary & testing.

TotoGe, 2016-02-21:
TMRh20  thank you very much again. and the library, and what is the time to newbies.
  All understood, I will seek

closed 2016-04-13 by Avamander