nRF24/RF24 #220: multiple radios..

WERAQS, 2016-02-20:
any one has any success with more than 3 nRF24 radios on RPi 2 model? 2 works but.. 3+ ?

Avamander, 2016-02-21:
This sounds an awful a lot like a X-Y problem. What's your goal, what do
you wish to do with the nRF24L01+ modules?

WERAQS, 2016-02-21:
faster scanner & logger with zero

TMRh20, 2016-02-23:
Per page 103/104 of [bcm2835 datasheet](https://www.raspberrypi.org/wp-content/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) RPI actually has 2 auxillery SPI interfaces, each with 3 chip-selects, but I don't think there are any supported drivers for them.

Another option may be to use manual CS pin toggling, but that will likely slow things down.

Technically more than 2 radios is not supported.

closed 2016-03-23 by TMRh20