nRF24/RF24 #221: pipes are leaking, literally.

WERAQS, 2016-02-22:
node names with 3xx are working at Ch80, others at 75, both are at pipe A4.. and Ch80 reads both Ch75 and Ch80, something is definitely wrong here..

NODE_128	3358	8386	2696
NODE_53	2991	8378	5071
NODE_306	3143	25155	989	3389
NODE_63	2946	8381	5744
NODE_52	2721	8385	5789
NODE_74	3421	8382	4613
NODE_29	2970	8378	4890
NODE_50	3603	8380	5849
NODE_306	3120	25154	989	3392
NODE_73	3695	8384	6518
NODE_64	2923	8386	5107
NODE_306	3073	25157	988	3398
NODE_140	2728	8378	5207

TMRh20, 2016-02-23:
Wut?

WERAQS, 2016-02-24:
Node_3xx operates at CH80 with Pipe 0xAAAAAAAAA4 open,
Node_others operates at CH75 with same pipe open,

base raspberry pi reads both CH75 & CH80, I wonder why.

TMRh20, 2016-02-24:
Some sort RF overlap. Try wider channel spacing, moving the modules apart, adjusting orientation.
Also 0xAAAAAAAAA4 is [not a good address to use](http://maniacalbits.blogspot.ca/2013/04/rf24-addressing-nrf24l01-radios-require.html)

WERAQS, 2016-02-25:
gotcha, back to 0xD0E0A0D0B2..8 again, thanks

closed 2016-03-23 by TMRh20