nRF24/RF24 #222: writeAckPayload does not return the result

gilperon, 2016-02-23:
When I am using this really nice library I always try to use writeAckPayload  as much as possible cause it brings LOTS of advantages, but there is a serious problem with it that now I am facing and see no solution: I need writeAckPayload to tell me if it was sucesfully or not when sending the ACK.

I have an arduino that sends information to another and after receiving the information that arduino replies back with some information. I could use "write" twice, when sending for the first time from ardino 1 -> 2 and write the information back from 2 -> 1 using "write". But using writeAckPayload it becomes much better and I dont even have to change the state of my modules.

Any suggestion to know if the writeAckPayload was succesfully or not?

TMRh20, 2016-02-24:
If you pre-load a payload, then whenever RX device receives a payload, it was successful in sending the ACK. See the [gettingstarted_call_response](http://tmrh20.github.io/RF24/GettingStarted_Call_Response_8cpp-example.html) example.

gilperon, 2016-02-25:
Thank you for replying but I think you are a little wrong. I took a look at the code you provided an check this line:

radio.writeAckPayload(pipeNo,&gotByte, 1 );

In the docs the function writeAckPayload does not return anything (void) so it's not possible to the receiver know that the ACK that it sent arrived to the module that sent the data. If I am wrong, could you please tell me which line of the code you provided that tells the receiver that the ACK was succesfully received by the module that initially sent the data?

TMRh20, 2016-02-25:
ACK stands for acknowledgement... What more needs to be said? If you think
I am incorrect please provide a code example demonstrating errant behavior.

gilperon, 2016-02-25:
I know what ACK is. What needs to be said is this: IS IT POSSIBLE TO KNOW IF THE ACK WAS RECEIVED BY THE SENDER OF THE MESSAGE? Cause if I am sending an important data using the payload of the ACK I need to make sure it arrived and so far I dont see how to make sure ACK was received. Is there any function I can call after "writeAckPayload" that tells me if the ACK was succesfully received?

TMRh20, 2016-02-25:
Yes, call radio.available();

Process:
1. RX device writes ACK to TX buffer
2. TX device sends to RX device
3. RX device a: receives and b: ACKNOWLEDGES reception
4. TX device retries if no ACK

If RX device has received a payload, it must have been acknowledged.

gilperon, 2016-02-25:
Thank you for clarifying this but I still didnt get it. I will try to be clearer, sorry if I was not before:

1) A sends 123 to B using "write";
2) B receives 123 from A and sends 456 to A using "writeAckPayload";
3) A receives the ACK and so "write" will return "true". Great, so A knows that B received it's message;
4) B does not know for sure if the ACK with payload 456 it sent arrived to A cause writeAckPayload does not return anything.

I know that if the ACK was not received by A it will retry to resend the information but let's suppose I configured 5 retries and all the first 4 retries failed and now we are in the very last retry (the fifth). Suppose that A sent the information to B and B is now replying with the ACK. B will never know if:

1) the ACK was indeed received by A or;
2) A gave up sending the message after 5 failed attempts;

TMRh20, 2016-02-25:
This is why I ask for code to demonstrate the behavior. Debates about the potential behavior of hardware is useless.  
  
There really is no limit to the number of retries, because, if you set it to 5 and it fails, you can just try again. It all comes down to your protocol, and the details of implementation.

gilperon, 2016-02-25:
Ok, it looks like you dont wanna help, I got it. Cause if you wanted to help you just had to take a look at the example at "gettingstarted_call_response" and try to check if you can confirm if the ACK was succesfully received. I will use MIRF library it has a much better support.

Avamander, 2016-02-25:
If you have enabled auto-ACK then isn't it built into the system to resend x amount of times until ACK is received? You don't get any boolean value for that but the packets are still send and acknowledged if not then resent and acknowledged.

gilperon, 2016-02-25:
I think you dont understand what is going on: I need to make sure A->B communication was succefull and ACK payload fro B->A was succesful bor BOTH modules. A and B need to know that the TWO communications were succesfully transmitted. With MIRF I dont have this kind of problem, but with your library I do have this problem.

Avamander, 2016-02-25:
That functionality (if I understand you correctly) does not exist in RF24. But you can detect manually if A has not received ACK from B as A will retransmit the message when no ACK from B is received.

TMRh20, 2016-02-25:
> Cause if you wanted to help you just had to take a look at the example at "gettingstarted_call_response" and try to check if you can confirm if the ACK was succesfully received.

So you are having a problem and you want me to do the testing for you? LOL wtf?

https://en.wikipedia.org/wiki/Wikipedia:Reference_desk/How_to_ask_a_software_question

gilperon, 2016-02-25:
"wtf" you are an idiot and dont know how to read a question. It's a bug. Actually let's be honest: in this thread you cleared showed you are ignorant and an idiot. You dont even understand the basics of this module and not even shockburst technology, othwersie you would know how to fix this. This behaviour of yours is of no surprise to me, it just demonstrates why this fucking library is so BUGGY. EVERYBODY on Google says this library sucks and has lots of bugs. Now it's clear why that happens, cause you have no idea wtf you are doing bi***. Go f* yourself I will be better off using MIRF.

TMRh20, 2016-02-25:
:0 Well, thanks for trying to help @Avamander  

@gilperon I'm not trying to offend you or make you angry, but without the following it is nearly impossible to help:
1. Describe your goal
2. Describe the problem
3. Post a minimal and complete example demonstrating the problem.

I don't think the mirf lib supports ack payloads btw.

Avamander, 2016-02-25:
@gilperon  Noone has before needed this kind of functionality. You can't blame TMRh20 for not implementing it. I also seriously doubt there's ACK ACK functionality by the hardware, in order to confirm that a packet is received the target needs to respond, you could go into infinite loop with ACKing ACKs.

ortegafernando, 2016-02-25:
@Avamander and @TMRh20 THANKS A LOT for your work. It is a "free work" for us. 

@gilperon Please, dont insult to anyone. They only want to help you. They work for free. If you have any solution to this problem, please, let us to know it.

Otherwise, what @TMRh20 ask you for is the minimum. A simple piece of code where you have that problem.

I have been using this library for months and it is great and well developed.

Please, read the SPECIFICATIONS os RF24 chip. But read it several times. It is very hard to read it, but if you read it you will understant ACKPAYLOAD perfectly.

Hope we could find the solution to your problem.

Regards,

TMRh20, 2016-02-27:
>  Avamander: I also seriously doubt there's ACK ACK functionality by the hardware

Yeah, thats why I say it comes down to protocol and implementation details. The TX device can indicate if prev transmissions failed, use a header with counter + retries, etc. if acceptable.

There are often so many ways to do things, it is usually best to attempt it, and then if problems, post the code&info (even if it seems *crappy*) and it makes it much easier to help. (and we don't have to act as free consultants) Plus it doesn't matter much what language ppl speak, because we all speak C++ 
  
Also I'm marking this as a bug and close it because ack payloads must be cursed or something...

closed 2016-02-27 by TMRh20

yongkianpeng, 2017-09-14:
Hi everyone, I have a question related to ACK, I understand the ACK will be returned to the sender immediately. I wondered could you tell me: 

[1] On the Receiver, I have ACK set to "ONE"
[2] If I later called writeAckPayload to place "TWO" as the ACK
[3] What will a Transmitter received when it send a request to the Receiver now?
[4] Would it be "ONE" or "TWO" ?

If it is "ONE", is there any way to "CLEAR" the ACK BUFFER so that it will return "TWO"?

Thank you very much.

Avamander, 2017-09-14:
I suspect this is again counterfeit behaviour or an mixed environment with counterfeits, until proven otherwise I'm going to leave the tag to warn future users coming from Google. 

@yongkianpeng Hi! Small question, you set ACK with what function how exactly?

yongkianpeng, 2017-09-14:
Hi Avamander, 

Thanks for the quick response.

On the Receiver setup(), I have set: radio.enableAckPayload();

In my code, I set ACK by: radio.writeAckPayload( 1, &Ack_Package, sizeof(Ack_Package) );
where Ack_Package = Byte Array.

Thank you.

batata004, 2017-09-14:
@yongkianpeng my friend, I know this is not the answer you are looking for but I strongly advise you to use MIRF. Since I migrated to MIRF I had 0 problems, I repeat: zero problems. This library that you are using is tremendously faulty, sooner or later, if you make serious use of it, you will realize it lacks lots of optimizations.

Avamander, 2017-09-14:
@batata004 If you'd provide constructive feedback that would be nice.

@yongkianpeng The parameters of writeAckPayload are:

```
pipe | Which pipe# (typically 1-5) will get this response.
buf   | Pointer to data that is sent
len   | Length of the data to send, up to 32 bytes max. Not affected by the static payload set by setPayloadSize().
```
If you set the response payload it's sent with the next message that needs to be ACKd. So if you set it to "TWO" and after that a message needs to be sent, "TWO" is sent.

From documentation:  
`Ack payloads are handled automatically by the radio chip when a payload is received. Users should generally write an ack payload as soon as startListening() is called, so one is available when a regular payload is received.`

batata004, 2017-09-14:
@Avamander  ok, I will give you a constructive feedback: ditch this library and move on to MIRF. This is as constructive I can get.

yongkianpeng, 2017-09-14:
*smiale* 👍 Thanks Avamander. Could you point me to the right lib for MIRF and maybe some good example on handling ACK? Thank you veyr much.

Avamander, 2017-09-14:
@yongkianpeng MIRF is not something I maintain, RF24 is what I maintain, you have to go look up it's documentation somewhere else. In RF24's case `/examples/pingpair_ack/pingpair_ack.ino` is a good example of using ACK without custom response.

@batata004 It's unfortunate you can't provide a single case where it's actually the library at fault and lack the debugging skills to actually determine what's going on if you're experiencing errors.

yongkianpeng, 2017-09-14:
@Avamander : Sorry, I thought you were suggesting to me to move to a better nrf24 lib. I am quite happy with nrf24 lib. 

I am just puzzled why after I performed writeAckPayload from the Receiver to write "TWO" as ACK, I still observed the Transmitter getting ACK "ONE" for 2 cycles then receive "TWO" .... Is there a "buffer" for ACK? If so, can it be cleared?

Thank you very much.

Avamander, 2017-09-14:
@yongkianpeng Hmm, could you please post the sender/receiver code here?

yongkianpeng, 2017-09-14:
@Avamander Thanks for your patient. 

I have plug out codes to simplify my illustration. Thank you in advance for your help. 

Below are two piece of codes, Tx and Rx. You can see that in the Rx codes, there is an IF statement to update the ACK twice when the ackCount is a factor of 10. So you will see the log from the Receiver setting the ACK when ackCount=10 (I have also attached the results below the codes and BOLT that particular line).

**Transmitter codes**
  #include <SPI.h>
  #include "RF24.h"
  
  RF24 nrf24(2, 15);
  
  const uint64_t pipes[3] = { 0xF0F0F0F0E1LL, 0xF0F0F0F0E2LL, 0xF0F0F0F0E3LL };
  
  int reqCount=0;
  int ackCount=0;
  
  void setup() {
  
    Serial.begin(9600);
    printf("\nSetup starts\n");
    
    nrf24.begin();
    nrf24.setRetries(15, 15);
    nrf24.setPayloadSize(8);
    nrf24.setPALevel(RF24_PA_HIGH);
    nrf24.enableAckPayload();
    
    nrf24.openWritingPipe(pipes[1]);
  
  }
  
  void loop() {
  
    printf("Sending ReqCount = %i\n",reqCount);
    bool ok = nrf24.write( &reqCount, sizeof(reqCount) );
  
    if (ok) {
      
      int respRetry=10;
      printf("Waiting for NRF response ");
      while (!nrf24.available() && respRetry>0) {
        printf(".");
        respRetry--;
        delay(100); // Pause for the Receiver to response
      }
      printf("\n");
    
      if (nrf24.available()) {
    
        nrf24.read(&ackCount,sizeof(ackCount));
        printf("Received AckCount = %i\n\n",ackCount);
      }
    
      reqCount++;
    }
  
    delay(5000);
  }

**Receiver codes**
#include <SPI.h>
  #include "RF24.h"
  
  RF24 nrf24(2, 15);
  
  const uint64_t pipes[3] = { 0xF0F0F0F0E1LL, 0xF0F0F0F0E2LL, 0xF0F0F0F0E3LL };
  
  int reqCount=0;
  int ackCount=0;
  
  void setup() {
  
    Serial.begin(9600);
    printf("\nSetup starts\n");
    
    nrf24.begin();
    
    nrf24.enableAckPayload();
    nrf24.setRetries(2,15);
    nrf24.setPayloadSize(8);
    nrf24.setPALevel(RF24_PA_HIGH);
    
    nrf24.openReadingPipe(1,pipes[1]);
    nrf24.startListening();
  }
  
  void loop() {
    
    if (nrf24.available()) {
  
      nrf24.read(&reqCount, sizeof(reqCount));
      printf("Received ReqCount = %i\n",reqCount);
  
      printf("Setting AckCount = %i\n\n",ackCount);
      nrf24.writeAckPayload( 1, &ackCount, sizeof(ackCount) );
  
      ackCount++;
  
      // Update ACK
      if (ackCount/10*10==ackCount) {
        printf("updating AckCount = %i\n\n",ackCount);
        nrf24.writeAckPayload( 1, &ackCount, sizeof(ackCount) );
  
        ackCount++;
      }
  
    }
  }

**Receiver logs**
Received ReqCount = 19
Setting AckCount = 6

Received ReqCount = 20
Setting AckCount = 7

Received ReqCount = 21
Setting AckCount = 8

Received ReqCount = 22
Setting AckCount = 9

**updating AckCount = 10**

Received ReqCount = 23
Setting AckCount = 11

Received ReqCount = 24
Setting AckCount = 12

**Transmitter log**
Sending ReqCount = 8
Waiting for NRF response 
Received AckCount = 7

Sending ReqCount = 9
Waiting for NRF response 
Received AckCount = 8

Sending ReqCount = 10
Waiting for NRF response 
Received AckCount = 9

Sending ReqCount = 11
Waiting for NRF response 
Received AckCount = 10

Sending ReqCount = 12
Waiting for NRF response 
Received AckCount = 11

Sending ReqCount = 13
Waiting for NRF response 
Received AckCount = 12

Sending ReqCount = 14
Waiting for NRF response 
Received AckCount = 13

yongkianpeng, 2017-09-14:
@Avamander I also noticed down the line in the Transmitter logs, I can see it skipping the tenth (20, 30 ...etc) .... what am I missing .... Any idea? Thanks.

Sending ReqCount = 368
Waiting for NRF response 
Received AckCount = 18

Sending ReqCount = 369
Waiting for NRF response 
Received AckCount = 19

Sending ReqCount = 370
Waiting for NRF response 
Received AckCount = 21

Sending ReqCount = 371
Waiting for NRF response 
Received AckCount = 22

TMRh20, 2017-09-14:
Referring to the original post, the chip will hold three ack payloads. You can clear the tx buffer with radio.flush_tx(); 

This lib has been intensively debugged, as github history shows, most "bugs" or issues relate to user or hardware issues.
"[1] On the Receiver, I have ACK set to "ONE"
[2] If I later called writeAckPayload to place "TWO" as the ACK
[3] What will a Transmitter received when it send a request to the Receiver now?
[4] Would it be "ONE" or "TWO" ?

If it is "ONE", is there any way to "CLEAR" the ACK BUFFER so that it will return "TWO"?

Thank you very much"

It is "ONE"

yongkianpeng, 2017-09-14:
@TMRh20 Thank you very much. With radio.flush_tx() it works perfectly now.