nRF24/RF24 #225: Intel edison

aafeliz, 2016-02-25:
NEW ISSUE on edison
cp utility/MRAA/includes.h utility/includes.h
g++ -Wall -fPIC  -c RF24.cpp
g++ -Wall -fPIC  -c utility/MRAA/spi.cpp
g++ -Wall -fPIC  -c utility/MRAA/gpio.cpp
gcc -Wall -fPIC   -c utility/MRAA/compatibility.c
g++ -shared -Wl,-soname,librf24-bcm.so.1 -lmraa   -o librf24-bcm.so.1.0 RF24.o spi.o gpio.o compatibility.o
[Installing Libs]
make: install: Command not found
Makefile:110: recipe for target 'install-libs' failed
make: *** [install-libs] Error 127

TMRh20, 2016-02-25:
``` make: install: Command not found ``` ???

Maybe try this?:
http://stackoverflow.com/questions/17444777/make-install-command-not-found

closed 2016-02-25 by Avamander

aafeliz, 2016-02-25:
thank you that worked 
export PATH=$PATH:/path/to/install-utility
then i used,
make install -B RF24_SPIDEV=1