nRF24/RF24 #226: Does the library work with Digispark?

gitgrimbo, 2016-02-26:
Hi, I get compilation errors when I select a Digispark board from Arduino IDE (v1.6.7) and try to compile `rf24ping85`.  Is the [Digispark](https://digistump.com/wiki/digispark/tutorials/basics) supported?  It is based on the Attiny85 so I hoped it would be.

[image]

I am trying to follow the [3-pin approach](http://nerdralph.blogspot.co.uk/2014/01/nrf24l01-control-with-3-attiny85-pins.html).

I set the CE and CS pins to be equal, as mentioned [in this comment](https://github.com/TMRh20/RF24/issues/54#issuecomment-68406799).

````c
// CE and CSN are configurable, specified values for ATtiny85 as connected above
#define CE_PIN 3
//#define CSN_PIN 4
#define CSN_PIN 3 // uncomment for ATtiny85 3 pins solution
````

This is the compilation output.  From what very little I know, my guess is the Digispark is not treated like a regular Attiny85.

Thanks.

```
In file included from c:\apps\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,

                 from c:\apps\arduino\hardware\tools\avr\avr\include\avr\interrupt.h:38,

                 from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/WProgram.h:8,

                 from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/Arduino.h:4,

                 from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h:17,

                 from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.cpp:14:

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h: In member function 'void SPISettings::init_AlwaysInline(uint32_t, uint8_t, uint8_t)':

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h:146:16: error: 'SPE' was not declared in this scope

     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |

                ^

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h:146:27: error: 'MSTR' was not declared in this scope

     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |

                           ^

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h:146:65: error: 'DORD' was not declared in this scope

     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |

                                                                 ^

In file included from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.cpp:14:0:

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h: In static member function 'static void SPIClass::beginTransaction(SPISettings)':

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h:203:5: error: 'SPCR' was not declared in this scope

     SPCR = settings.spcr;

     ^

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h:204:5: error: 'SPSR' was not declared in this scope

     SPSR = settings.spsr;

     ^

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h: In static member function 'static uint8_t SPIClass::transfer(uint8_t)':

[... 463 lines cut ...]

In file included from c:\apps\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,

                 from c:\apps\arduino\hardware\tools\avr\avr\include\avr\interrupt.h:38,

Multiple libraries were found for "RF24.h"
 Used: C:\Users\grimbo\Documents\Arduino\libraries\RF24
                 from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/WProgram.h:8,

 Not used: C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\RF24
                 from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/Arduino.h:4,

                 from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.h:17,

                 from C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.cpp:14:

C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\SPI\SPI.cpp:74:18: error: 'SPE' was not declared in this scope

     SPCR &= ~_BV(SPE);

                  ^

exit status 1
Error compiling.
```

Avamander, 2016-02-26:
Your digistump installation is broken. The SPI library is broken it seems
and it's not caused by RF24. Open the blink example and include SPI
library, initalize it in setup() and see if it compiles.

You also have duplicate installation of RF24. Delete
C:\Users\grimbo\AppData\Local\Arduino15\packages\digistump\
hardware\avr\1.6.7\libraries\RF24 and
C:\Users\grimbo\Documents\Arduino\libraries\RF24
and redownload the library and place it to C:\Users\grimbo\Documents\
Arduino\libraries\.

gitgrimbo, 2016-02-26:
Thanks for your quick reply.  I can get the errors down to the following (I have done a fresh install of Arduino IDE 1.6.7 and removed most of the digispark-added libraries), but it still looks like the `#define`s and `#include`s that should be part of compilation for Digispark (as an Attiny85) are not run (these `#define`s would cause `<SPI.h>` to be `#include`d).

````
In file included from C:\Users\grimbo\Documents\Arduino\libraries\RF24/RF24.h:18:0,

                 from C:\Users\grimbo\Documents\Arduino\rf24ping85\rf24ping85.ino:58:

C:\Users\grimbo\Documents\Arduino\libraries\RF24/RF24_config.h:77:19: fatal error: SPI.h: No such file or directory

   #include <SPI.h>

                   ^

compilation terminated.

exit status 1
Error compiling.
````

If I add the following to `C:\Users\grimbo\Documents\Arduino\libraries\RF24\RF24_config.h`

````c
#define __AVR_ATtiny85__
````

then the sketch compiles.  But this doesn't feel right to me.  I feel like this ought to be picked up automatically at compile time based on my chosen board (Digispark).

Apologies if this isn't very clear, I'm very new to Arduino/Digispark.

Avamander, 2016-02-26:
The boards do not identify themselves right. It's digispark dev's fault (or just a weird design feature). I'll try to create a version of RF24 for you to test.

Avamander, 2016-02-26:
Try this https://github.com/Avamander/RF24/archive/master.zip, does it compile without any modification?

gitgrimbo, 2016-02-27:
Same error I'm afraid.

wmarkow, 2016-02-27:
Digispark doesn't contain a valid SPI implementation as ATTiny85 doesn't embedd SPI interface; it embedd USI instead. Including SPI will not work. There is Atmel's application note on how to implement SPI over USI.

Avamander, 2016-02-27:
The library would work if I'd know the board tag (as it does not identify
itself as __AVR_ATtiny85_) it would be a easy fix. I'll try one more thing.

gitgrimbo, 2016-02-27:
I've tried searching for 'digis' (case insensitive, to try and find any references to **digis**park/**digis**tump) in various files/folders with no real success.  In what files do existing boards/chipsets do their 'platform `define`s'?  (apologies if my terminology is confusing)

Avamander, 2016-02-27:
https://github.com/Avamander/RF24/archive/master.zip does it work now?

gitgrimbo, 2016-02-27:
Nope, sorry.

If I add the following to my sketch (rf24ping85.ino) it compiles, but I guess that's pretty obvious.

````c
#define AVR_DIGISPARK__
````

Board is still selected as:

[image]

wmarkow, 2016-02-27:
I don't know how did you managed to compile it. It doesn't compile for me even if I include the 
```
#define __AVR_ATtiny85__
```
in the RF24_config.h

I'm using Eclipse C++ with this arduino-eclipse-plugin which binds to arduino IDE 1.6.5. Digispark does define the __AVR_ATtiny85__ and the correct files are included:
first **arduino-1.6.5/hardware/tools/avr/avr/include/avr/io.h**
then **arduino-1.6.5/hardware/tools/avr/avr/include/avr/iotn85.h**
then **arduino-1.6.5/hardware/tools/avr/avr/include/avr/iotnx5.h**

In these file there is no definitions of standard SPI registers (like DORD for example) as AtTiny85 doesn't have a hardware SPI; it has USI instead. I think it will not work even if you get the code compiled: the SPI registers will be mapped to some memory addresses that will probably make no sense for AtTiny85 and the compiled program may  not work.

Digispark [SPI implementation](https://github.com/digistump/DigistumpArduino/tree/master/digistump-avr/libraries/SPI) have almost the same files as [generic Arduino library](https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/libraries/SPI). I do not know why Digispark included a library that will not work :)
There is also the [ATTinyCore library](https://github.com/SpenceKonde/ATTinyCore) but it looks like it doesn't support SPI for AtTiny85.

What you need is - I think - an SPI implementation over USI hardware interface. Take a look at [Atmel's AVR319](http://www.atmel.com/images/doc2582.pdf). You can check out these projects:
 * https://github.com/JChristensen/tinySPI
 * https://github.com/tessel/avr-usi-spi

The best solution would be to port SPI over USI implementation into Digispark's SPI library so it would be compatible with standard Arduino API.

Avamander, 2016-02-27:
RF24 has ATTiny85 support it's just the digispark boards have been configured so that the compiler preprocessor doesn't use the right "#include"-s. Someone could ask what's the compiler preprocessor tag from digispark themselves. I have no idea where one could find it. Until that use https://github.com/TMRh20/RF24/ version and add `#define __AVR_ATtiny85__` when you're using the board. If it doesn't work then your RF24 is either outdated or installation broken.

wmarkow, 2016-02-27:
I'm sorry @Avamander - I have edited my comment (added two lines at the end) while you were writing your comment.

Avamander, 2016-02-27:
I did too update my comment :D

wmarkow, 2016-02-27:
@Avamander, now I know. My Eclipse took the RF24 library from Digispark distribution instead of TMRh20's. I have throw away that Digispark's RF24 library and used the correct one :) Now I managed to compile it; it compile however in Eclipse I don't have to define __AVR_ATtiny85__. There is SPI implementation at the end of RF24 library (TMRh20's); this implementation uses the **USI** interface which is embedded in ATTiny85.
I can't test it now since I don't have (yet) any Digispark boards.

@Avamander - thanks for clarifications.

w-mo, 2016-03-08:
additionally to deleting the RF24 libary from the digispark subfolder I had to copy the spi.h from RF24/utility/ATTiny to /RF24 . I also renamen it to SPI.h but I don't know if it does anything.

This thread helped me a lot! Also it was the only source I found for my problem, so I guess this is reference now.

Avamander, 2016-03-08:
Please don't replace the SPI library, use the `#define` instead to ensure hassle free compatibility.

w-mo, 2016-03-09:
@Avamander I didn't replace anything.

closed 2016-03-13 by Avamander