nRF24/RF24 #227: License for example/GettingStarted/*

simon1573, 2016-02-27:
This might be a stupid issue, but I'll throw it here anyway;

Many of the individual sketches got its license in its header, e.g. [Transfer.ino](https://github.com/TMRh20/RF24/blob/master/examples/Transfer/Transfer.ino#L1) etc.

The exception is the GettingStarted-sketches:

- [GettingStarted.ino](https://github.com/TMRh20/RF24/blob/master/examples/GettingStarted/GettingStarted.ino)
- [GettingStarted_CallResponse.ino](https://github.com/TMRh20/RF24/blob/master/examples/GettingStarted_CallResponse/GettingStarted_CallResponse.ino)
- [GettingStarted_HandlingData.ino](https://github.com/TMRh20/RF24/blob/master/examples/GettingStarted_HandlingData/GettingStarted_HandlingData.ino)

What kind of license are they under? GPL2?

TMRh20, 2016-02-27:
Officially TBD. This just came up with rf24ethernet and I need to review,
ensure all contributors and borrowed code is accounted for in docs etc.
especially for rf24 which was initially a merging of info from anywhere I
could find it at the time.

It seems GPL 2 or 3 are the only likely options to keep the code free and
open.

Avamander, 2016-02-27:
If possible GPLv3 should be chosen as it benefits the users the most.

Avamander, 2017-04-15:
Basically, as the core has GPLv2, you can't link anything else with it. Thus anything committed into this repository has to be GPLv2 otherwise they would break the core's license.

closed 2017-04-15 by Avamander