nRF24/RF24 #228: Running `./setup.py build` on RPi B Rev1 seems to run out of memory

gitgrimbo, 2016-02-28:
### Summary

Running `./setup.py build` seems to fail because of a lack of memory.  If an RPi B Rev1 is not capable of building this code then perhaps a warning can be added to the docs (and maybe in the script itself).

I can run the build successfully on my RPi 2 Model B.

### Steps

I am following the instructions on this page http://tmrh20.github.io/RF24/Python.html.

Using this site, http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/, here is my RPi version.

````
pi@raspberrypi1:~/rf24libs/RF24/RPi/pyRF24 $  cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 2.00
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 0002
Serial          : 000000003e433a87
````

Revision **0002** maps to **Model B Revision 1.0 	256MB**.

Here is the console output.  Sometimes I don't see the failure text and it seems to hang instead.

````
pi@raspberrypi1:~/rf24libs/RF24/RPi/pyRF24 $ ./setup.py build
running build
running build_ext
building 'RF24' extension
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c pyRF24.cpp -o build/temp.linux-armv6l-2.7/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
virtual memory exhausted: Cannot allocate memory
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
````

Here is a screenshot of `top` where you can see `kswapd0` and `cc1plus` fighting it out for memory and CPU.  Memory is 62.2% here, but I see it reach 80+.

[image]

Avamander, 2016-02-29:
https://github.com/TMRh20/RF24/issues/176#issuecomment-165544564

closed 2016-02-29 by Avamander

gitgrimbo, 2016-03-01:
Ah thanks, didn't spot those other issues.

Do you think it's worth adding some output to the script to warn a newbie like myself that the build might not work on older RPis?

Avamander, 2016-03-01:
If you feel like improving the script then feel free to do so and create a
pull request. Bit busy myself.

mz-fuzzy, 2016-03-01:
Error reported is my opinion fully sufficient and understandable, nothing additional should be added to the setup script. Maybe just to add a note to documentation.

gitgrimbo, 2016-03-01:
@mz-fuzzy, I understand that the error is technically correct, sufficient and understandable as far as compilation is concerned.

My point is that someone with less experience like myself wondered if it was something I was doing wrong.  If it is basically impossible to run the script successfully on a particular 'common' platform (e.g. an RPi 1 Model B Rev1 in my case) then it could be useful for the script to report that up-front.  But I agree a hint/warning in the docs would be good.

Avamander, 2016-03-02:
It is not impossible, you just need enough swap.