nRF24/RF24 #230: RF24Audio

lazaroalvarado7, 2016-03-02:
Hello.

Can you help me with two questions?

1. what version of Windows and Arduino Uno is compatible to work with the NRFL01 module?
2. What is the code to run the transmission module and the module of reception?

Thank you.

Avamander, 2016-03-02:
1. Irrelevant. 

2. See examples.

closed 2016-03-03 by Avamander

lazaroalvarado7, 2016-03-03:
I am using Windows version 10 and version 1.6.7 of Arduino and I generated a lot of mistakes.

I will try with previous versions of Arduino.

thanks in advance.

<<Original translation>>
Estoy utilizando la versión 10 de Windows y la version 1.6.7 de Arduino y me genera muchos errores.

Intentaré con otras versiones anteriores de Arduino.

Agradecimiento adelantado.