nRF24/RF24 #231: makefile improvements

mz-fuzzy, 2016-03-03:
Hi guys,

I've done some improvements into my clone:
https://github.com/mz-fuzzy/RF24/tree/makefile_improvements

What is there:

- introduction of "configure" script. Run "./configure --help" for info.
- including suport of LittleWire into main makefile & adaptations needed for that
- BBB driver renamed to SPIDEV (because it is SPIDEV :-)
- examples_RPi renamed to examples_linux
- pyRF24 dir moved one level up directly to library dir
- SPIDEV now works with any /dev/spidevX.X [ /dev/spidev2.3 -> RF24(n, 23); ]
- autodetection of driver - RPi, SPIDEV, MRAA, LittleWire
- autodetection of HW/SoC - BCM2835/RPi, BCM2836/RPi2, AM33XX/BBB, A10, A13, A20 with assignment of optimal gcc optimizing flags
- support of cross compiling including remote upload/installation (also for python wrapper)
- python wrapper compilation uses same cpu flags/optimizations as main rf24 lib
- python installer now uses setuptools instead of distutils (needed for cross compiling)

At the moment I it's not tested enough to be ready for a pull request and maybe some of changes need discussion, but I think it's pretty usable, so I'd be glad if you can check it out and provide feedback/opinions.

to clone:
`git clone https://github.com/mz-fuzzy/RF24.git -b makefile_improvements`

to compile on an arm hosts:
`./configure`
`make`
`make install`

to cross-compile rf24 library:
- get cross compiler environment, for example for RPi

`git clone https://github.com/raspberrypi/tools rpi-tools`
- set path to binaries there, like

`export PATH=$PATH:/your/dir/rpi-tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin`
- best if you have your target machine accessible via ssh now. Assure password-less login and sudo rights for user which you'll use on it.
- run configure script:

`./configure --remote=pi@your_pi_host`
- it should detect all needed information about the destination host. Now you can compile

`make`
- and install - this should install it into cross-environment; this is important for cross-compiling examples:

`make install`
- now you can upload/install to the remote machine:

`make upload`
- compile examples:

` cd examples_linux`

`make`
- and upload them to the remote host:

` make upload`

- then you should have it ready at the remote machine.
- you can cross-compile also without having remote machine available, then you have to set target SoC and driver:

`./configure --soc=BCM2836 --driver=RPi`

- python wrapper cross-compiling: it's a bit more difficult. It's not fully automatized and one needs to get python and libboost libs to cross-compiling environment. I got it in a simple way downloaded /usr and /lib directories from the target machine where both python and libboost are installed to a separate say rpi-root directory. Then it's needed to set path to that and to python headers:

`export CFLAGS="--sysroot=rpi_root -Irpi_root/usr/include/python2.7/"

- then you can build the wrapper:

` ./setup.py build compiler=crossunix`

- and make the egg package:

`./setup.py bdist_egg --plat-name=cross`

- then you get the .egg file. Upload it to the target machine and install there:

`easy_install <your_egg.egg>`

TMRh20, 2016-03-03:
Very cool, I was looking at this a few days ago and overall some really
good ideas. Can't wait to try it out a bit!

https://github.com/mz-fuzzy/RF24/tree/makefile_improvements

TMRh20, 2016-03-17:
@mz-fuzzy Apologies, but I haven't been able to properly test or review this. Everything I have looked over is great. I don't question your knowledge or programming expertise at all, so IMHO, you have free reign to implement this as desired.

TMRh20, 2016-03-24:
Ok, I finally got a chance to test this a bit, and it is working very nicely so far.

**Results:**

**Tested on:** RPiB v2.0 w/Raspbian 4.1.20+ armv6l
**With:** BCM, SPIDEV, & MRAA
**Using:** Auto detection & manual driver selection

Only one issue/confusion I ran into is the lib name change to rf24 from rf24-bcm. Changing it seems that all libs that depend on RF24 need to change, so I think the default should maybe revert to rf24-bcm here: https://github.com/mz-fuzzy/RF24/blob/makefile_improvements/configure#L317 

**Cross Compilation&Remote Upload:** Very nice to have. Seems to work OK but got an error ``` /sbin/ldconfig.real: /usr/local/lib/librf24-bcm.so is not a symbolic link ``` due to name differences in librf24-bcm.so.1 vs librf24-bcm.so.1.0 I think

One additional question I have is regarding the line https://github.com/mz-fuzzy/RF24/commit/0de24018dc4f50c0c397738c9a7acd347ea61d17#diff-850bb3124375ff25ac4198addd330b6eL33

I believe the ``` defined (LINUX) ) && defined( __arm__ ) ``` part has to do with Intell x86 devices working as both Arduino and Linux devices, so we may need somebody with an Intel board to test this change.

:+1:

mz-fuzzy, 2016-03-24:
@TMRh20 thanks for testing.
- library name - that's eternal clash between the cleanup (the lib is not bcm-only anymore) and keeping backward compatibility... I vote for cleanup and using ./config switches for the case when compatibility needs to be kept. Of course other approach is also possible.

- changing
`defined (LINUX) ) && defined( __arm__ )`
is not a mandatory change, but is unnecessary limiting non-arm devices from using SPIDEV etc... if it has its sense there for some cases then it can be kept as it was.

it still needs additional refinements beside issues you listed:
- it would be probably good to add 'config' target to the makefile so the ./configure does not need to be called explicitly (also good for backward compatibility)
- extra/interrupts example makefiles need to be yet adapted to the new concept

- to move a bit forward, I pulled the change into your repo into a separate branch and rebased to current master HEAD. Feel free to submit changes there if you need (also for others).

prophet-roshak, 2016-03-25:
@mz-fuzzy @TMRh20 great addition you've done to us all!

**Results**
**Tested on:** Intel Galileo Gen 1 (Rev D.)
**With:** SPIDEV (autodetecton), MRAA ( --driver=MRAA)

Lib compiles and links ok both ways, with target detection like this:
```
[SECTION] Detecting target machine.
[OK] machine detected: SoC=unknown, Type=unknown, CPU=i586.
```

examples_linux build OK, haven't tested real transmission yet. I had to tinker all of them to use MRAA, but i guess it's ok, as one will still need to set correct pins and bus numbers to make them work.

mz-fuzzy, 2016-03-26:
@prophet-roshak thanks;

@TMRh20 you are right, the condition 
`...defined (LINUX) ) && defined( __arm__ )...`
is needed for intel boards in arduino ide.
However it's limiting for non-arm devices in linux.

The mentioned condition is there for detecting whether to include `utility/includes.h`, which is generated by `configure` script. My assumption is that it's for all cases except arduino ide is used, meaning the condition could be just:
`#if (!defined(ARDUINO))`

Is that correct?

All other solutions that came to my mind have some drawbacks, this one seems to be best, so I'm wondering whether it works really for all cases. As far as I checked different boards on arduino ide it's working.

TMRh20, 2016-03-27:
@mz-fuzzy The ``` #if (!defined(ARDUINO)) ``` sounds like a much better solution.

Also apologies for the delayed reply, but I've going through the RF24 stack, fighting bugs and forgot to reply... but I finally think I killed the big ones :+1:

mz-fuzzy, 2016-03-28:
So I guess I resolved also issues about library naming with backward compatibility using symlinks:
ef1914b47506750d26ddbd89110700b689fea1c2

@TMRh20 From my point of view it's now finished enough to be merged to master, it would be good if you once again verify and give a 'go' if everything is ok :-)

TMRh20, 2016-03-31:
I sat down to give this a good trial run, but I'm running into some odd issues with SPIDEV breaking the gettingstarted example. I was sure I tested it before, but can't remember if I just commented it out maybe and forgot about it?

The errant line seems to be [108](https://github.com/TMRh20/RF24/blob/makefile_improvements/examples_linux/gettingstarted.cpp#L108) ``` getline(cin,input); ``` where it just hangs. If I comment it out, the code works fine. I tested on two separate RPi. If I go back to the old code, it seems ok.

This took a while to test, so I'll see if I can take a closer look soon.

mz-fuzzy, 2016-04-01:
Yep, I was successful with getting this behavior as well. However I think it's sometimes there, sometimes not. I can also see this in RF24 master branch when I use "-O2" optimization level when compiling. It also disappears if I comment out this line https://github.com/TMRh20/RF24/blob/makefile_improvements/examples_linux/gettingstarted.cpp#L91

Well, I'll set all optimizations as they are in master branch; there is also missing -pthread, also in master - which should be there so I'll add it.

But, together with all strange behavior that people report in other threads I have feeling that there is some kind of memory problem somewhere (not necessarily directly in RF24).
Unfortunately valgrind does not work correctly on arm platforms; I'll still try to detect where the problem is.

mz-fuzzy, 2016-04-01:
Ok, i think I got this one. Here
https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L576
`SPI()`
means spi instance is created and destroyed immediately. In SPI's destructor in SPIDEV there is close(this->fd), and fd is not initialized without calling SPI::init(). Thus the close closes a random file, and if there are good conditions, it closes stdin, and as a consequence getline(cin, ...) is not working properly.
So I can see 2 problems there:
- why there is such suspicious construction SPI() called in RF24.cpp? This does not make any sense to me.
- close() cannot be called in SPI's destructor unconditionally on unset fd

TMRh20, 2016-04-01:
Well that was fast...

The initial commit looks to be https://github.com/TMRh20/RF24/commit/09269a355759c49854ae844a903ac74143ac7fe8 and I 'adapted' the code from https://github.com/mcrosson/RF24/blob/master/librf24-rpi/librf24/RF24.cpp#L361 and as you can see it looks to be my bad/sloppy coding. I never really considered the call to SPI() or what exactly it did since it was slapped in there.

Many others must have seen it too tho... it is like, right there.
Good eye @mz-fuzzy !!

TMRh20, 2016-04-01:
Just to confirm also, the issue tests OK by commenting out ```SPI()``` and adding to spi.cpp:
```
    if (this->fd > 0){    
        close(this->fd);
    }
```

mz-fuzzy, 2016-04-01:
@TMRh20 I've fixed the spidev issue in master. Can you see any other issue with makefile_improvements branch?

TMRh20, 2016-04-02:
I haven't run through it in full since the SPI() issue, but didn't find any problems besides that. I think it should be about ready to merge and test further if you'd like, but I'll give it a run tonight.

TMRh20, 2016-04-02:
All checks out here, plus the SPI bugfix! Things are working nicely.

mz-fuzzy, 2016-04-04:
makefile_improvements rebased to master and tested again, seems to be working. Hopefully not much damage caused :-)

@TMRh20 thanks for the latest bugfix and all testing!

TheMeddlingMonk, 2016-04-07:
I'm having some issues now that this branch has been merged into the master. When I try to do a clean install (either with the install script or using `sudo make install -B`), it just hangs part way through the configure:

`[Running configure]`
`[SECTION] Detecting arm compilation environment.`
`  [OK] arm-linux-gnueabihf-gcc detected.`
`  [OK] arm-linux-gnueabihf-g++ detected.`
`[SECTION] Detecting target machine.`
`[OK] machine detected: SoC=BCM2835, Type=RPi, CPU=armv6l.`
`[SECTION] Detecting DRIVER`
`  [OK] DRIVER detected:RPi.`
`[SECTION] Detecting OS.`
`  [INFO] OS detected:LINUX.`
`[SECTION] Preparing configuration.`
`[SECTION] Saving configuration.`
`[SECTION] Cleaning previous builds.`

This doesn't happen if I run configure and make separately: `make clean; ./configure; make install`.

mz-fuzzy, 2016-04-07:
try now
6c8e302ad0abf1a90053246ce7434216fbd17696

TheMeddlingMonk, 2016-04-08:
Thanks; the issue appears to be resolved now.

mz-fuzzy, 2016-04-18:
considering this topic as closed; if you find an error, report it as a separate issue.

closed 2016-04-18 by mz-fuzzy