nRF24/RF24 #234: RF24 for non root (/dev/gpiomem)

yasriady, 2016-03-22:
Dear guys,

As we know gpio access only allowed for root id as the library open access to /dev/mem, but recent released on linux kernel provide non root access to gpio by using /dev/gpiomem instead.  

RF24 library seem using old version of bcm2835.c, while latest release http://www.airspayce.com/mikem/bcm2835/bcm2835-1.50.tar.gz is now available and give rootless execution of the library.

On Raspberry Pi2, we have to sudo the application to make it running. While in some circumstance, we run GUI program in non root account and could not sudo the GUI application simply.

Do you plan to upgrade the library to make used of bcm2835-1.50.tar.gz, if so, it would be very nice.

Best regards,
Dedy Yasriady / Indonesia

TMRh20, 2016-03-23:
I believe the BCM driver only allows GPIO as non-root, SPI seems to require root.

mz-fuzzy, 2016-03-24:
/dev/spidev* allows non-root access [if the user is in appropriate group](http://quick2wire.com/non-root-access-to-spi-on-the-pi/).

https://github.com/TMRh20/RF24/issues/194#issuecomment-172883559

yasriady, 2016-03-24:
mz-fuzzy, 
I will make a try.. thanks anyway ...

closed 2016-04-13 by Avamander