nRF24/RF24 #235: available() returns true

shirish47, 2016-03-28:
when I start to listen and wait for data to be available for first time.. once I receive data and available() returns true. for next all check even after using read, available returns true even when the other radio is trued of and it keeps on returning true. I think some buffer is not cleared when is RX buffered cleared? I think it should clear on itself when read? is it? can you check out.
` nrf24.startListening();
   Serial.println("waiting..");
   unsigned long started_waiting_at =millis();
   bool timeout=false;
   while ( ! nrf24.available() ){                             // While nothing is received 
      if ((millis() - started_waiting_at) > 5000 ){            // If waited longer, indicate timeout and exit while loop
          timeout = true;
          break;
      }else
          timeout=false;
    }
     if ( timeout ){                                             // Describe the results
        Serial.println(F("Failed, response timed out."));
    }else{

   Advertise a;
   uint8_t len= sizeof(a);

    nrf24.read( &a, len );
      
   Serial.print("Dev ID: ");
   Serial.println(a.dev_Id);
   Serial.print("Token ID: ");
   Serial.println(a.token_Id);

   Serial.print("Adv type: ");
   Serial.println(a.adv_type);
  //just to ensure that location is cleared
  a.dev_Id=0;
  a.token_Id=0;
  a.adv_type= 0;
  }
 Serial.println("___________has the buffered erased?____________");
   Serial.print("Dev ID: ");
   Serial.println(a.dev_Id);
   Serial.print("Token ID: ");
   Serial.println(a.token_Id);
   Serial.print("Adv type: ");
   Serial.println(a.adv_type);
  Serial.println("___________are all of above zeroed?____________");
  
 Serial.print("nrf available test: "); Serial.println(nrf24.available()); //returns true once it returns true :(
 Serial.print("timeout :");
 Serial.println(timeout);
nrf24.stopListening();// thought this would clear rx buffer :( `

TMRh20, 2016-03-28:
Try this:
```
   Advertise a;
   uint8_t len= sizeof(a);
    
   while(nrf24.available()){
      nrf24.read( &a, len );
   }
   Serial.print("Dev ID: ");
   Serial.println(a.dev_Id);
```

The radio has 3 - 32byte FIFO buffers. If the radio is connected and receiving data properly etc. this should solve it.

The RX buffer is only cleared by the radio driver if:
1. On startup (```radio.begin()```)
2. User calls ```radio.read()``` until ```radio.available()``` returns 0
3. If a dynamic payload size is invalid (corrupt dynamic payload) (```radio.getDynamicPayloadSize()```)
4. User calls ```radio.flush_rx()```

shirish47, 2016-03-28:
ohh ok I will try that, yes I tried to find flush() for rx  but I did not find it [here](http://tmrh20.github.io/RF24/classRF24.html#ab7f54decbe9d06cb026a5d3dfb505116)

TMRh20, 2016-03-28:
My bad flush_rx is private so not user accessible

<http://tmrh20.github.io/RF24/classRF24.html#ab7f54decbe9d06cb026a5d3dfb505116>

shirish47, 2016-03-28:
I think I have blown my module because of a tiny peice of wire on my board. so can't check it now. I would let you know the results when I try it. thanks buddy.

shirish47, 2016-03-29:
hi so here is it is ...
I used it as u mentioned.. 
``C
   Advertise a;
   uint8_t len= sizeof(a);    
     while ( nrf24.available() ){
       Serial.println("woow some data .. ASDASDASDSAFASFFSF");
      nrf24.read( &a, len );
    }
``

but it never comes out of while even when I power of transmitter ... :(  
And can someone tell me how to lable issues ??

TMRh20, 2016-03-29:
This can happen if wires are loose or connected improperly. I would be leaning towards hardware or wiring issues. I would suggest testing with the example sketches.

To get debug output, add the following to the gettingstarted_call_response example:
```
#include <SPI.h>
#include "RF24.h"
#include <printf.h>
```
in setup add:
```
printf_begin();
```
after Serial.begin();
and then uncomment ```radio.printDetails()``` to get debug output on Arduino.

Usually you can tell if its correct, because everything but addressing will look the same on both nodes.

Issue labeling is mostly just an admin thing, so only myself or designated 'Collaborators' can label issues. Collaborators are @mz-fuzzy, @Avamander, and @Oitzu. (To become a collaborator, start collaborating! )

shirish47, 2016-03-29:
RF24/examplesx00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
RF24/examples/GettingStarted_CallResponse
*** PRESS 'T' to begin transmitting to the other node
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e31 0x65646f4e32
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e31
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
NRF_CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

both my radio shows this even on turning on transmit mode I get sending failed. even at 10 cms I also have caps on the NRF modules ... but seems some problem Now I think it must be some hardware trouble which I need to first solve than will able to give more feed on available() problem.

shirish47, 2016-03-29:
OK fine My hardware is working and I am using Gettingstarted code and it works fine on same hardware.

TMRh20, 2016-03-29:
Then something is wrong with your code. Would need a more complete example
of both sides.

shirish47, 2016-03-30:
hi so here is code. .. just use a data structure anything for testing... https://gist.github.com/shirish47/1c082b26217708ba3f8ee7f8e7b45f5f

TMRh20, 2016-03-30:
Nothing really looks out of sorts with the code. 

I can't test now, but I'm thinking you have something weird going on, and I really don't know what. A loose wire, or slightly oxidized connector,  hardware oddity... something odd.

TMRh20, 2016-03-30:
I bet its this:
``` nrf24.setCRCLength(RF24_CRC_8);```

Radio spam?

Try:
```nrf24.setCRCLength(RF24_CRC_16);```

closed 2016-07-27 by TMRh20