nRF24/RF24 #236: ATtiny84 Pin Issue

Raynman77, 2016-03-28:
Hi, I'm having a problem with the transmission using the RF24 library. On the tiny84, i'm using the example diagram that you have provided for my wiring scheme in the example "rf24ping85" for the ATtiny84 and it's not transmitting anything. On the receiver i'm using the example "getting started". When I use the 5 wire hookup on an ATtiny85, it works just fine. However, when I wire up a tiny84 and load the same sketch, I get nothing. Here's the sketch:

ATtiny24/44/84 Pin map with CE_PIN 8 and CSN_PIN 7
	Schematic provided and successfully tested by Carmine Pastore (https://github.com/Carminepz)
                                  +-\/-+
    nRF24L01  VCC, pin2 --- VCC  1|o   |14 GND --- nRF24L01  GND, pin1
                                             PB0  2|     |13 AREF
                                             PB1  3|     |12 PA1
                                             PB3  4|     |11 PA2 --- nRF24L01   CE, pin3
                                             PB2  5|     |10 PA3 --- nRF24L01  CSN, pin4
                                             PA7  6|     |9  PA4 --- nRF24L01  SCK, pin5
    nRF24L01 MOSI, pin7 --- PA6  7|      |8  PA5 --- nRF24L01 MISO, pin6
                                  +----+
*/
// CE and CSN are configurable, specified values for ATtiny85 as connected above
#define CE_PIN 3
#define CSN_PIN 4
//#define CSN_PIN 3 // uncomment for ATtiny85 3 pins solution
int switchPin = 1;

#include "RF24.h"

RF24 radio(CE_PIN, CSN_PIN);

byte addresses[][6] = {
  "1Node","2Node"};

void setup() {
  pinMode(switchPin, INPUT);
  digitalWrite(switchPin, HIGH);
  // Setup and configure rf radio
  radio.begin(); // Start up the radio
  radio.setAutoAck(1); // Ensure autoACK is enabled
  radio.setRetries(15,15); // Max delay between retries & number of retries
  radio.openWritingPipe(addresses[1]); // Write to device address '2Node'
  radio.openReadingPipe(1,addresses[0]); // Read on pipe 1 for device address '1Node'
  radio.startListening(); // Start listening
}

void loop(void){
  
  digitalRead(switchPin);
  const char payload[] = "opened";
  if (switchPin == LOW) {
    radio.stopListening(); // First, stop listening so we can talk.
    radio.write( &payload, sizeof(payload) );
    radio.startListening(); // Now, continue listening
  }
  unsigned long started_waiting_at = micros(); // Set up a timeout period, get the current microseconds
  boolean timeout = false; // Set up a variable to indicate if a response was received or not

  while ( !radio.available() ){ // While nothing is received
    if (micros() - started_waiting_at > 200000 ){ // If waited longer than 200ms, indicate timeout and exit while loop
      timeout = true;
      break;
    }

  }

  if ( !timeout ){ // Describe the results
    unsigned long got_time; // Grab the response, compare, and send to debugging spew
    radio.read( &got_time, sizeof(unsigned long) );
  }

  // Try again 1s later
  delay(1000);
}

TMRh20, 2016-03-29:
Well the ATtiny pin config is super confusing, so my first assumption is that. 

See here for an example of my confusion: https://github.com/TMRh20/RF24/pull/155

Please double check, because you have verified the radio is working, and ATTiny is usually straight forward once connected properly.

Raynman77, 2016-03-29:
I totally agree with you. It is very confusing, especially when trying to follow some of the examples. So swapping MISO and MOSI according to the way the diagram in the rf24ping85.ino is listed still doesn't help. My first impression was how does the tiny84 communicate with slave if MISO on the tiny84 (PA5) is considered DO but there is no DI pin listed for the tiny84 on the MOSI pin (PA6). I mean isn't MISO and MOSI bus only available for programming the tiny84, and if this is true, then how would the tiny84 ever be able to communicate with an NRF24L01? It's very confusing.

zador-blood-stained, 2016-03-29:
> My first impression was how does the tiny84 communicate with slave if MISO on the tiny84 (PA5) is considered DO but there is no DI pin listed for the tiny84 on the MOSI pin (PA6). 

Both official datasheets [[1]](http://www.atmel.com/images/doc8006.pdf) and [[2]](http://www.atmel.com/images/8006s.pdf) list PA6 as DI for USI.

> So reading confusion: #155 , does this mean that in order to keep the MCU from switching between master and slave I need to use the SPI library within my sketch?

No, SPI support code is implemented inside RF24 library, so it should work out of the box.

What software are you using? I assume Arduino IDE with some kind of attiny support library? Last time I checked Arduino IDE versions higher than 1.6.5 didn't work well with google-code attiny core.

Raynman77, 2016-03-30:
Right, and initially thats what I assumed. That SPI is included in the library. Which to me doesn't help, because I have tried the rf24ping85.ino example on the tiny85 chip and the getting started.ino example on a arduino pro mini and it worked with no problems. However if I use the same setup but with the given wiring for a tiny84 (with the rf24ping85.ino on a tiny84), it doesn't transmit. The arduino IDE i'm using is 1.6.5. The only attiny support i'm using is the rf24 libraries provided from the TMRh20 fork.

zador-blood-stained, 2016-03-30:
> The only attiny support i'm using is the rf24 libraries provided from the TMRh20 fork.

And what device do you select in `Tools->Board` menu?

Anyway, I looked again at your first post.

> ATtiny24/44/84 Pin map with CE_PIN 8 and CSN_PIN 7

This says that provided wiring diagram needs `CE_PIN` set to 8 and `CSN_PIN` set to 7, but you left default values - 3 and 4. Please edit this part of your sketch and check again.

```
// CE and CSN are configurable, specified values for ATtiny85 as connected above
#define CE_PIN 3
#define CSN_PIN 4
```

Raynman77, 2016-03-30:
Tried that too and it did not work either.

Raynman77, 2016-03-30:
// CE and CSN are configurable, specified values for ATtiny85 as connected above
#define CE_PIN 8
#define CSN_PIN 7
//#define CSN_PIN 3 // uncomment for ATtiny85 3 pins solution
#include "RF24.h"
RF24 radio(CE_PIN, CSN_PIN);

Raynman77, 2016-03-30:
Ill try again when im not at work. But i did try ce>8 & ce>7 and still didnt work, although i might have made a simple wiring mistake as it was very late at night. Ill post back as soon as get it wired up and tested.

wmarkow, 2016-03-30:
> My first impression was how does the tiny84 communicate with slave if MISO on the tiny84 (PA5) is considered DO but there is no DI pin listed for the tiny84 on the MOSI pin (PA6). I mean isn't MISO [...]

Both ATtiny84 and ATtiny85 doesn't have a hardware SPI interface. They have an USI (Universal Serial Interface) that has a different registers map inside these chips. Standard SPI software library will not work - more over it will not compile at all. However - it is possible to configure the USI interface to act as an SPI interface - this little piece of code is included in RF24 attiny85 example/implementation (it is a SPI over USI implementation).

I do not know why your code doesn't work on ATtiny84. What device (board) have you selected in Arduino IDE? There is a chance that software compiled for ATTiny85 will not work when you upload it to ATtiny84. May it be because of chips internal hardware differences?

zador-blood-stained, 2016-03-30:
> There is a chance that software compiled for ATTiny85 will not work when you upload it to ATtiny84. May it be because of chips internal hardware differences?

Normally programmer (avrdude) won't let you upload if MCU signature doesn't match selected target MCU.

> There is a chance that software compiled for ATTiny85 will not work when you upload it to ATtiny84. May it be because of chips internal hardware differences?

It definitely won't work due to different register and port addresses.

> What device (board) have you selected in Arduino IDE?

And what speed and fuse settings?

Slartitorto, 2016-03-30:
I use nRF24L01 with attiny84 since 1-2 years ago and they work fine together. Thanks to TMRh20 libraries.
Try to call RF24 lib as:
`RF24 radio(CSN pin, CE pin);`
where your "CSN pin" = "PA3) and "CE pin" = "PA2"; so:
`RF24 radio(PA3, PA2);`

[Here my project](http://slartitorto.blogspot.it/2015/01/nrf24l01-and-tmp36-as-remote-temp.html) (I'm sorry, it is in Italian), and it work. I use an old version of TMRh20 libs, so I don't know if the code work with the last release; but you can take a look.

Raynman77, 2016-03-30:
Wmarkow, im using the board selection for the tiny84. Burnt the boot for 8mhz. And using the USBtiny programmer.

Raynman77, 2016-03-30:
Slartitorto, defining the actual port "A" for ce and csn works?

Slartitorto, 2016-03-30:
Raynman77, I believe you are wrong when you assign CE and CSN pin numbers:
```
...
PB3 4| |11 PA2 --- nRF24L01 CE, pin3
PB2 5| |10 PA3 --- nRF24L01 CSN, pin4
...
#define CE_PIN 3
#define CSN_PIN 4
...
RF24 radio(CE_PIN, CSN_PIN);
...
```
In this way, you call RF24 lib with "3" and "4", but they are the pins on nRF24L01 module, they are not the pins on the microcontroller attiny84; I believe you need to pass the correct pin numbers that are attiny84's pins (PA3 and PA2); so I'd try:

```
...
PB3 4| |11 PA2 --- nRF24L01 CE, pin3
PB2 5| |10 PA3 --- nRF24L01 CSN, pin4
...
#define CE_PIN PA2
#define CSN_PIN PA3
...
RF24 radio(CSN_PIN, CE_PIN);
...
```

Raynman77, 2016-03-30:
I will try defining ce and csn with PA2 & PA3 instead of using just  2 & 3 when i get off work. It will be one of the things I do in my testing.

Slartitorto, 2016-03-30:
Be sure to connect:
```
attiny84 pin 11 (PA2) ---> nRF24L01 pin 3 (CE)
attiny84 pin 10 (PA3) ---> nRF24L01 pin 4 (CSN)
```

Also, try both:

`RF24 radio(CSN_PIN, CE_PIN);`
and
`RF24 radio(CE_PIN, CSN_PIN);`

TMRh20, 2016-03-30:
@Raynman77 Thanks for your input. In Arduino IDE if you go to Tools > Boards > Boards Manager, do you find 'ATTiny Core by Spence Konde'  ?

The 'Arduino' pin numbers are defined in a file called pins_arduino.h as per https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/variants/tinyX4/pins_arduino.h#L114

I think the problem lies in the fact that the ATTiny support was built and documented for the old google ATtiny core, which doesn't even exist any more. I think the Arduino pin numbers are different in this core, so the PA2, PA3 thing should work, but in theory it should also work with the corresponding pin numbers. 

Thanks to all the comments, I think the problem is related to the change in ATTiny core, which can be fixed easy enough. I think this came up before, but the issue got closed or something.

zador-blood-stained, 2016-03-30:
@TMRh20 pin map you linked matches google-code pin map for tiny84 ([this](https://github.com/TMRh20/RF24/issues/236#issuecomment-203324723))

TMRh20, 2016-03-30:
@zador-blood-stained Thanks, the darn project page was removed so I couldn't find it! Long thread for such a small chip.

zador-blood-stained, 2016-03-30:
https://code.google.com/archive/p/arduino-tiny/ ?

TMRh20, 2016-03-30:
Yup:
> 404. That's an error.
The project arduino-tiny was not found.

zador-blood-stained, 2016-03-30:
This link works fine for me (sorry for big screenshot)
[image]

Raynman77, 2016-03-30:
No...my board manager shows "attiny by David A. Mellis ver. 1.0.1

Raynman77, 2016-03-30:
I will add the core by Spence Konde when i get off work. I had a feeling this could be an issue..will post back with results later this evening.

TMRh20, 2016-03-30:
What a trip with so many bugs lately... is this all somehow related to the leftpad thing lol?

The google page doesn't work for me in chrome at all. I tried using a VPN also and it doesn't matter where I connect from. It fails in Firefox, but if I refresh it comes up... LOL... I don't even know...

@Raynman77 https://github.com/damellis/attiny/blob/master/variants/tiny14/pins_arduino.h would be Arduino D2 and D3 ??

Raynman77, 2016-03-30:
Gotcha, thanks. Will try D2 and D3 for ce & csn pin defintitions first.

Raynman77, 2016-03-30:
Ok, so quick question. When I go to define the pin using D2 & D3 for as PA2 and PA3, it compiles just fine. However I'm having trouble declaring the correct pin value for my int switchPin, in which my switchPin is the reed switch IO device. I've tried PA1 and 12 but these pins never go HIGH. Bear in mind i'm still testing this using the David Mellis tiny core.

Raynman77, 2016-03-30:
This does not work using David A. Mellis tiny core and I even swapped the MISO and MOSI pins on the breadboard:
`// CE and CSN are configurable, specified values for ATtiny85 as connected above
#define CE_PIN 2
#define CSN_PIN 3
//#define CSN_PIN 3 // uncomment for ATtiny85 3 pins solution
int switchPin = 1;

#include "RF24.h"

RF24 radio(CE_PIN, CSN_PIN);

byte addresses[][6] = {
  "1Node","2Node"};

void setup() {
  pinMode(switchPin, INPUT);
  // Setup and configure rf radio
  radio.begin(); // Start up the radio
  radio.setAutoAck(1); // Ensure autoACK is enabled
  radio.setRetries(15,15); // Max delay between retries & number of retries
  radio.openWritingPipe(addresses[1]); // Write to device address '2Node'
  radio.openReadingPipe(1,addresses[0]); // Read on pipe 1 for device address '1Node'
  radio.startListening(); // Start listening
}

void loop(void){
  
  digitalRead(switchPin);
  digitalWrite(switchPin, HIGH);
  const char payload[] = "Garage Opened";
  if (switchPin == HIGH) {
    radio.stopListening(); // First, stop listening so we can talk.
    radio.write( &payload, sizeof(payload) );
    radio.startListening(); // Now, continue listening
  }
  unsigned long started_waiting_at = micros(); // Set up a timeout period, get the current microseconds
  boolean timeout = false; // Set up a variable to indicate if a response was received or not

  while ( !radio.available() ){ // While nothing is received
    if (micros() - started_waiting_at > 200000 ){ // If waited longer than 200ms, indicate timeout and exit while loop
      timeout = true;
      break;
    }

  }

  if ( !timeout ){ // Describe the results
    unsigned long got_time; // Grab the response, compare, and send to debugging spew
    radio.read( &got_time, sizeof(unsigned long) );
  }

  // Try again 1s later
  delay(1000);
}`

Raynman77, 2016-03-30:
Sorry for the big post, just wanted it up and case anyone spots any issues in the code. However, there shouldn't be considering it runs just fine with the ATtiny85 chip. So I guess my next attempt will be to use a different core for the tiny. The core by Spence Konde mentioned by TMRh20 and zador-blood-stained.Will post back the results.

Raynman77, 2016-03-31:
Does anyone have any tips or suggestions for installing the alternate ATtiny core for the arduino IDE? I'm not sure if it is wise to remove the David A. Mellis attiny core before installing the Spence Konde attiny core, but it seems like the logical thing to do.

Raynman77, 2016-03-31:
I installed the Spence Konde attiny core to the Arduino IDE 1.6.5 and wired everything up, uploaded the code in past posts above with the exception of these changes and is still not working.

`#include "RF24.h"

int switchPin = 1;

RF24 radio(PA2, PA3);
`
evened swapped 'ce & csn' for physical pins on the tiny84 '11 & 10'

Raynman77, 2016-03-31:
A funny thing, when I use the tiny85 chip and program it with the exact same program as above, except I define the reed switch as int switchPin = 1; , in which is the reset pin of the tiny85. When I do this everything works fine with the exception when the NC reed switch is closed the tiny85 is constantly resetting itself due to using the reset pin as an IO without fusing it to act as an IO pin. (Only for testing purposes). Also with the David Mellis core and using a tiny84 chip, I can set the int switchPin = 1; , which is PA1 physical pin location 12 and the internal resistors are set to the pin I commanded pin PA1. However, I am not getting any data transmitted out from the rf24. This is beginning to be very upsetting. I'm almost out of hair to pull out.

TMRh20, 2016-03-31:
I would recommend this:
Delete/remove everything ATtiny using Arduino Board Manager, then verify/delete manually:
     (On Windows)
      Check Documents/Arduino/hardware  -- delete all tiny stuff
      Check C:\Users\<YOURUSER>\AppData\Local\Arduino15\packages -- delete all tiny stuff

Get the google ATtiny core that is tested and the examples were built with: (you may have to use firefox or a browser that works there??) https://code.google.com/archive/p/arduino-tiny/  [Download](https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/arduino-tiny/arduino-tiny-0150-0020.zip)  It should work with the others, but lets go with the known working one.

Run the known working example rf24ping85 using RF24 radio(8,7);  or RF24 radio(PA2,PA3);

Laugh or cry...

Raynman77, 2016-04-01:
I removed everything tiny from my hardware folder within my sketch book folder and removed all the tiny stuff from my C:\Users<myname>\RoamingAppData\Local\Arduino15\packages folder and unzipped and installed the "tiny" folder within my hardware folder in the sketchbook folder, copied the tiny84 internal 8 mhz board entry from the perspective boards.txt file into a file named boards within the sketchbook folder and opened the arduino IDE selected the tiny84 board in the board entry field. Loaded the rf24ping85.ino sketch as before. However before attempting to test it out I used a multimeter to test that the switchPin does in fact show voltage when HIGH. It does not. My entry for the switchPin looks like this:

`#include "RF24.h"

int switchPin = 1;

RF24 radio(PA2, PA3);`
Would int switchPin = 1; be the same as selected PA1 as my input, btw it is called to fuse HIGH in my sketch? Or do I have the wrong pin?

TMRh20, 2016-04-01:
I'm not familiar at all with the switchPin/fuse thing, but the comment https://github.com/TMRh20/RF24/issues/236#issuecomment-203324723 shows the pinmap.

If you specify a number like 9, it uses the D9 arduino pin, else using the PA/PB numbers is pretty straight forward I think. Maybe @zador-blood-stained  would be more knowledgeable than I?

Raynman77, 2016-04-01:
I tested PA1 before setting switchPin = 9 and it was LOW, afterwards I set my variable switchPin = 9 and tested it again and my meter read 3.3 volts concluding that using the straight numeral value works.

Raynman77, 2016-04-01:
I only bought 2 of these 84 chips to play around with. So far it looks like the only thing they are good for at the moment is for running basic scripts.

TMRh20, 2016-04-01:
Yeah, I would probably resort to looking at hardware or coding a [loopback test](http://www.tonylabs.com/arduino-spi-lookback-test.html) with MISO connected directly to MOSI to see if SPI is working or what. Change ``` #include <SPI.h> to <RF24.h> ``` should allow it to compile and work on ATTiny

Raynman77, 2016-04-01:
@TMRh20 Funny you mention that, when I swapped MISO on the radio to PA6 on the 84, and MOSI to PA5, the voltage meter showed a varying voltage indicating that communication was taking place. But I swapped the cables around voltage only showed on PA5.

Raynman77, 2016-04-01:
This is what I get if I try to #include <SPI.h> in my sketch.

"rf24ping84_sw.ino:57:17: fatal error: SPI.h: No such file or directory
compilation terminated.
Error compiling."

TMRh20, 2016-04-01:
GitHub formatting blocked out your ``` #include ``` but if you are getting that error with ``` #include <RF24.h> ```, maybe you need to do the same thing with RF24 library and delete everything RF24 manually before reinstalling with library manager preferably. 

If ATtiny is selected, RF24 will include SPI functions automatically for all supported devices.

Raynman77, 2016-04-01:
@TMRh20 Im getting this error when I attempt to include SPI.h in my sketch

TMRh20, 2016-04-01:
Standard spi lib does not support attiny. Rf24.h has spi code for attiny.

Raynman77, 2016-04-01:
@TMRh20 With the David Mellis RF24 library?

Raynman77, 2016-04-01:
@TMRh20 Just want to be clear on which library to install. I have removed the RF24 libraries from David A. Mellis

TMRh20, 2016-04-01:
This https://github.com/TMRh20/RF24 is the RF24 library.

For the record, the problems at https://code.google.com/archive/p/arduino-tiny/  with chrome were user error (a plugin blocking it).

Thanks for all the input everybody.

zador-blood-stained, 2016-04-01:
> Would int switchPin = 1; be the same as selected PA1 as my input

Yes if your attiny support core was installed and configured correctly. But there are many things that could go wrong (especially since you had to edit/copy-paste board definitions manually).

Without an oscilloscope or logic analyzer I would recommend starting from scratch with simple blinking LED sketch to verify pin mappings, after that test input with pull-up (`switchPin`), and after that start to play with RF24 library and hardware.

Raynman77, 2016-04-01:
@zador-blood-stained I will do that, again. However, I have tested the pins with a meter for simple HIGH and LOW state changes according to the pins that I have selected as IO functions and they are correct as far as IO pin mapping is concerned.

Raynman77, 2016-04-01:
[image]

This is a picture of the blink LED working correctly as "int switchPin = 9;" the LED is attached to pin PA1. You can't see that it is blinking because its a picture, bit it is. Confirming that it is not my pin selection for IO

Raynman77, 2016-04-01:
@TMRh20 I had no problems downloading the zip file for [https://code.google.com/archive/p/arduino-tiny/](url)
I selected this version for download [arduino-tiny-0150-0020.zip](url)

Raynman77, 2016-04-03:
Is there anyone whom has tested this setup and has a working example using Arduino IDE 1.6.5, Dave Mellis Tiny-core, and TMRh20's RF24 fork? I would like to compare with my setup.

Raynman77, 2016-04-03:
Further testing with my meter shows that CE stays high and never goes low and the MISO pin on the nrf24l01+ never shows any voltage. If I swap the MISO & MOSI leads nothings shows up on either on my meter. Why does this sound more like SPI is working correctly? I may be wrong only because I don't have hard proof by using a Logic Analyzer, but when I have more time, I will test it once I get the hardware for doing so.

zador-blood-stained, 2016-04-03:
Unfortunately local store where I buy electronic components doesn't have attiny44/84 in DIP packages, but I will try to test this core with attiny45 and attiny4313 hopefully this week.

Raynman77, 2016-04-03:
I'd be more than happy to send you a tiny84 if you'd prefer. This is driving me crazy. I feel like punching my face in.lol

zador-blood-stained, 2016-04-04:
@Raynman77 
After spending some time testing core and library version with damellis core on tiny85, I checked pin maps for tiny84 again.
[This](https://github.com/damellis/attiny/blob/master/variants/tiny14/pins_arduino.h#L33-L43)
```
// ATMEL ATTINY84 / ARDUINO
//
//                           +-\/-+
//                     VCC  1|    |14  GND
//             (D 10)  PB0  2|    |13  AREF (D  0)
//             (D  9)  PB1  3|    |12  PA1  (D  1) 
//                     PB3  4|    |11  PA2  (D  2) 
//  PWM  INT0  (D  8)  PB2  5|    |10  PA3  (D  3) 
//  PWM        (D  7)  PA7  6|    |9   PA4  (D  4) 
//  PWM        (D  6)  PA6  7|    |8   PA5  (D  5)        PWM
//                           +----+
```
doesn't match google-code core pin map and, most important, it doesn't match tiny84 [pin map](https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L1539-L1545) used in this (RF24) library internally.

**TL;DR** damellis core pin map for tiny84 is not compatible with RF24 library. Please use google-code core instead.

Raynman77, 2016-04-04:
So with the Google tiny-core this actually worked for your 4313? 

I will try this just as soon as I get off today..thank you very much. I hope this works.

zador-blood-stained, 2016-04-04:
> So with the Google tiny-core this actually worked for your 4313?

I used google-code pin map to add attiny4313 support to this library (#152, #155), so it should work. Today I checked only tiny85, because damellis core doesn't support x313 anyway.

The only problem is - google-code core needs to be patched to work with arduino 1.6.x (If I remember correctly), so you can try this version: [tiny.zip](https://github.com/TMRh20/RF24/files/202916/tiny.zip)

Raynman77, 2016-04-04:
I will give this tiny core a go and follow the pin map as well as checking the RF24.cpp file. Will let you know as soon as I have tested it....hey thanks so much @zador-blood-stained

Raynman77, 2016-04-05:
@zador-blood-stained I'm finding some strange behavior. I couldn't get this test run to work, so I decided to load a simple blink sketch to the tiny84. My leds weren't blinking so I took out the meter and tested voltage on all except the reset pin. In the sketch I wrote 

`int ledGreen = PA1;
int ledBlue = PA2;`

.....And when I tested all the pins, I found the blinking on "PB1 and PB2"...?!?!?!

Raynman77, 2016-04-05:
@zador-blood-stained After I wrote a short sketch one after the other for each digitalWrite pin, this is what I get on each write for each pin HIGH.
`//                           +-\/-+
//                     VCC  1|    |14  GND
//             (D 0)  PB0  2|    |13  AREF (D  didn't test)
//             (D  1)  PB1  3|    |12  PA1  (D  9) 
//                     PB3  4|    |11  PA2  (D  8) 
//  PWM  INT0  (D  2)  PB2  5|    |10  PA3  (D  7) 
//  PWM        (D  3)  PA7  6|    |9   PA4  (D  6) 
//  PWM        (D  4)  PA6  7|    |8   PA5  (D  5)        PWM
//                           +----+`

[61 more replies]

closed 2016-04-05 by Raynman77